grblHAL/STM32F4xx #1: STM32 hardfault saving settings

dresco, 2021-03-12:
I'd noticed that my STM32F411 Nucleo test board was ending up in a hard fault after updating settings. Seems to be coming from here;

https://github.com/grblHAL/STM32F4xx/blob/906d7b12a72548fd64c8faa2a7659967c29c9413/Src/flash.c#L61-L69

nvs_alloc() is adding some extra bytes to hal.nvs.size, making it no longer divisible by four;

`hal.nvs.size = GRBL_NVS_SIZE + hal.nvs.driver_area.size + 1;`

Because of this, remaining never hits 0 (for me, it starts at 1034, wrapping around from 2), and the flash write just carries on until it presumably runs outside of the eeprom emulation area..

I expect this will impact the F1 and F3 drivers also..

terjeio, 2021-03-12:
> nvs_alloc() is adding some extra bytes to hal.nvs.size, making it no longer divisible by four;

The extra byte is for the checksum. Adding this line at the start of nvs_buffer_init()

`hal.nvs.size = ((hal.nvs.size - 1) | 0x03) + 1;`

so it becomes:

```
//
// Switch over to RAM based copy.
// Changes to RAM based copy will be written to physical storage when grblHAL is in IDLE state.
bool nvs_buffer_init (void)
{
    hal.nvs.size = ((hal.nvs.size - 1) | 0x03) + 1;

    if(nvsbuffer) {
...
```

should fix it for all drivers.

dresco, 2021-03-12:
Thanks, I have added that line in nvs_alloc(), just after the existing size adjustment and it looks good (size is rounded up to 1036 & no errors)..

```
hal.nvs.size = GRBL_NVS_SIZE + hal.nvs.driver_area.size + 1;
hal.nvs.size = ((hal.nvs.size - 1) | 0x03) + 1;
```

closed 2021-04-30 by terjeio