grblHAL/STM32F4xx #3: Issue with Black Pill

troth530, 2021-04-27:
Hi, I use a typical gCode sender (commercial) and get this error.  
Is there some gCode standard that is being used for the core or is this a driver issue?
Is this because of advanced gCode commands and is there a way to disable to be compatible with typical senders?
Thanks!

During a typical jog or any movement I get this error.
error:9
G-code locked out during alarm or jog state.

terjeio, 2021-04-27:
Error 9 is due to the controller beeing in an alarm state. This could be due to grblHAL defaulting to normally closed (NC) switches, in new setups typically because the reset input is not connected to ground. Check this [wiki page](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings) for tips.

troth530, 2021-04-30:
@terjeio  thank you for all of your hard work. Issue is closed.

closed 2021-04-30 by troth530