grblHAL/STM32F4xx #4: Compile error - driver.c - no member named 'write_char'

keithstrachan, 2021-04-27:
when compiling within STM32CubeIDE 1.6, selecting any configuration, the following error is displayed.

../Src/driver.c: In function 'driver_init':
../Src/driver.c:1449:15: error: 'io_stream_t' {aka 'struct <anonymous>'} has no member named 'write_char'
 1449 |     hal.stream.write_char = serialPutC;
      |               ^
arm-none-eabi-gcc "../Src/syscalls.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DDEBUG '-DPID_LOG=1000' -DNUCLEO_F446 -DSTM32F446xx '-DUSB_ENABLE=1' -c -I../Inc -I"C:/Users/keith/STM32CubeIDE/workspace_1.4.0/GRBL Driver STM32F401" -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../FatFS -I../USB_DEVICE/App -I../USB_DEVICE/Target -Og -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Src/syscalls.d" -MT"Src/syscalls.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Src/syscalls.o"
arm-none-eabi-gcc "../Src/sysmem.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DDEBUG '-DPID_LOG=1000' 

please let me know if you need any further information

terjeio, 2021-04-27:
It looks like your [grblHAL core](https://github.com/grblHAL/core/blob/e1554ddd863b17b53b1cf00efff89040ca56cc57/grbl.h#L37) is outdated. You need to pull inn the latest submodules when updating the repo. See [the wiki](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL) for info.

keithstrachan, 2021-04-28:
Thanks terjeio, 
i need to look at my sync.. it said it was, clearly didn't..
i created new repo, sync'd and it's all ok.
closing issue as resolved, operator error ;)

keithstrachan, 2021-04-28:
need to ensure that your repository is upto date, or get a fresh copy of the required repo from Github and retest before opening an issue.

closed 2021-04-28 by keithstrachan