grblHAL/STM32F4xx #5: Build Fail for F411 black pill

envyvan, 2021-05-02:
Hi forks,

I git clone the project currently, It fail with the message below:

21:54:40 **** Incremental Build of configuration Debug for project GRBL Driver STM32F401 ****
make -j32 all 
arm-none-eabi-gcc "../Src/driver.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DDEBUG '-DN_AXIS=4' -DSTM32F411xE -c -I../Inc -I"D:/CNC/Minimal-Black-Pill--STM32F4xx-BOB-for-grblHAL-main/Minimal-Black-Pill--STM32F4xx-BOB-for-grblHAL-main/STM32F4xx" -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../FatFS -I../USB_DEVICE/App -I../USB_DEVICE/Target -Og -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Src/driver.d" -MT"Src/driver.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Src/driver.o"
../Src/driver.c:1454: error: unterminated #else
 1454 | #if USB_SERIAL_CDC
      | 
make: *** [Src/subdir.mk:63: Src/driver.o] Error 1
"make -j32 all" terminated with exit code 2. Build might be incomplete.

It show something happend about the error of the USB_SERIAL_CDC.
I disable the "define USB_SERIAL_CDC" in the my_machine.h file since the usb funtion of my blackpill was not working.
I was able the build the firmware using the old grblHAL-master file with the same config.
[image]

terjeio, 2021-05-02:
Strange, the missing #endif is in the repo and in all previous versions as well. Did you somehow delete it by accident?

https://github.com/grblHAL/STM32F4xx/blob/905db6be74218f9437fff79cfd68845f7b4ae42c/Src/driver.c#L1456-L1460

envyvan, 2021-05-03:
Cool, It works.
Thanks for helping.
I don't think I will change any code coded the except the configuration.

terjeio, 2021-05-03:
Glad to hear that.

closed 2021-05-03 by terjeio