grblHAL/STM32F4xx #8: STM32F407 build failing

vumario, 2021-05-18:
Hi,

I'm trying to build grblHAL for an SKR PRO board that is based on STM32F407 mcu, but there seem to be some issues. 

First thing I came across was that there was CMSIS header missing for STM32F407 and after some time trying to figure it out, I found that in commit 5457ad062da78fd599a50880a5c680ce780f9960 those headers were deleted. So after copying those headers back, I could build the code but only if I disable USB_SERIAL_CDC in my_machine.h.

If I try to leave the config as is with USB_SERIAL_CDC enabled, I get following errors:

c:\st\stm32cubeide_1.6.1\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: Src/stm32f4xx_it.o: in function `OTG_FS_IRQHandler':
stm32f4xx_it.c:(.text.OTG_FS_IRQHandler+0x8): undefined reference to `hpcd_USB_OTG_FS'
c:\st\stm32cubeide_1.6.1\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: Src/usb_serial.o: in function `usb_write':
usb_serial.c:(.text.usb_write+0x1c): undefined reference to `CDC_Transmit_FS'
c:\st\stm32cubeide_1.6.1\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: usb_serial.c:(.text.usb_write+0x32): undefined reference to `CDC_Transmit_FS'
c:\st\stm32cubeide_1.6.1\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: Src/usb_serial.o: in function `usbInit':
usb_serial.c:(.text.usbInit+0x2): undefined reference to `MX_USB_DEVICE_Init'
c:\st\stm32cubeide_1.6.1\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: Src/usb_serial.o: in function `usbPutC':
usb_serial.c:(.text.usbPutC+0xc): undefined reference to `CDC_Transmit_FS'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:73: GRBL Driver STM32F401.elf] Error 1

That was about as far as I could get with my knowledge, so any help would be appreciated.

terjeio, 2021-05-18:
I recently switched to the latest ST driver code, is your download recent/complete as the source files needed are in the repo in the [USB_DEVICE](https://github.com/grblHAL/STM32F4xx/tree/master/USB_DEVICE) folder?

This is what I get when I search for `CDC_Transmit_FS`:

[image]

terjeio, 2021-05-18:
It looks like there are missing source locations for the build, try with changing to match this via _File > Properties_ for the project:

[image]

vumario, 2021-05-18:
Thanks, It's working now,

Middlewares and USB_DEVICE source folders were missing, after adding them, I can successfully build the project.

closed 2021-05-18 by vumario

terjeio, 2021-05-18:
Ok, sorry for causing problems.

I guess you are going to make a board map file for the SKR PRO, please make a PR for that if so.

Ethernet support would be nice to have? I have a STM32F756 Nucleo board waiting for me to write a driver for it, it has an Ethernet port as well and I intend to support that. With luck the code needed for F407 will be very similar to, or even the same as the code for the F756 and thus easy to port over...

vumario, 2021-05-18:
Shure, I'll make a PR as soon as I have a working board.

Ethernet support would be very nice to have. Looking forward to it...