grblHAL/STM32F4xx #11: Possible typo in driver.c

kfmut, 2021-05-27:
Hi,

I'm trying to understand pin mapping for STM32 MCU's and noticed that in the pin assignment function for steppers rotation direction STEP_OUTMODE is used in place of DIRECTION_OUTMODE(line 315). Should it be that way?

https://github.com/grblHAL/STM32F4xx/blob/b9426a4f1b321b77ff9dede7ff67df362c42a2c9/Src/driver.c#L311-L325

Also I'm curious what is the practical difference from end-user point of view between different INMODE and OUTMODE options: GPIO_SHIFTx, GPIO_MAP, GPIO_BITBAND?

As I've understood GPIO_SHIFTx is for pins going one after another on the same port, GPIO_MAP is for pins on the same port and GPIO_BITBAND for pins on different ports. Is there any speed difference or other valuable difference for using different options? I'm really too far from low-level MCU programming so I’m lacking any expertice here.

terjeio, 2021-05-27:
> Should it be that way?

No, and it was fixed the other day when I refactored the code. See PR #9. I'll likely commit the changed code today or tomorrow.

> Also I'm curious what is the practical difference from end-user point of view between different INMODE and OUTMODE options

Not really, but I do not like to waste processor cycles. At least the step outputs should be handled as fast as possible as every cycle added decreases the max step rate.

>  Is there any speed difference or other valuable difference for using different options?

No significant differences, shift is fastest but only a few instructions less than map. Bitband is slowest but flexible as pins can be assigned to different ports.

kfmut, 2021-05-27:
@terjeio That's nice, thank you! Now it's time for me to change mapping for my Blackpill in a more thoughtful way.

> See PR #9. I'll likely commit the changed code today or tomorrow.

Wow, 6-axis for stm32f4, it’s almost too much :-) But maybe it's not THAT late to make numbered axes like in fra589’s grbl port for mega2560 https://github.com/fra589/grbl-Mega-5X/blob/edge/grbl/config.h
So they could be relatively easily cloned, renamed? Also auto squaring wouldn’t be too much of a problem, simple change of homing procedure:

```
#define HOMING_CYCLE_0 (1<<AXIS_3) // Z
#define HOMING_CYCLE_1 (1<<AXIS_1) // X
#define HOMING_CYCLE_2 ((1<<AXIS_2)|(1<<AXIS_4)) // Y
```

closed 2021-05-27 by kfmut

terjeio, 2021-05-27:
> But maybe it's not THAT late to make numbered axes like in fra589’s grbl port for mega2560

It is not too late but I need a compelling use case before spending time on changing the core for that. Senders has to support different lettering too for it to be useful...?

> Also auto squaring wouldn’t be too much of a problem, simple change of homing procedure:

The homing procedure in the core already supports auto squaring, the driver has to be expanded to support it. An there is no need to recompile to set up the procedure as settings for this, and many other, compile time settings are possible to configure at run time with [$ settings](https://github.com/grblHAL/core/wiki/Additional-or-extended-settings).

kfmut, 2021-05-27:
@terjeio

> It is not too late but I need a compelling use case before spending time on changing the core for that.

Wouldn't it benefit to one of roadmap targets(https://github.com/grblHAL/core/wiki/Roadmap)

`Add uniform way to specify ganged/auto-squared axes to my_machine.h. Add this option to drivers capable of handling it.`

for all drivers simultaneously?

Specific breakout boards with corresponding map files could almost be treated as black boxes with N available axes and some basic inputs/outputs(cycle start button, hold button, probe, spindle enable, etc). In some distant future it could go as far as listing pins capable of fast and accurate pulse generation for step-signals that could be assigned to any LETTER-axis, some pins which should/could be handled via interrupts, some general IO pins for non-realtime machine controls or even for specific M-codes. For now it’s at least a straightforward configuration for end-users with different(!) requirements via single file.

Maybe it’s already realised in other drivers? I’ve seen that for some drivers configuration is simpler and more feature rich(auxiliary IO pins)...which driver is most advanced and what place driver for STM32 MCUs helds in that rating?

> Senders has to support different lettering too for it to be useful...?

I'm OK with relatively simple senders lacking rich visualisation capabilities(long term bCNC user here) so maybe I'm missing potential problems with senders? Or it is not about visualisation?

> An there is no need to recompile to set up the procedure as settings for this, and many other, compile time settings are possible to configure at run time with [$ settings](https://github.com/grblHAL/core/wiki/Additional-or-extended-settings).

Thank you for pointing that out. I'm still thinking in classical grbl way :-)

terjeio, 2021-05-27:
> Wouldn't it benefit to one of roadmap targets(https://github.com/grblHAL/core/wiki/Roadmap)
Add uniform way to specify ganged/auto-squared axes to my_machine.h. Add this option to drivers capable of handling it.
for all drivers simultaneously?

Sure, but it does not do itself - and auto-squaring is not just about writing code hoping it works, it takes a lot of time to set up and verify for each driver. And some boards do not have the pins required and are not widely used either so I see no reason to spend my time on work that does not benefit anybody or only a few. Get a more capable controller instead.

> Specific breakout boards with corresponding map files could almost be treated as black boxes with N available axes and some basic inputs/outputs(cycle start button, hold button, probe, spindle enable, etc). 

This is what the drivers (and plugins) represents for the grblHAL core, still the contents of the black box has to be made. I do have longer term plans for pin assignments, at least some should be possible to assign at run time.

> Maybe it’s already realised in other drivers?

I am just about to commit an update for some that uses the style that you see in the PR mentioned.

> I’ve seen that for some drivers configuration is simpler and more feature rich(auxiliary IO pins)...which driver is most advanced and what place driver for STM32 MCUs helds in that rating?

iMXRT1062, STM4xx and MSP432 based drivers are among the most advanced. The RP2040 (Pi Pico) can become interesting...
I want to see networking added for the STM4xx driver, I have a STM32F756 dev board I want to make a driver for (it has tons of I/O) - the network code I am going to add for that should be possible to backport.

It can be mentioned that I have just managed to get FTP access up and running for SD card management, this will IMO open up new avenues for standalone controllers...

STM32F103 is on its way out, it has limited memory - it is right at the edge of beeing usable. So I am not going to put any effort into bringing that up to speed. It will at some point be left frozen against the latest core that it can support.

>  Or it is not about visualisation?

Visualising the DRO is part of that, graphical visualisation is another story... My own sender supports up to 6 axes, and it will be some work to change the parser to support new axisletters, I would need a very good reason for updating that. Not many supports more than 3 so less of a issue for them...

kfmut, 2021-05-28:
@terjeio Thank you for the elaborate answer and insight into future plans!

> Get a more capable controller instead.

Sorry, my assumption was that any modern 32-bit MCU would be more capable than dated atmega328 MCU, my router setup with last grbl release has auto-squaring.

terjeio, 2021-05-28:
> Sorry, my assumption was that any modern 32-bit MCU would be more capable than dated atmega328 MCU, my router setup with last grbl release has auto-squaring.

Some of the 32-bit controllers grblHAL has drivers for are limited when it comes to number of GPIO pins, some does not have a FPU and some does not have the peripherals needed for stuff such as spindle sync - and finally I do not have unlimited time to implement every feature possible for all of them. My efforts will gravitate towards those controllers that are most capable and/or popular among users. If a specific feature is lacking for a given controller/board either ask me to add it or add the needed code yourself, test it and create a PR for it.

kfmut, 2021-05-28:
@terjeio 

> My efforts will gravitate towards those controllers that are most capable and/or popular among users.

Now I understand that and I will spread the word about it so others interested in grblHAL would be looking into decent controllers right from the start :-)