grblHAL/STM32F4xx #12: I2C on Blackpill boards

kfmut, 2021-05-27:
Hi,

I'm curious to try grblHAL on my router and the keypad plugin is one of my prime interests in this endeavour. For now I have a Blackpill devboard on STM32F411 MCU from WeActStudio. My problem is that I2C pins by default are mapped to PB10 and PB11 but PB11 on Blackpill boards is not routed to headers:

[image]

Is it OK to change this check and add CNC_BLACKPILL to line 206?

https://github.com/grblHAL/STM32F4xx/blob/b9426a4f1b321b77ff9dede7ff67df362c42a2c9/Inc/driver.h#L205-L211

So it would look something like this:
`#if defined(NUCLEO_F411) || defined(NUCLEO_F446) || defined(CNC_BLACKPILL)`

Would it be possible to connect I2C-devices like EEPROM and keypad to PB8 and PB9 pins after that? Any side effects to this?

terjeio, 2021-05-27:
> Would it be possible to connect I2C-devices like EEPROM and keypad to PB8 and PB9 pins after that?

Yes, but only if the board map you intend to use do not assign these pins to something. You can modify the board map if there is a clash, even copy it to _my_machine_map.h_ and select that in _my_machine.h_. If you copy it it will not be overwritten on an update.

kfmut, 2021-05-27:
@terjeio Thank you for clarification!

Yes, I'm intending to create a custom mapping file for my current tests since the stock file also has conflict between PROBE and KEYPAD_STROBE signals.

closed 2021-05-27 by kfmut