grblHAL/STM32F4xx #14: Blink led PC13

Juank23, 2021-05-29:
I would like to know how to add the led PC13 so that when it connects with the program it blinks like rx and tx it remains pending thanks

terjeio, 2021-05-29:
Blinking while there is traffic can be achieved by adding a plugin that redirecs input to itself and blinks the LED when there is activity. It is not possible to detect connect/disconnect events for UART traffic, if serial over USB it is perhaps possible to hook into the ST USB driver somehow. But I do not know if it is possible or how to do it - you have to read the documentation or study the USB cod to find out.

The [YModem protocol](https://github.com/grblHAL/Plugin_SD_card/blob/master/ymodem.c) is an example for how to listen for input, it does that by redirecting  the `grbl.on_unknown_realtime_cmd` However you would want to do the same by redirecting `hal.stream.enqueue_realtime_command` instead as all input passes over this,  `grbl.on_unknown_realtime_cmd` is only called when an unknown realtime command character is received.

If you want to blink the LED at a predefined interval then redirect `grbl.on_execute_realtime` to your code and keep track of time by calling _hal.get_elapsed_ticks()_ . Set a flag when there is traffic to know when to blink. Tics are incremented every millisecond. Again the YModem protocol code can act as an example of how to do this.

A plugin can be added without modifying a single line of code in the controller if you implement it by adding it in a [my_plugin.c](https://github.com/grblHAL/Templates/blob/master/my_plugin.c) file.

Juank23, 2021-05-29:
Good morning, thanks for your answer, if I wanted to know how to make the controller's ledPC13 blink, I'm a newbie in this, thank you very much

Juank23, 2021-05-29:
[image]
good afternoon, this is it, could you help me with an example ... thank you very much

terjeio, 2021-05-29:
Here is how I would blink the LED from within grblHAL:

```
#include "driver.h"

static on_report_options_ptr on_report_options;
static on_execute_realtime_ptr on_execute_realtime;

// Add info about our plugin to the $I report.
static void on_report_my_options (bool newopt)
{
    on_report_options(newopt);

    if(!newopt)
        hal.stream.write("[PLUGIN:Blink LED v1.00]" ASCII_EOL);
}

static void blink_led (sys_state_t state)
{
    static bool led_on = false;
    static uint32_t ms = 0;

    if(hal.get_elapsed_ticks() >= ms) {
        ms = hal.get_elapsed_ticks() + 500; //ms
        led_on = !led_on;
        if(led_on)
            GPIOC->ODR |= GPIO_PIN_13;
        else
            GPIOC->ODR &= ~GPIO_PIN_13;
    }

    on_execute_realtime(state);
}

void my_plugin_init (void)
{
    // Add info about our plugin to the $I report.
    on_report_options = grbl.on_report_options;
    grbl.on_report_options = on_report_my_options;

    // Add blink LED function to grblHAL foreground process
    on_execute_realtime = grbl.on_execute_realtime;
    grbl.on_execute_realtime = blink_led;

    // Enable PC13 as output
    GPIO_InitTypeDef GPIO_InitStructure = {
        .Mode      = GPIO_MODE_OUTPUT_PP,
        .Speed     = GPIO_SPEED_FREQ_VERY_HIGH,
        .Pin       = GPIO_PIN_13,
    };
    HAL_GPIO_Init(GPIOC, &GPIO_InitStructure);
}

```
Create _Src/my_plugin.c_ , copy this code and recompile/flash and the LED is flashing once every second..

Juank23, 2021-05-29:
Thank you very much friend, excellent work

closed 2021-05-29 by Juank23