grblHAL/STM32F4xx #15: No motion after latest updates

kfmut, 2021-05-29:
Hi,

I was still experimenting with my Blackpill F411 and noticed that you have updated stm32f4 driver(including i2c pins fix) so I've also updated source to latest and after that motion has stopped. Inputs are working. Pins map file is stock.

Backdated my source to May 18 and it's working, updated to latest - no luck.

What I'm doing wrong? :-)

$I:
```
[VER:1.1f.20210517:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:ENUMS,RT+,TC]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32F411]
[DRIVER VERSION:210525]
[BOARD:Minimal BlackPill grblHAL]
```

$$:
```
$0=10.0
$1=25
$2=0
$3=0
$4=0
$5=7
$6=1
$7=0
$10=511
$11=0.010
$12=0.002
$13=0
$14=7
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0
$21=0
$22=0
$23=0
$24=25.0
$25=500.0
$26=250
$27=1.000
$28=0.100
$29=0.0
$30=1000.000
$31=0.000
$32=0
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$62=0
$63=2
$64=0
$65=0
$80=1.000
$81=0.010
$82=0.000
$84=0.000
$85=10.000
$90=0.000
$91=0.000
$92=0.000
$95=0.000
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=100.0
$347=5.0
$348=2.500
$349=25.000

```

terjeio, 2021-05-29:
Could it be that the  your grbl folder is out of sync?

There is a new file [grbl/stepdir_map.h](https://github.com/grblHAL/core/blob/master/stepdir_map.h) that has been modified, check that your version has this symbol defined at line 188:

`#define USE_STEPDIR_MAP 1`

kfmut, 2021-05-29:
@terjeio

Yes, it has. I've updated my source with
`git pull --recurse-submodules`

terjeio, 2021-05-29:
Ouch, a regression - add these lines after `#if !TRINAMIC_ENABLE` at line 155 in driver.c:
```
#ifdef STEPPERS_DISABLE_PORT
    { .id = Output_StepperEnable,   .port = STEPPERS_DISABLE_PORT,      .pin = STEPPERS_DISABLE_PIN,    .group = PinGroup_StepperEnable },
#endif
```
I'll commit a fix later, will be busy for a while. Please report back.

kfmut, 2021-05-29:
@terjeio

Stepper started to make noise but look like it's making one single step per single jog. If I start nc-file from ioSender single step is made at beginning after that no movement.

kfmut, 2021-05-29:
@terjeio 

Actually stepper connected to X-axis makes single step if I jog ANY axis.

terjeio, 2021-05-29:
Try with $4=7

kfmut, 2021-05-30:
@terjeio

> Try with $ 4 = 7

Yep, that obviously helps alot, looks like I've forgot to re-establish this for driver.c fix after countless reflashings. I think those single steps when I jogged other axis were connected with microstepping when driver failed to mantain stepper position, so after each stepper enable signal steppers rotor was returned to intended position. Couldn't confirm that, I've proper toasted my test drv8825 driver :-)

Issue could be closed if you are OK with this results.

closed 2021-06-13 by terjeio