grblHAL/STM32F4xx #17: option to set output pins to OD

langwadt, 2021-06-05:
Would it not make sense to add the option to set output pins to opendrain? Many external drives with optocoupler inputs spec a minimum of 5V and since most pins on the STM32 is 5V tolerant that would be possible with the output set as OD

terjeio, 2021-06-05:
A compile time option specified in the map file and per [pin group or per pin](https://github.com/grblHAL/core/blob/master/crossbar.h)? Or a run time configuration with new $-settings bound to the map file? 

It has to be per map file as some boards buffer outputs and changing the pins to OD for these may render them unusable.

What about the risk of frying the MCU if 5V is applied to a non-OD output pin? Are STM32 pins still 5V tolerant in push-pull (PP) mode?

langwadt, 2021-06-05:
compile time option per group I'd say. The pins are not 5V tolerant in PP or with pullup enabled, but they will never see 5V, only 5V minus the forward voltage of the opto and a resistor 

My drivers (DM556) seems to work with 3.3V PP but it is outside spec, with 5V and PP the optos never turn off, with 5V and OD it works and is within spec

terjeio, 2021-06-05:
Ok, then it will be like this:

```
...
#define STEP_OUTMODE            GPIO_MAP
//#define STEP_PINMODE            PINMODE_OD // Uncomment for open drain outputs
...
```
same for direction and enable signals.

closed 2021-06-06 by langwadt

terjeio, 2021-06-06:
Change is now up.

langwadt, 2021-06-06:
thanks, I'll try it out

langwadt, 2021-06-07:
works as expected. I must say that a getting fork of something with changes in submodules updated seems less that intuitive

dresco, 2021-06-08:
> I must say that a getting fork of something with changes in submodules updated seems less that intuitive

Wouldn't disagree, but fwiw you can configure your repo to automatically update submodules when pulling changes..

`git config --local submodule.recurse true`