grblHAL/STM32F4xx #18: The USB virtual serial port test failed for F446RET6

hankecnc, 2021-06-13:
Hello, I am an amateur CNC enthusiast from China. Previously, I learned and made the circuit board based on the grbl version of https://github.com/robomechs/6-AXIS-USBCNC-GRBL on this page. My Taobao store sells (an online store, web address https://item.taobao.com/item.htm?spm=a2oq0.12575281.0.0.18a21debZt82Qh&ft=t&id=639866857811), the effect is not bad, but because of the price of the F103C8T6 chip It skyrocketed, and I had to stop sales (because the circuit board is sold at a low price, and the postage is only 45 yuan, so I can't afford the price increase of the chip), I saw your grblhal a few days ago, and I bought a F446RET6 minimum system board , I tested your grblhal using the serial port, it is very good, supports all drilling cycles, and many new things, but when using the USB virtual serial port through #define USB_SERIAL_CDC 1 the computer recognition fails (the USB virtual serial port driver has been installed, and Using 6-AXIS-USBCNC-GRBL is normal), I'm pretty sure it doesn't seem to be my problem, my F446RET6 circuit board uses USB1, that is, PA11, PA12, I also confirm that it is not a line connection problem, because if it is not enabled #define USB_SERIAL_CDC 1, there will be no response when connecting to the computer via USB, so I am very sure that it is a program problem. I don't want to connect to the computer through the ch340 serial port. On the one hand, it increases the unreliability, and on the other hand, welding will also occur. Bring a little trouble,

hankecnc, 2021-06-13:
I don't know how to program MCU, so I can't solve this problem

terjeio, 2021-06-13:
Which F466 development board? Does it have a crystal for the main oscillator and what is the frequency?
Have you removed the `NUCLEO_F446` symbol from the compilation?

I am asking since the clock frequency supplied to the USB peripheral is critical and I have not verified the clock setup for USB when building for F446 as I have no board to check with.

hankecnc, 2021-06-13:
After enabling USB, it is responsive to plug in the computer, but the unknown device is recognized, it is likely to be a problem with the program.

hankecnc, 2021-06-13:
I used the F446RET6 circuit board purchased by Taobao. I tested his USB with a multimeter. It is the same as F103C8T6, which is also PA11 and 12.
[image]

hankecnc, 2021-06-13:
He is using 8M crystal oscillator, using GRBLhal serial port pb2, pb3 communication, he works very well

hankecnc, 2021-06-14:
Regarding your source code, I swear that I haven't made too many changes, I am the grblhal core program on this page https://github.com/grblHAL/core
Download it with https://github.com/grblHAL/STM32F4xx to your computer, and put all the files after core-master decompression into the STM32F4xx-master grbl folder, click the 🔨 icon, select Release F446 Nucle064, in my_machine. h #define BOARD_MORPHO_CNC, and #define USB_SERIAL_CDC 1
Then compile, there will be an error message, I deleted that #if defined(IS_NUCLEO_BOB)&& USB_SERIAL_CDC...#endif,
Compile again without any errors, and download the program to the microcontroller,
Other than that, nothing else has been changed

[image]
[image]
[image]
[image]
[image]

hankecnc, 2021-06-14:
[STM32F4xx-masterusbusbusb.zip](https://github.com/grblHAL/STM32F4xx/files/6644899/STM32F4xx-masterusbusbusb.zip)
This is my modified file, it can work normally with b2 and b3 serial port

hankecnc, 2021-06-14:
I have installed st's USB virtual serial port driver, https://github.com/robomechs/6-AXIS-USBCNC-GRBL USB virtual serial port on this page can work normally

terjeio, 2021-06-14:
For now replace `SystemClock_Config() `in main.c with this code:

```
void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};
  RCC_PeriphCLKInitTypeDef PeriphClkInitStruct = {0};

  /** Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = 7;
  RCC_OscInitStruct.PLL.PLLN = 294;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 7;
  RCC_OscInitStruct.PLL.PLLR = 7;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
  /** Initializes the CPU, AHB and APB buses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK)
  {
    Error_Handler();
  }
  PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_CLK48;
  PeriphClkInitStruct.Clk48ClockSelection = RCC_CLK48CLKSOURCE_PLLQ;
  if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
}
```
Then remove the NUCLEO_F446 compilation symbol and add symbol HSE_VALUE set to 8000000 - [see here](https://github.com/grblHAL/STM32F4xx) for where to do that.

This should hopefully set up the clock tree correctly for USB operation.

closed 2021-06-16 by hankecnc

terjeio, 2021-06-16:
Is USB working with the changed code?

hankecnc, 2021-06-16:
After I added the HSE_VALUE setting to 8000000 and replaced the content of the SystemClock_Config() function, the USB virtual serial port was still not recognized successfully.
I'm really not too embarrassed to ask you for help again.

It doesn't matter, I am planning to change the firmware of the grbl control board I sold from https://github.com/robomechs/6-AXIS-USBCNC-GRBL to your grblhal, I have tested your firmware on this page https://github.com/robomechs/6-AXIS-USBCNC-GRBL ://github.com/grblHAL/STM32F1xx, the USB virtual serial port can work normally,

hankecnc, 2021-06-16:
> USB 是否可以使用更改后的代码？

After I added the HSE_VALUE setting to 8000000 and replaced the content of the SystemClock_Config() function, the USB virtual serial port was still not recognized successfully.
I'm really not too embarrassed to ask you for help again.

It doesn't matter, I am planning to change the firmware of the grbl control board I sold from https://github.com/robomechs/6-AXIS-USBCNC-GRBL to your grblhal, I have tested your firmware on this page https://github.com/robomechs/6-AXIS-USBCNC-GRBL ://github.com/grblHAL/STM32F1xx, the USB virtual serial port can work normally,

hankecnc, 2021-06-16:
> USB 是否可以使用更改后的代码？

我刚刚查看了您的grblhal F103版本,引脚与6-AXIS-USBCNC-GRBL 相同 ,非常好,谢谢你

terjeio, 2021-06-16:
> I'm really not too embarrassed to ask you for help again.

Well, I am here to help and it is in my interest to get it working but it is not possible for my to test the code when I do not have a board to check with. So to get it to work I need your help/input or I have to buy the same development board so I can test myself.

The F4xx driver supports a number of different variants and I know the USB code itself is working as I have a F411 board that runs fine. It can be mentioned that I have had problems with getting the clock tree configuration right for both F3 and F4 series when it comes to USB operation, if the frequency is not 100% correct it will not work.

> ... STM32F1xx, the USB virtual serial port can work normally.

This firmware works with the F446 processor? The F1 and F4 series are very different so I am surprised if this is true.

hankecnc, 2021-06-16:
No, you misunderstood, this may be a problem caused by Google Translate.

Since the previous GRBLHAL F4XX USB virtual serial port test failed, I am now testing your GRBLHAL F103.
I am planning to replace your GRBL HAL F1XX firmware with the grbl F103 control board that I am selling

[image]

This is a grbl control card that I learned about PCB boards and made. I like CNC very much. My job is also CNC. I like the open source nature of grbl very much.

hankecnc, 2021-06-18:
> 现在`SystemClock_Config() `用以下代码替换main.c 中：
> 
> ```
> void SystemClock_Config(void)
> {
>   RCC_OscInitTypeDef RCC_OscInitStruct = {0};
>   RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};
>   RCC_PeriphCLKInitTypeDef PeriphClkInitStruct = {0};
> 
>   /** Configure the main internal regulator output voltage
>   */
>   __HAL_RCC_PWR_CLK_ENABLE();
>   __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
>   /** Initializes the RCC Oscillators according to the specified parameters
>   * in the RCC_OscInitTypeDef structure.
>   */
>   RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
>   RCC_OscInitStruct.HSEState = RCC_HSE_ON;
>   RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
>   RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
>   RCC_OscInitStruct.PLL.PLLM = 7;
>   RCC_OscInitStruct.PLL.PLLN = 294;
>   RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
>   RCC_OscInitStruct.PLL.PLLQ = 7;
>   RCC_OscInitStruct.PLL.PLLR = 7;
>   if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
>   {
>     Error_Handler();
>   }
>   /** Initializes the CPU, AHB and APB buses clocks
>   */
>   RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
>                               |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
>   RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
>   RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
>   RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
>   RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;
> 
>   if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK)
>   {
>     Error_Handler();
>   }
>   PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_CLK48;
>   PeriphClkInitStruct.Clk48ClockSelection = RCC_CLK48CLKSOURCE_PLLQ;
>   if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK)
>   {
>     Error_Handler();
>   }
> }
> ```
> 
> 然后删除 NUCLEO_F446 编译符号并添加符号 HSE_VALUE 设置为 8000000 -[请参阅此处](https://github.com/grblHAL/STM32F4xx)了解在哪里执行此操作。
> 
> 这应该有望为 USB 操作正确设置时钟树。

Hi, hello, I tried your method again today, and it succeeded. It's exciting, your code is correct!!

hankecnc, 2021-06-18:
> 现在`SystemClock_Config() `用以下代码替换main.c 中：
>
> ```
> void SystemClock_Config(void)
> {
>   RCC_OscInitTypeDef RCC_OscInitStruct = {0};
>   RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};
>   RCC_PeriphCLKInitTypeDef [...]

I would like to ask you another question. Is grbl difficult to implement for servo spindle tapping? As far as I know, for tapping, the spindle speed has a strict relationship with the feed rate of the Z axis. I think if grbl can tap Is grbl perfect for 3-axis CNC?

terjeio, 2021-06-19:
Great that the USB now works, I was about to get a USB socket so that I could test with my Nucleo-64 board - not needed anymore. I will add this change to the next commit including a build configuration for it.

Tapping should be possible to implement, how difficult I do not know. Adding it would require access to a machine for testing and likely a lot of broken taps? It is not a top priority for me to add it as I do not have access to a suitable machine and I have plenty of other stuff on my todo list.