grblHAL/STM32F4xx #19: Some excessive definitions in mapping files and configs(?) from IDE in repository

kfmut, 2021-06-19:
Hi,

why is `STEPPERS_DISABLE_MASK` definition is present in many mapping files? I couldn't find any other place in source where it's used, maybe it's obsolete or something else.

Also is it possible to delete from repository the `.settings` directory? I think it's modified in background by STM32CubeIDE, so file `.settings/language.settings.xml` is always reported as modified by `git status` and interfere with `git pull` command.

terjeio, 2021-06-19:
> why is STEPPERS_DISABLE_MASK definition is present in many mapping files? I couldn't find any other place in source where it's used, maybe it's obsolete or something else.

It inherited fom the driver it the code is based on (STM32F103) where it is used, to be (re)moved. I have some changes coming, amongst them a shared file for adding this and some other masks even if not used by a particular driver.

> Also is it possible to delete from repository the .settings directory? I think it's modified in background by STM32CubeIDE, so file .settings/language.settings.xml is always reported as modified by git status and interfere with git pull command.

I'll add it to the .gitignore in the next commit.

kfmut, 2021-06-19:
@terjeio Thank you!

closed 2021-06-19 by kfmut