grblHAL/STM32F4xx #20: strange auto squaring behaviour

langwadt, 2021-07-04:
I'm trying to get auto squaring to work, so far just with motors on the table so I don't crash my machine. 
I get a strange behavior; if I try to home Y it looks like it is working  as expected stopping one motor at a time as the limits gets triggered, but only if I trigger the Y2 limit first. If I trigger the Y limit first it instantly fails with and error 15

terjeio, 2021-07-05:
I have now retested with my simulator and it does not fail for me.

I guess you are triggering the switches by hand, IMO that is next to impossible do correctly. Could that be the reason?
If so dual axis settings ($347 - $349) are quite strict, you may have to relax these to get testing by hand to work reliably:

```
$347: Dual axis length fail in percent, range: 0 - 100
$348: Dual axis length fail min in mm
$349: Dual axis length fail max in mm
ok
$347=5.0
ok
$348=2.500
ok
$349=25.000
ok
```

langwadt, 2021-07-05:
it's not easy but doable by hand. It seems it works very differently depending on which limit triggers first.  I'd expect it to be the same.

It is possible to successfully complete homing by hand if Y2 is triggered first, if Y is triggered first it is an instant fail

terjeio, 2021-07-05:
Well, it works as it should when I am testing with the simulator. Do you have any specific tests you want me to perform? E.g. I do not simulate bouncy switches - perhaps that could be an issue?

Most of the processing takes place in the [core](https://github.com/grblHAL/core/blob/b685e18411e5837fc05b5bd9c4d6ed8cc4fe6843/limits.c#L228-L459) which is know to work with other drivers, only the switches are read in the [driver](https://github.com/grblHAL/STM32F4xx/blob/7779affc3264f1ef3bb2f7007bba9743fb6739ca/Src/driver.c#L764-L840).

BTW I found a bug in reading the primary switches if the board map selected is BOARD_CNC3040, I'll fix that soon.

langwadt, 2021-07-05:
I found it, and it was already fixed in a commit you did 6 days ago. 

step_outbits_2.mask was set after step_outbits_1.mask  resulting in one of the limit switches disabling both motors and not just one motor like the other limit switch

closed 2021-07-05 by langwadt