grblHAL/STM32F4xx #21: Help!!If a G code has an error, the correct G code sent later will reply with the same error

hankecnc, 2021-07-08:
hi, this problem has troubled me for a long time, it prevents me from using grblhal,
This problem will not occur in 6-AXIS-USBCNC-GRBL, while GRBLhal and GRBL-Advanced will do so.
I guess this is a safety mechanism? Because wrong G code can cause serious accidents.
I hope grblhal can be like 6-AXIS-USBCNC-GRBL, for the subsequent normal G code to be executed,
I looked in config.h, there is no option to turn off this behavior, but I seem to have seen it somewhere, but I can't remember it, please help me

hankecnc, 2021-07-08:
[image]
[image]

hankecnc, 2021-07-08:
Picture 1 is 6-AXIS-USBCNC-GRBL, which can be executed normally for the subsequent G code, instead of returning the error encountered by the last wrong g code

hankecnc, 2021-07-08:
G40 G17 G49 G21 G94 G54 G80
G28 G91 Z0.
G90 G54 G0 X1.303 Y2.632 M03 S5000
G43 Z10. H01 M08
Z0.355
G3 X-2. Y0. Z-0.5 I-0.603 J-2.632 F300.
G1 Y-2.771 F1000.
X2.
Y2.771
X-2.
Y0.
X-5.
Y-5.771
X5.
Y5.771
X-5.
Y0.

hankecnc, 2021-07-08:
The error code is in the second line G28 G91 Z0.
And the fourth row G43 Z10. H01 M08

hankecnc, 2021-07-08:
How should I change it to be the same as 6-AXIS-USBCNC-GRBL? Perform subsequent normal G codes?

hankecnc, 2021-07-08:
Seems to be for this reason?
[image]

terjeio, 2021-07-08:
This behaviour is [per design](https://github.com/grblHAL/core/wiki/Changes-from-grbl-1.1) and is a safety feature, the original behaviour is IMO insane where the controller just continues execution of whatever is fed to it after an error.
It can be turned off by setting [compatibilty level](https://github.com/grblHAL/core/wiki/Compatibility-level) > 0.

Note that a G28 command should not be sent to the controller unless the machine is homed and the G28 position is set. From the LinuxCNC specification (which I use as a reference for most of the grblHAL behaviour):

>__Warning__
Only use G28 when your machine is homed to a repeatable position and the desired G28 position has been stored with
G28.1.

closed 2021-07-08 by hankecnc