grblHAL/STM32F4xx #23: PnP Project

qiwalker, 2021-08-03:
hi,

im on going develop PnP machine, and for the core will use the black pill
anyway thank you for creating GRBLHAL its great!

back to the machine , it using 4 head for pick and place and each vacuum has vacuum sensor, is it possible to get more 4 output and 4 input using GRBL HAL ?

thanks

terjeio, 2021-08-03:
The driver currently supports up to 6 inputs and 3 outputs via GPIO and can easily be extended. You will have to create a map file for your board, however I suspect the Blackpill does not have enough GPIOs and you will have to add a GPIO extender and support code for that? A better choice would be a Nucleo-64 board?

qiwalker, 2021-08-04:
ouh i see, i will try to use due core board

but still need your help to change the EEPROM port in the code

meanwhile i will design the expansion PCB.

terjeio, 2021-09-01:
> dow you any plan to implemented 5 or 6th motion planner ?

Not in the near future, but have been looking into how to separating planner and stepper code from the rest of the core in a way that would allow it to run on a second core. I believe this would simplify replacing it with higher order motion planners later on.

With luck maybe somebody else will come up with the needed code?

qiwalker, 2021-09-02:
i hope so, did u have take a look for g2core the motion is good but little bit slow than grbl

qiwalker, 2021-09-03:
hi, can i get your contact ?
i had project need your help as freelance any discussion let's continue by email.

terjeio, 2021-09-03:
Please add a comment with your email adress then and edit it out afterwards.

qiwalker, 2021-10-31:
hi terje
any update ?

thanks