grblHAL/STM32F4xx #32: Saving GRBL settings for BTT-SKR2

natrix45772, 2021-09-12:
I have succsessfully compiled grblHAL for the SKR2 Board. But after saving parameters the bootloader is overwritten. This is because the board has a 32k bootloader and uses the address 0x8004000 which grblHAL uses for saving the settings. I've changed the settings address to 0x8Exxx but this doesn't work neither! (?). Any ideas to solve this problem?

terjeio, 2021-09-12:
Settings should be moved to sector 2 at 0x08008000 instead then? The [.ld file](https://github.com/grblHAL/STM32F4xx/blob/master/STM32F407VGTX_FLASH.ld) has to be modified for that. It would be nice if you could give it a try as I do not have a board to test with.

natrix45772, 2021-09-13:
sector 2 is the program start addressed by bootloader. I took the last sector 11 for the EEPROM_EMUAL (this sector is 128k, but we have 1M in total). Now it seems to work!!! I have to make some tests to be really shure... 

My changes were:
1. **STM32F407VGTX_FLASH.ld**

_/* Memories definition */
MEMORY
{
  RAM    (xrw)    : ORIGIN = 0x20000000,   LENGTH = 128K
  BOOT_FLASH    (rx)    : ORIGIN = 0x8000000,   LENGTH = 32K
  FLASH    (rx)    : ORIGIN = 0x8008000,   LENGTH = 1024K - 160K
  EEPROM_EMUL(xrw)      : ORIGIN = 0x80E0000,   LENGTH = 128K
}_

2.  **flash.c**

_#define FLASH_SECTOR 11
#define FLASH_SECTOR_ADDR 0x80e0000_

I will leave a confirmation the next days. Thank you for your fast reply

Ray

fitch22, 2021-09-13:
This was an issue on my list as well.  I did not know if the loader script could accept preprocessor directives or not.  Is there a way to make the start address conditional so that just the skr-2 starts beyond the boot loader?

terjeio, 2021-09-15:
Here is a new loader script, _STM32F407VGTX_BL32K_FLASH.ld_, and modified _flash.c_ that uses symbols from the loader script for programming.
A updated _.cproject_ that adds "Release F407 8MHz 32K Bootloader" to the build menu in Eclipse is also included. Hopefully this should work. Please verify and report back.

[STM32F407VGTX_BL32K_FLASH.zip](https://github.com/grblHAL/STM32F4xx/files/7167129/STM32F407VGTX_BL32K_FLASH.zip)

Note that it is possible to add an external EEPROM (or FRAM) to this board as a I2C bus is brought out.

BTW is there any information, or even source code, available for the bootloader somewhere?

fitch22, 2021-09-15:
I don’t think the original source is open.  I was able to find copies of the original here, and it looks like there might be source for a functional equivalent.  However I can only vouch for the binary.  I had to use it to restore my board after blowing away the bootloader.

https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader

Bill

fitch22, 2021-09-16:
Just a quick update.  Turns out, all of my previous builds (debug and release) had been using STM32F401CCUX_FLASH.ld which was an oversight.  I have tried STM32F407VGTX_FLASH.ld and STM32F407VGTX_BL32K_FLASH.ld, and neither work for me at all.  I am getting hard traps for some reason.  The first trap occurs in HAL_RCC_OscConfig(), but not if I single step through that function.  I do not understand why the memory addresses would have any affect on setting up the clocks.  I am starting to wonder if my MCU has some bad flash that is used with the full memory linker script but not when using the linker script from the F401.  I will update when I have more information.  For now, I cannot verify the code from above, sorry.

Bill

fitch22, 2021-09-16:
Well, I don't think I know how to debug the program properly with FLASH starting address of 0x8008000.

If I use a starting address of 0x8000000 then everything is fine.  If I then change the address in the .ld file, then it is also fine.  But, if I do a complete chip erase and then load the program in the debugger, it dies with a hard fault in HAL_RCC_OscConfig() as described above.  There appears to be something left over at address 0x800000 that lets things work and when I erase the flash, whatever is needed is no longer there.

I found some options in the run/debug settings of STM32CubeIDE to change the vector table address.  I tried setting that to 0x8008000 but no joy.

Anyone know what I am doing wrong?

Thanks,
Bill

fitch22, 2021-09-17:
OK, figured out how to get the debugger to work:-).  Need to modify VECT_TAB_OFFSET from 0x00 to 0x08008000 in system_stm32f4xx.c.

Now, regarding new flash.c.  In the debugger, both _EEPROM_Emul_Start and _EEPROM_Emul_Sector don't seem to be correct values inside flash.c.  I think _EEPROM_Emul_Sector should be 11 as the last sector in the flash.  This comes in as 0xff which I think is erased flash.  Often _EEPROM_Emul_Start comes in as 0xffffffff which also looks like erased flash, but the last time I got 0x1313.

I modified the original flash.c as per Ray above, and that seems to work just fine.

I see both the _EEPROM_* definitions in the .map file, but they don't look to me like variables, more like a definition (they don't appear to have storage).  I have no idea why the extern qualifier in flash.c does not get a complaint when there is no such extern variable.  Also, I don't understand the use of & with these items.

Bill

fitch22, 2021-09-17:
At long last, I think I got to the bottom.

1) We need to change the VECT_TAB_OFFSET in system_stm32f4xx.c to match the starting address of FLASH in the loader script.
2) We need to change line 31 of flash.c to "extern uint8_t _EEPROM_Emul_Start;".  Having it declared as a void pointer messed things up.  It is still just an address and the & works.

With the above changes, I can erase entire flash, reprogram the bootloader, copy the new .bin file to SD card and insert it into the slot on my SKR-2.  The bootloader programs the flash from the .bin file and renames it to firmware.cur.  All seems good to me.

Bill

terjeio, 2021-09-17:
>  Often _EEPROM_Emul_Start comes in as 0xffffffff which also looks like erased flash, but the last time I got 0x1313.

0xFF... is correct when you have erased the flash before programming. 0x13 is the settings structure version number which is also correct after the settings has been flashed.

> I think _EEPROM_Emul_Sector should be 11 as the last sector in the flash.

This is an error from my side - the file I uploaded has _EEPROM_Emul_Sector set to 1, it should be 11. Sorry about that.

> This comes in as 0xff which I think is erased flash. 

_EEPROM_Emul_Sector is used as a constant, not as a pointer into the flash.

> We need to change the VECT_TAB_OFFSET in system_stm32f4xx.c to match the starting address of FLASH in the loader script.

Shouln't the bootloader set the vector address register before entering the code? If it does not then VECT_TAB_OFFSET should be defined in the build configuration - not by changing the source code as this messes up all the other build configurations.

BTW the link to the bootloader code you provided is for the LPC176x MCU - perhaps parts of it is the same for STM32F4xx but certainly not all. So I did not understand this statement:

> I had to use it to restore my board after blowing away the bootloader.

You used a LPC176x binary to restore the STM407 board? Or is a STM32 binary provided that I have overlooked?

> We need to change line 31 of flash.c to "extern uint8_t _EEPROM_Emul_Start;". Having it declared as a void pointer messed things up. It is still just an address and the & works.

How this is declared should not matter, void * works for me and seems more logical. _EEPROM_Emul_Start is not an 8 bit value...

fitch22, 2021-09-17:
Terje,
Well, I think I finally understand why I saw FF's when hovering over the _EEPROM* variables.  They are entries in the symbol table, but have no storage.  When I hovered over them in the debugger, it would show the value of what they pointed to, and indeed it pointed into blank flash, so all FF's is correct.

I verified that (void *) type for _EEPROM_Emul_Start works.

The bootloader will set the vector table location properly when it loads the firmware, so for a Release build, the VECT_TAB_OFFSET does not need to change.  However, for debugging it does need to change, at least for the SKR-2.  What I did just now is set another symbol for my debug build.  The system_stm32f4xx.c file checks if it is defined and used the already defined value if it is.  This works, but I wish there was a better way to have this set automatically when DEBUG is defined for the SKR-2.  If you have a better idea, please let me know.

I guess I am ready to bless the proposed changes with only setting the flash sector to 11.  Everything else is OK.  Good job!

The link I sent you is a collection of bootloaders for many of the BTT boards.  If you look in a folder called bootloader_bin you will find directories for the boards and there are copies of the original bootloaders as programmed at the factory.  I used a factory copy to restore my board.

Thanks,
Bill

terjeio, 2021-09-17:
> If you have a better idea, please let me know.

Is debugging when flashing a bootloader build really needed? I do not think I will provide a build configuration for that.
IMO it would be better to debug with a "normal" debug build and reflash the bootloader when done and then use the release version?
Modifying or adding a build configuration and adding the VECT_TAB_OFFSET symbol to that is AFAIK the cleanest way to do it if you really want to debug a bootloader version.

> I guess I am ready to bless the proposed changes with only setting the flash sector to 11.

Thanks for verifying - I'll commit an update soon.

>  If you look in a folder called bootloader_bin you will find directories for the boards ...

Thanks, I did not browse that far when I saw the source was for LPC176x. I think I will add a link to the driver page so it will be easy to find.

fitch22, 2021-09-17:
Terje,
I think the SKR-2 is a bit special.  It has a bootloader from the factory, and that is used by Marlin and RepRap firmwares and probably others.  So it clearly makes sense to have a linker script that puts the firmware in the right place to work with the same bootloader.  For a Release build, everything is good.  The problem is if someone wants to build a debug version, everything else is already configured merely by setting a DEBUG symbol.  Unfortunately with the SKR-2 the build won't work because the vector table is not in the standard place.

There is nothing wrong with having the debug build put everything in the standard place and then the vector table is also in the right place, but then we need a different linker script.

If we want to build both versions with the same linker script, then the vector table offset needs to change somehow.

One other way around this is to just document how to debug this particular build by setting the symbol VECT_TAB_OFFSET.

My preference would be to make this as painless as possible for the programmer if it is easy to do.  But documenting it is also a solution.  No matter what, I think something has to be done.  Not everyone will waste hours/days chasing down the vector table offset, but I did.

I have been thinking that the .map files may be good places to stick a bunch of documentation specific to a board.  Maybe a couple sentences about debugging could go in the .map file.

Any other ideas come to mind?

Bill

terjeio, 2021-09-18:
I'll add information about this in the readme changelog and possibly in the [wiki](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL) as well.

Since this driver can be used for many F4xx processors in different configurations I am considering adding a build configuration vs. board table as well.

fitch22, 2021-09-18:
Terje,

Sounds great.  I think the issue can be closed as well.

 

Bill

fitch22, 2021-09-20:
Terje,

I think the bootloader for the SKR Pro model is also 32K, but I don’t have one and cannot confirm.  If so, it might want to use the new loader script as well.

 

Bill

closed 2023-08-26 by terjeio