grblHAL/STM32F4xx #35: I don't understand why you have been devoting your energy to making grblhal drivers for different processors?

hankecnc, 2021-09-29:
How are you?

Your energy is limited. Does this make you tired?

I think that too many grblhal drivers will not help the promotion and use of grblhal. Most CNC enthusiasts do not understand electronic technology, and they will not spend a lot of money to buy a development board to play grbl.

What was the initial success of grbl? I don’t think it relied on the arduino development board, but the open source, low cost, and the popularity of 3D printing. grbl is excellent in 3D printing and laser performance, but it is very insufficient in CNC.

hankecnc, 2021-09-29:
In the past few days, I tried to make F401CCU6 (used chip 8 RMB) for grblhal, but due to the QFN package, there was a problem with the soldering. I will try again next month. The esp32 (10 RMB) version is already available for soldering and sale, but I Kind of hate esp32, I think it is unreliable and weird thing

Do you know when the chip will return to the normal price? If the stm32f103 can return to the normal price, I will still choose the f103 chip to make the circuit board for grblhal,

After I saw your support for the RP2040 chip, I went to find out, it is very good compared to the f103, the price is 7 yuan, but the Taobao seller told me that you can only buy one, not in bulk.

Regarding the shortcomings of grblhal? I don't know, because I can't buy cheap chips to make the circuit board, I didn't evaluate it, I just ran it on the development board, and the added drilling cycle is very good.

For professional CNC, GRBL does not depend on tool length and tool radius compensation?

Do you know the five-axis RTCP? Chinese CNC DIY enthusiasts seem to be very keen to discuss this, there is an open source 5-axis rtcp for sale on Taobao, 1000 yuan
[UsbMotionController18.24(32).zip](https://github.com/grblHAL/STM32F4xx/files/7253190/UsbMotionController18.24.32.zip)

hankecnc, 2021-09-29:
For the ioSender you wrote, I can’t Chinese. I tried to open your source code with VS. It reported a lot of errors. It was scary. I couldn’t solve it at all. I just want to change the text on the button to Chinese.

hankecnc, 2021-09-29:
gcode expression? Will anyone learn and use that ancient thing?

I think that thing is a substitute for the previous unpopular CAM software?
Among the DIY CNC enthusiasts I have contacted, most of their professions are related to the machinery industry. It is not difficult to learn to use CAM software, or wood carving enthusiasts use Autodesk ArtCAM to make G-code toolpaths.

Well, no matter what, it is still good,

closed 2021-09-30 by hankecnc

terjeio, 2021-09-30:
> Do you know when the chip will return to the normal price? If the stm32f103 can return to the normal price, I will still choose the f103 chip to make the circuit board for grblhal,

After the world get rid of Covid? I would not base a new design around that chip as it has limited memory, no FPU and is relatively slow.

> For professional CNC, GRBL does not depend on tool length and tool radius compensation?

Tool length offset is already supported (for all axes) and tool table support is optional, currently for up to 16 tools. Tool radius compensation (and path blending) is on my todo list - but I suspect both will require rather large changes to the core.

> Do you know the five-axis RTCP?

No, so I checked around a bit. It requires a expensive machine and CAD/CAM tools so is perhaps not something grblHAL should aim to support? It will be rather expensive to implement both in terms of money and time?

> For the ioSender you wrote, I can’t Chinese. I tried to open your source code with VS. It reported a lot of errors. It was scary. I couldn’t solve it at all. I just want to change the text on the button to Chinese.

You need to [install the libraries](https://github.com/terjeio/ioSender/wiki/Building) ioSender is dependent on to compile. However, making ioSender translatable is on my todo list so I will advice against it if your idea was to change the text strings in code. If you are willing to do the translation I can put this high on my list as I am going to take a break from grblHAL coding.

> gcode expression? Will anyone learn and use that ancient thing?

Perhaps not, it is a part of making grblHAL standards compliant. Parameter support is IMO important as it provides an API to get at core data in a read-only fashion, useful for plugin developers.

terjeio, 2021-09-30:
Like this?

[image]

hankecnc, 2021-10-05:
> 像这样？
> 
> [image]

Yes

hankecnc, 2021-10-05:
Is it a misunderstanding caused by different languages?

What I mean is that I directly modify the text of the control by opening the source code to Chineseize the software.

If you have a better software translation method, I can try the method you provide.

terjeio, 2021-10-05:
> Is it a misunderstanding caused by different languages?

Yes, it could easily be a misunterstanging.

> What I mean is that I directly modify the text of the control by opening the source code to Chineseize the software.

Not a good idea, when a new version comes out the translation has to be done again. 

> If you have a better software translation method, I can try the method you provide.

I have, there are a number of options and I have decided to to use [baml for localization](https://docs.microsoft.com/en-us/dotnet/desktop/wpf/advanced/wpf-globalization-and-localization-overview?view=netframeworkdesktop-4.8).

I have already tagged most of the UI with the required Uid and generated the file that is to be used for translation. With the _LocBaml_ tool a translation can then be added _without_ recompiling.

What I find a bit problematic is that there are no good tools (that I can find) for maintaining translations. In addition the file generated contains a lot of information that, in my opinion, can easily be corrupted by a translator.
To mitigate this I have started to experiment with loading the file generated into a relational database and adding fields to that so that translation status can be tracked. From this I can generate a file(s) for translation that will be a lot simpler for the translator to handle. However, there is a downside to this - a translation has to be submitted so that I can apply it...

Give me a bit more time to experiment, I need that before I decide how to proceed.

hankecnc, 2021-10-05:
> > 是不是语言不同造成的误会？
> 
> 是的，这很容易造成误解。
> 
> > 我的意思是我直接通过打开源代码修改控件的文字来汉化软件。
> 
> 这不是一个好主意，当一个新版本出现时，必须再次进行翻译。
> 
> > 如果你有更好的软件翻译方法，我可以试试你提供的方法。
> 
> 我有，有很多选择，我决定使用[baml 进行本地化](https://docs.microsoft.com/en-us/dotnet/desktop/wpf/advanced/wpf-globalization-and-localization-overview?view=netframeworkdesktop-4.8)。
> 
> 我已经用所需的 Uid 标记了大部分 UI，并生成了用于翻译的文件。使用_LocBaml_工具，_无需_重新编译即可添加翻译。
> 
> 我觉得有点问题的是没有好的工具（我可以找到）来维护翻译。此外，生成的文件包含大量信息，在我看来，这些信息很容易被翻译人员破坏。 为了缓解这种情况，我开始尝试将生成的文件加载到关系数据库中并添加字段，以便可以跟踪翻译状态。由此，我可以生成一个翻译文件，这对于翻译人员来说处理起来要简单得多。但是，这有一个缺点 - 必须提交翻译才能应用它......
> 
> 给我多一点时间进行实验，在我决定如何继续之前，我需要它。

OK 👌。
Hi, I just used GRBLhal to connect my Android phone via USB and sent him "G1Z10F100". The machine moved. I think this is a very interesting development direction because people have many old Android phones or tablets.

MuhammadQaisarAli, 2023-02-21:
Heo @terjeio .
Can you please contact me on muhammadqaisarali1@gmail.com, I would be thankful.