grblHAL/STM32F4xx #43: Board config - STM32F429VGT6

boboxx, 2021-11-16:
I see there is a SKR2 that has a STM32F429VGT6 (https://www.mouser.ca/datasheet/2/389/dm00071990-1797623.pdf) has anyone done any board config for this board? This MCU seems to be a bit faster running at 180MHz then the STM32F407VGT6 running at 168MHz.

The board is here:
https://www.amazon.ca/gp/product/B0943WVM5B/

terjeio, 2021-11-16:
> has anyone done any board config for this board?

Not that I know about, if like the other boards it is adding setup of the clocks in main.c and adding a build configurations for it. To do that one has to get hold of the crystal frequency.

If the pins are routed differently that the F407 SKR2 a pin map has to be added as well.

boboxx, 2021-11-16:
I have the board on it's way, how would I "get hold of the crystal frequency"?

terjeio, 2021-11-16:
Either from the schematic or from the crystal itself, the little black rectangle just below the processor in the image from the link above?

boboxx, 2021-11-17:
Oh ok I was thinking internal crystal. That all looks was to complicated for my knowledge! I have ordered a second SKR2 with a STM32F407.

boboxx, 2021-11-18:
I received the board, the crystal is a DO81a but I'm unable to located a datasheet for it.

acaballero6270, 2022-01-31:
I compiled the firmware for SKR2. I added the symbols HSE_VALUE 8000000, STM32F407xx and USE_HAL_DRIVER. Removed all others. In STM32CubeIDE. Also commented out line 37 on btt_skr_2.0_map.h. As per Terje's advise. It mostly works except for Trinamic drivers.

dwkdnvr, 2022-04-04:
Following up on the 429 version of the SKR2. I have the symptoms described above by @acaballero6270 - firmware flashed successfully and I can connect, but steppers (2209) don't turn. Built with STM32IDE, and I think my flags were
USE_HAL_DRIVER
HSE_VALUE=8000000
HAS_BOOTLOADER
STM32F407
TRINAMIC_UART_ENABLE
N_AXES=5
TRINAMIC_ENABLE=2209
BOARD_BTT_SKR_20

From reading other issues here it seems @acaballero6270 ultimately got this working, so I'm hoping I'm missing something obvious (since the I'm new to the building firmware game)

acaballero6270, 2022-04-04:
All I used were:
HSE_VALUE 8000000
STM32F407xx
USE_HAL_DRIVER
TRINAMIC_ENABLE 2209

Also #define BOARD_BTT_SKR_20  in my_machine.h

dwkdnvr, 2022-04-08:
Ok, I have movement. Not sure whether I was just dumb and impatient and didn't realize the jog settings were low. I re-flashed with a platformIO build and tweaked the settings and it now moves.
Unfortunately I seem to have the 'controller disconnects after a short period' problem that someone else had. I'm hoping it's my janky power supply, but either way more investigation

gjcollie, 2022-04-16:
Can someone post a compiled firmware for the SKR 2 with STM32F429.  I am able to compile but when I install the firmware via SD the I no longer get a USB connection.  I am not sure whether it is something in my configuration or that the firmware is not functioning correctly.

acaballero6270, 2022-04-17:
Here is one:
[grblHAL F407 8MHz.zip](https://github.com/grblHAL/STM32F4xx/files/8500821/grblHAL.F407.8MHz.zip)
You would have to use STMLink

gjcollie, 2022-04-17:
> Here is one: [grblHAL F407 8MHz.zip](https://github.com/grblHAL/STM32F4xx/files/8500821/grblHAL.F407.8MHz.zip) You would have to use STMLink

Thanks for the file.  Is there nothing that will work with the built in bootloader?  STM32CubeProgrammer is throwing an error when I try to connect "Error" Activatingdevice: KO. Please verify the boot mode configuration and check the serial port configuration. Reset your device then try again..."  

I know there is no issue with the serial port config because its working fine to connect to the installed Marlin firmware.  Is there some trick to place the device into boot mode, I cannot find any documentation on this.

Thanks

RileyMcCarthy, 2022-12-24:
Has anyone got the F429 version of the SKR 2 working? I am trying to use grblHAL for my laser engraver but I got the F429 variant.

terjeio, 2023-01-27:
Can you try the grblHAL STM32F429 BTT SKR 2.zip binary available [here](http://www.io-engineering.com/downloads)?
You may also try the F407 version as the clocks setup is identical.

acaballero6270, 2023-01-27:
I tried it. I can’t get a connection though USB.

terjeio, 2023-01-28:
Did you program the board with a programmer or via bootloader (SD card)?
And have you tried the F407 version?

closed 2023-08-26 by terjeio