grblHAL/STM32F4xx #44: SKR2 - compile error

boboxx, 2021-11-17:
Just did a fresh pull from git:
git clone --recurse-submodules https://github.com/grblHAL/STM32F4xx.git

Created a new board config for platformio from a copy of the other SKR2 but removed the trinamic as I will be using some stepper driver adapter module. But i'm getting a error:

error: 'cs' undeclared (first use in this function)

Bellow is my config and logs:

[env:btt_skr_2]
# Untested and might not boot.  Please report issues at:
# https://github.com/grblHAL/STM32F4xx/issues
board = genericSTM32F407VGT6
board_build.ldscript = STM32F407VGTX_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_BTT_SKR_20=
  -D HSE_VALUE=8000000
  #-D USB_SERIAL_CDC=1
  # Boot loader offset (32K)
  -D VECT_TAB_OFFSET=0x8000
lib_deps = ${common.lib_deps}
  eeprom
lib_extra_dirs = ${common.lib_extra_dirs}
# Upload is not supported for this board since BOOT0 is tied to GND.
# With the default boot loader, you must deploy new firmware by copying
# .pio/build/<env name>/firmware.bin (produced by `pio run`) to the SD card.

Enabled my board in my_machine.h by removing the "//" using  #define BOARD_BTT_SKR_20 

C:\Users\user\Documents\Projects\STM32F4xx>pio run --environment btt_skr_2
Processing btt_skr_2 (board: genericSTM32F407VGT6; platform: ststm32; framework: stm32cube)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VGT6.html
PLATFORM: ST STM32 (15.0.0) > STM32F407VG (128k RAM. 1024k Flash)
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES:
 - framework-stm32cubef4 1.26.2
 - tool-ldscripts-ststm32 0.1.0
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 77 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <bluetooth>
|-- <grbl>
|-- <keypad>
|-- <laser>
|-- <motors>
|-- <odometer>
|-- <spindle>
|-- <Core>
|-- <Class>
|-- <App>
|-- <Target>
|-- <eeprom>
Building in release mode
Compiling .pio\build\btt_skr_2\libbfb\grbl\maslow.o
Compiling .pio\build\btt_skr_2\libbea\keypad\keypad.o
Compiling .pio\build\btt_skr_2\lib17a\laser\coolant.o
Compiling .pio\build\btt_skr_2\lib17a\laser\ppi.o
Compiling .pio\build\btt_skr_2\lib02a\motors\trinamic.o
Compiling .pio\build\btt_skr_2\lib392\odometer\odometer.o
Compiling .pio\build\btt_skr_2\lib2c9\spindle\huanyang.o
Compiling .pio\build\btt_skr_2\lib2c9\spindle\modbus.o
Compiling .pio\build\btt_skr_2\lib535\App\usbd_cdc_if.o
Compiling .pio\build\btt_skr_2\libf00\eeprom\eeprom_24AAxxx.o
Compiling .pio\build\btt_skr_2\libf00\eeprom\eeprom_24LC16B.o
Compiling .pio\build\btt_skr_2\src\btt_skr_2.0.o
Compiling .pio\build\btt_skr_2\src\diskio.o
Src\btt_skr_2.0.c: In function 'add_cs_pin':
Src\btt_skr_2.0.c:511:7: error: 'cs' undeclared (first use in this function)
       cs[X_AXIS].port = (GPIO_TypeDef *)gpio->port;
       ^~
Src\btt_skr_2.0.c:511:7: note: each undeclared identifier is reported only once for each function it appears in
At top level:
Src\btt_skr_2.0.c:541:13: warning: 'if_init' defined but not used [-Wunused-function]
 static void if_init(uint8_t motors, axes_signals_t enabled)
             ^~~~~~~
Compiling .pio\build\btt_skr_2\src\driver.o
Compiling .pio\build\btt_skr_2\src\flash.o
*** [.pio\build\btt_skr_2\src\btt_skr_2.0.o] Error 1

terjeio, 2021-11-17:
Change [line 503](https://github.com/grblHAL/STM32F4xx/blob/77760893fc7d1e008e989d36e6b1a77536f7a5a8/Src/btt_skr_2.0.c#L503) to
`#if TRINAMIC_ENABLE`
and recompile.

I'll commit an update a little later.

boboxx, 2021-11-17:
Thank you that solved the problem!

hankecnc, 2021-11-18:
> 将[第 503 行](https://github.com/grblHAL/STM32F4xx/blob/77760893fc7d1e008e989d36e6b1a77536f7a5a8/Src/btt_skr_2.0.c#L503)更改为 `#if TRINAMIC_ENABLE` 并重新编译。
> 
> 稍后我会提交更新。

> 将[第 503 行](https://github.com/grblHAL/STM32F4xx/blob/77760893fc7d1e008e989d36e6b1a77536f7a5a8/Src/btt_skr_2.0.c#L503)更改为 `#if TRINAMIC_ENABLE` 并重新编译。
> 
> 稍后我会提交更新。

hi, Terje Io, do you have a plan to make stm32h7 driver? I have made grbl circuit board of esp32 for sale, aimed at beginner CNC enthusiasts (they use aluminum plate or aluminum profile to make small desktop cnc), I want to make one more This is a circuit board for professional CNC enthusiasts (they use castings). I think STM32H743VIT6 is very good. This chip is very popular and easy to buy. But I can't program. So do you have a plan to make stm32h7? I will wait

closed 2023-08-26 by terjeio