grblHAL/STM32F4xx #46: Dual axis for X and Y and auto squaring on SKR PRO 1.2

olivierOT, 2021-11-25:
Hi,

I have an MPCN running on SKR pro 1.2, motors runs onto 5 TMC2209 in UART. CNC is using a Neje PWM diode laser mounted on it

My machine has dual axis on X and Y, using respectively E0 and E1, and has 2 endstop on X and 2 on Y for auto squaring homing.

I'm currently using Marlin 2 but i would like to migrate to grblHAL.
The thing is that i'm completly lost and i have no clue of where to get start to make corresponding setup in configuration files before to go to compile.

Any a good soul here to help me on it ?

terjeio, 2021-11-26:
Configuration of the driver is in [my_machine.h](https://github.com/grblHAL/STM32F4xx/blob/master/Inc/my_machine.h), SKR1.2 can use the same map file as SKR 1.1?

Currently Trinamic drivers are not supported for this board but code is available for soft UART in SKR 2.0 board specific code. If you wait a bit I will add it to SKR 1.1 with the changes required. Be aware that I am not able to test this so I cannot guarantee that it will work.

olivierOT, 2021-11-26:
Hi, thanks for your reply.
Theorically yes SKR pro 1.2 has the same specs and pinout than 1.1.

I had a look on my_machine.h but i didn't found anything about dual axis X->X/ reversed E0 and Y->Y/reversed E1.
 
I can wait for sure ! I will also be pleased to test it and to give you feedbacks.

terjeio, 2021-11-26:
> I had a look on my_machine.h but i didn't found anything about dual axis X->X/ reversed E0 and Y->Y/reversed E1.

See [here](https://github.com/grblHAL/STM32F4xx/blob/b60d8b86b392eed8ccf642ec37ad21800ffab3ce/Inc/my_machine.h#L71-L77).

$8 was recently added for reversing the second motors by setting instead of reversing by swapping motor wires.

> I will also be pleased to test it and to give you feedbacks.

Sounds good.

terjeio, 2021-11-28:
New version out now. Start with no ganged motors if you want to test.

terjeio, 2021-11-29:
It compiles for me with STM32CubeIDE 1.5.1 so these errors are from the pio framework. I do not have that installed so have no idea where to start other than install it and learn how to configure it. @dresco Do you have any idea about what could be the reason?

olijouve, 2021-11-29:
I'm using my personnal github account.

> It compiles for me with STM32CubeIDE 1.5.1

Thanks i've build it successfully using STM32CubeIDE, I'll try to find time to flash the board this evening.

dresco, 2021-11-29:
> @dresco Do you have any idea about what could be the reason?

Is building okay here, I just took a fresh clone & added the following test environment to platformio.ini (just adding the 2209 support and auto-squaring to the original skr_pro_1_1 entry)..

platformio run --environment btt_skr_pro_1_1_buildtest

```
[env:btt_skr_pro_1_1_buildtest]
# Untested and might not boot.  Please report issues at:
# https://github.com/grblHAL/STM32F4xx/issues
board = genericSTM32F407VGT6
board_build.ldscript = STM32F407VGTX_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_BTT_SKR_PRO_1_1=
  # 8MHz crystal
  -D HSE_VALUE=25000000
  # Boot loader offset (32K)
  -D VECT_TAB_OFFSET=0x8000
  # TMC2209 stepper drivers
  -D TRINAMIC_ENABLE=2209
  # Motor ganging & auto-squaring
  -D X_GANGED=1
  -D X_AUTO_SQUARE=1
  -D Y_GANGED=1
  -D Y_AUTO_SQUARE=1
lib_deps = ${common.lib_deps}
  eeprom
  trinamic
lib_extra_dirs = ${common.lib_extra_dirs}
# Upload is not supported for this board since BOOT0 is tied to GND.
# With the default boot loader, you must deploy new firmware by copying
# .pio/build/<env name>/firmware.bin (produced by `pio run`) to the SD card.
```

olijouve, 2021-11-29:
Hum, i cloned again STM32F4xx repository, added your btt_skr_pro_1_1_buildtest section env in platformio.ini and tryied to build it but still have errors:

```
MacBook-Pro-de-Olivier:STM32F4xx $ platformio run --environment btt_skr_pro_1_1_buildtest
Processing btt_skr_pro_1_1_buildtest (board: genericSTM32F407VGT6; platform: ststm32; framework: stm32cube)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VGT6.html
PLATFORM: ST STM32 (8.0.0) > STM32F407VG (192k RAM. 1024k Flash)
HARDWARE: STM32F407VGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES: 
 - framework-stm32cube 2.0.200813 
 - framework-stm32cubef4 1.26.2 
 - tool-ldscripts-ststm32 0.1.0 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 52 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <bluetooth>
|-- <grbl>
|-- <keypad>
|-- <laser>
|-- <motors>
|-- <odometer>
|-- <spindle>
|-- <Core>
|-- <Class>
|-- <App>
|-- <Target>
|-- <eeprom>
|-- <trinamic>
Building in release mode
Compiling .pio/build/btt_skr_pro_1_1_buildtest/lib87a/bluetooth/hc_05.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/alarms.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/coolant_control.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/corexy.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/errors.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/gcode.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/grbllib.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/ioports.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/limits.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/maslow.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/motion_control.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/my_plugin.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/ngc_expr.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/ngc_params.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/nuts_bolts.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/nvs_buffer.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/override.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/pid.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/planner.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/protocol.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/regex.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/report.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/settings.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/sleep.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/spindle_control.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/state_machine.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/stepper.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/stream.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/system.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/tool_change.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/libee4/grbl/wall_plotter.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/lib197/keypad/keypad.o
Compiling .pio/build/btt_skr_pro_1_1_buildtest/lib62c/laser/coolant.o
keypad/keypad.c:26:10: fatal error: driver.h: No such file or directory

[... 27 lines cut ...]
laser/ppi.c:24:10: fatal error: driver.h: No such file or directory

****************************************************************
* Looking for driver.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:driver.h"
* Web  > https://platformio.org/lib/search?query=header:driver.h
*
****************************************************************

 #include "driver.h"
          ^~~~~~~~~~
compilation terminated.
*** [.pio/build/btt_skr_pro_1_1_buildtest/lib62c/laser/ppi.o] Error 1
====================================================================================== [FAILED] Took 3.94 seconds ======================================================================================

Environment                Status    Duration
-------------------------  --------  ------------
btt_skr_pro_1_1_buildtest  FAILED    00:00:03.940
================================================================================= 1 failed, 0 succeeded in 00:00:03.940 =================================================================================
```

My PlatformIO Core version is 5.2.3

Anyway it builded nicely with STM32CubeIDE so i keep going with it for now

olijouve, 2021-11-29:
I've flashed my board, green diode blinked as usual on new firmware.bin

I'm stucked as I don't have any USB serial port recognized for the board on my laptop i can connect on it.

olijouve, 2021-11-30:
According to schematic, crystal frequency on that board seems to be 8Mhz
https://raw.githubusercontent.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/master/Schematic/SKR-PRO-V1.1%EF%BC%88SCH%EF%BC%89.pdf

I'll give it a new try with HSE_VALUE set to 8000000 as soon as possible

terjeio, 2021-11-30:
The bootloader version has never worked? See issue #45for more info.

Do you have access to a ST-Link programmer? Voltmeter?

boboxx, 2021-11-30:
Looks like you are missing some modules, make sure you are adding recursive to your pulls

it should be:
git clone --recurse-submodules https://github.com/grblHAL/STM32F4xx.git

olijouve, 2021-11-30:
@terjeio what do you mean by "The bootloader version has never worked?" i'll read #45...
No i don't have a ST-link, but i do have a voltmetre.

@boboxx, `git clone --recurse-submodules https://github.com/grblHAL/STM32F4xx.git` is exactly how i cloned the project.

dresco, 2021-11-30:
@olijouve It should compile from a fresh recursive clone with no code changes. Your log indicates it's failing to find driver.h from the Inc folder, is ARDUINO defined somewhere (that would change the include path in keypad.c)? 

If you want to pursue the platformio errors, could you maybe post a copy of the build output somewhere - a gist or pastebin? I'll compare that with my working env.. Cheers.

```
git clone --recurse-submodules https://github.com/grblHAL/STM32F4xx.git
cd STM32F4xx
platformio run --target clean
platformio run --environment btt_skr_pro_1_1 --verbose > build.log 2>&1
```

olijouve, 2021-11-30:
Thanks @dresco i should have some old arduino relics in my plateformio workspacd.
i've delete my ~/.platformio folder and it built successsfully.

I'll try to flash board through sdcard now...

olijouve, 2021-12-01:
Not better, bootloader seems to work as i can see green blinking led when i put .pio/build/btt_skr_pro_1_1/firmware.bin in sdcard, and the file is then renamed to FIRMWARE.CUR

terjeio, 2021-12-01:
@olijouve Which build configuration are you using the binary from? Only _Release F407 8MHz 32K Bootloader_ has the offset I believe is needed.

dresco, 2021-12-01:
> i've delete my ~/.platformio folder and it built successsfully.

Cool, good news..

> Only _Release F407 8MHz 32K Bootloader_ has the offset I believe is needed.

Looks like the bootloader offset is defined in the btt_skr_pro_1_1 platformio environment, the crystal value may still be wrong though? The description and value don't match..

olijouve, 2021-12-01:
Yes it's what i've seen.
I had tried to compile with try HSE_VALUE set to 8000000 but in  skr_pro_v1_1_map.h there is that test that does not allow the build:
```
#if !defined(STM32F407xx) || HSE_VALUE == 8000000
#error "This board has STM32F407 processor with a 25MHz crystal, select a corresponding build!"
#endif
```

I tried to comment the error line but without more chance but i suppose it a bit far more complicated than that

terjeio, 2021-12-01:
What is needed is a way to check that the downloaded code get started by the bootloader. Without much code changes that would be to measure the volatage on an input pin. This should be 3.3V due to the pullup resistor beeing enabled.
Change the RESET_PORT and RESET_PIN definitions to GPIOC and 1 and measure on the J46 (I/O expansion port)? AFAICT this pin has no pullup resistor connected externally that would mess up the measurement. 

Another would be to toggle an output pin, I see that the fan and heater outputs has LEDs connected.

olijouve, 2021-12-01:
>  Which build configuration are you using the binary from? Only Release F407 8MHz 32K Bootloader has the offset I believe is needed.
I also tried this one to without more success, usb seriel port can't be seen on my Mac.

Looking at Marlin's plateformio.ini we can say that  BigTreeTech SKR Pro v1.1 /v1.2 boards are shipped with a bootloader
that is located in the first 0x8000 bytes of the STM32F4 flash memory.

I've flashed back a Marlin 2 firware just to ensure nothing were bricked and it's fine. As soon it's been flashed, usb serial device is seen again by my OS.

>Change the RESET_PORT and RESET_PIN definitions to GPIOC and 1 and measure on the J46 (I/O expansion port)
I can't find the J46 you mention:
https://teamgloomy.github.io/skr_pro_pins.html

> Another would be to toggle an output pin, I see that the fan and heater outputs has LEDs connected.
Didn't find by myself how and where in code  i can toggle an output pin...

terjeio, 2021-12-01:
PC1 is here in the schematic:

[image]

I believe toggling an output pin can be done by adding [this plugin](https://github.com/grblHAL/Templates/tree/master/my_plugin/stm32F411blinky) to the src folder, modifying it for the port/pin you want and recompile/reflash. The LEDs I mentioned seems to need the board to be powered by 12V.

olijouve, 2021-12-01:
my board is powered by 24V could it be risky to test that ?

terjeio, 2021-12-02:
> my board is powered by 24V could it be risky to test that ?

No, if designed for 24V you will be ok.

> ... but if not loaded i should get 0V ?

You may measure some voltage since the pin is (or should be?) floating, but IMO not 3.3V.
If we can verify that the code has been started then it is likely that the issue is with the clock setup for USB. Even better would be to get a blinking LED as this will stop if the MCU goes into a hard fault.

olijouve, 2021-12-02:
> Thanks for J46 PC1, i'll try to test tomorrow. So just to understand, if the firmware is well loaded i should mesure 3.3V between that pin and gnd because the RESET_PORT and RESET_PIN definitions set to GPIOC and 1, but if not loaded i should get 0V ?

I've just tested it and still have nearly 0V after flashing the board with RESET_PORT and RESET_PIN definitions set to GPIOC and 1.

I have a FTDI232, do you think trying to connect to UART could show some usefull stuff ?

terjeio, 2021-12-02:
> I have a FTDI232, do you think trying to connect to UART could show some usefull stuff ?

Yes, try that. If nothing shows up I believe the code is not entered by the bootloader. A pity that the source for it is nowhere to be found.

Maybe taking a look at the Marlin binary (or .elf) could be the next step?

olijouve, 2021-12-02:
It's exactly the same with FTDI232 on uart, no usb serial is created on my host

olijouve, 2021-12-02:
Here is the Marlin's SKR PRO board json description, if it could help...

Marlin/uildroot/share/PlatformIO/boards/BigTree_SKR_Pro.json

```
{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32F4 -DSTM32F407xx -DSTM32F40_41xxx",
    "f_cpu": "168000000L",
    "hwids": [
      [
        "0x1EAF",
        "0x0003"
      ],
      [
        "0x0483",
        "0x3748"
      ]
    ],
    "mcu": "stm32f407zgt6",
    "variant": "BIGTREE_SKR_PRO_1v1"
  },
  "debug": {
    "jlink_device": "STM32F407ZG",
    "openocd_target": "stm32f4x",
    "svd_path": "STM32F40x.svd"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "STM32F407ZG (192k RAM. 1024k Flash)",
  "upload": {
    "disable_flushing": false,
    "maximum_ram_size": 196608,
    "maximum_size": 1048576,
    "protocol": "stlink",
    "protocols": [
      "stlink",
      "dfu",
      "jlink"
    ],
    "offset_address": "0x8008000",
    "require_upload_port": true,
    "use_1200bps_touch": false,
    "wait_for_upload_port": false
  },
  "url": "https://www.st.com/en/microcontrollers-microprocessors/stm32f407zg.html",
  "vendor": "Generic"
}
```

terjeio, 2021-12-02:
> It's exactly the same with FTDI232 on uart, no usb serial is created on my host

Is your FTDI breakout board broken? It should show up even when not connected to the controller. Mine does...

olijouve, 2021-12-02:
I don't know, i have it for years unoppened in it's antistatic bag, i've bought it on aliexpress for less than 3$ so it could be defective.
I've installed a FTDI232 driver for mac but i could'nt say if it worked.

I'll check if i don't have one in an elegoo arduino starter kit i'd bought years ago

olijouve, 2021-12-02:
It's my old mini usb cable that was faulty, i've found a new one and i can see FTDI port on my Mac, I tried to connect to it at 9600 115200 and 250000 bauds but nothing.
I've connected SKR UART TX and RX to FTDI232 RX and TX(also tried reversed), uart gnd and 5v to FTDI gnd and vcc.

olijouve, 2021-12-02:
maybe this could be usefull :
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/25#issuecomment-613133435

terjeio, 2021-12-02:
I have flashed the bootloader on a STM32F446 dev board and see signalling on the SD-card pins. This means it runs, but it has not grabbed a firmware.bin. I flashed one directly with ST-LINK Utility but it is not started. Next step is to find out why...

terjeio, 2021-12-02:
I managed to track it down. It is a sloppy bootloader that does not relocate the vector table, the LPC176x bootloader does...
Add these line at the start of [driver_init()](https://github.com/grblHAL/STM32F4xx/blob/88f1b648023a82251783e5d664bd958788082c25/Src/driver.c#L1896-L1899):

```
    __disable_irq();
    SCB->VTOR = 0x8008000 ;
    __DSB();
    __enable_irq();
```

Perhaps try with the FTDI breakout first, I am not able to test USB.

olijouve, 2021-12-02:
It compile but i'm not able to get anything on serial with FTDI, i also tried to run it at 3.3v(can do both 5V or 3.3V) with vcc attached to 3.3v on skr board.

Just to not being lost, what config do you suggest me to run with that last change ?

terjeio, 2021-12-03:
>  i also tried to run it at 3.3v(can do both 5V or 3.3V) with vcc attached to 3.3v on skr board.

I do not connect any supply to my FTDI breakout when using that, it is an output  pin.

> Just to not being lost, what config do you suggest me to run with that last change ?'

Enable the board and USB_SERIAL_CDC, also disable settings storage in flash - I had to do that for my F446 board likely because differing memory map. To disable settings [change this line](https://github.com/grblHAL/STM32F4xx/blob/88f1b648023a82251783e5d664bd958788082c25/Src/driver.c#L1989) to `#elif xFLASH_ENABLE`.

Check the voltage on PC1 if still no connection.

Wakingtester88, 2021-12-08:
Have you figured out your problem? Are you wanting grbl so you can run your laser? I am running lightburn on my ender 5 to control my 30w neje laser,  with btt skr-pro-1.2 board and marlin! I can put my print head back on and print like normal.. when i do this i run 12v to the board and leave the 24v for everything else since you can run them at different power levels and not hurt the laser, most of the time they are 12v(the laser).

olijouve, 2021-12-09:
@Wakingtester88, yes i would prefer to run grbl for laser engraving mostly, but then also for a spindle.
I don't have issue running laser for cutting on my CNC with Marlin but i'm not satisfied with the render i have with engraving.

From what i've red, grbl performs better with pwm reduction on deceleration and it results better engraving on borders.
Also sometime i have annoying recalcitrant usb connection issue with Lightburn / SKR Pro / Marlin 2, forcing me to use another software to release the serial port as a workaround. I wish it could be resolved by switching to grblHAL...

@terjeio, i didn't found spare time yet those last few days to test what you pointed out on your last comment. I'll do it asap.

terjeio, 2021-12-10:
@olijouve I have committed the changes needed for the bootloader, tested with a F411.

> From what i've red, grbl performs better with pwm reduction on deceleration and it results better engraving on borders.

FYI this driver supports PPI (Pulses Per Inch) mode, [Lightburn supports that too](https://lightburnsoftware.github.io/NewDocs/CutSettings-Line.html#override-ppi) but only for Trocen controllers. In grblHAL PPI mode is set up with M-codes, I'll try to find out how Trocen does it and see if grblHAL can support the same method.

I nearly always use PPI mode for cutting with my 40W CO2 laser, very delicate cutting can be made with it enabled:

[image]

olijouve, 2021-12-10:
> @olijouve I have committed the changes needed for the bootloader, tested with a F411.

I've freshly cloned, built and flashed but no more chance on usb serial with that config:
`[env:btt_skr_pro_1_1]
# Untested and might not boot.  Please report issues at:
# https://github.com/grblHAL/STM32F4xx/issues
board = genericSTM32F407VGT6
board_build.ldscript = STM32F407VGTX_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_BTT_SKR_PRO_1_1=
  # 8MHz crystal
  -D HSE_VALUE=25000000
  # Boot loader offset (32K)
  -D VECT_TAB_OFFSET=0x8000
  -D TRINAMIC_ENABLE=2209
  -D USB_SERIAL_CDC=1

lib_deps = ${common.lib_deps}
  eeprom
  trinamic
lib_extra_dirs = ${common.lib_extra_dirs}
# Upload is not supported for this board since BOOT0 is tied to GND.
# With the default boot loader, you must deploy new firmware by copying
# .pio/build/<env name>/firmware.bin (produced by `pio run`) to the SD card.
`

terjeio, 2021-12-11:
> board_build.ldscript = STM32F407VGTX_FLASH.ld
must be changed to
`board_build.ldscript = STM32F407VGTX_BL32K_FLASH.ld`
and
 > -D VECT_TAB_OFFSET=0x8000
to
`-D HAS_BOOTLOADER`

Note that this requires the latest version. If you still cannot make it start then I can upload builds made with STM CubeIDE.

BTW the schematic has the crystal specified as 8 MHz for both 1.1 and 1.2, does it not match the actual one?

[image]

olijouve, 2021-12-11:
Big change !! 
Now i can see green led blinking longer at flash and the usb serial is seen on my Mac:
/dev/cu.usbmodem377A396E34361

I used thoses parameters:
  -D BOARD_BTT_SKR_PRO_1_1=
  -D HSE_VALUE=25000000
  -D HAS_BOOTLOADER
  -D TRINAMIC_ENABLE=2209
  -D USB_SERIAL_CDC=1

However i can't get any working communication over that port.

Your are right for the 8MHz clock specified in schematics, so i tried with -D HSE_VALUE=8000000 by comment line 25 of Inc/btt_skr_pro_v1_1_map.h, but usb serial port is not reconnized with that value.

terjeio, 2021-12-11:
Great!
Try send a `?` to fetch a realtime report. If the controller enters some alarm modes [it does not respond to anything else](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings).
Or try connecting with your FTDI breakout - the startup message(s) are often lost with a USB connection.

> Your are right for the 8MHz clock specified in schematics, so i tried with -D HSE_VALUE=8000000 by comment line 25 of Inc/btt_skr_pro_v1_1_map.h, but usb serial port is not reconnized with that value.

The the schematic is incorrect...

olijouve, 2021-12-11:
Unfortunatly even with FTDI i don't get anything... i tried at 115200 bauds

terjeio, 2021-12-12:
Hmm, a bit odd.
Can you check the voltage as [discussed before](https://github.com/grblHAL/STM32F4xx/issues/46#issuecomment-983486890).
And try with changing
https://github.com/grblHAL/STM32F4xx/blob/fdef66850504d3c3ad75b7556340ab61d78c8fb5/Src/driver.c#L1995
to
`#elif xFLASH_ENABLE`
if still 0V as this may cause a hard fault.

fughilli, 2021-12-17:
Dunno if you're building the standalone image (no bootloader), but if you are, the board will not boot due to a bad vector table relocation. The vector table is located by the linkerscript STM32F407VGTX_FLASH.ld at 0x0, but then the startup code relocates it to 0x8000 as configured by the project config. As a result, SysTick firing during HAL init causes the CPU to jump to some invalid address when loading whatever happens to be at 0x8000 + SysTick_offset.

I have fixed it here: https://github.com/grblHAL/STM32F4xx/commit/d1365cdf35c8529bf744c14dacd455598115582b

Don't have time to open a PR right now, but just thought I'd chime in in case anyone else was having trouble with this.

olijouve, 2021-12-17:
Thanks @fughilli, I think this point was resolved in #337 few days ago as i can now use SDCard bootloader and grblHAL loads on my board.

terjeio, 2021-12-18:
@fughilli I have a fix for the map file ready to be commited, `GPIO_H` clock init is done in `MX_GPIO_Init()` so no need to add that to `main()`. FYI I have used the F401 startup code for all variants, this will be changed in the next commit since user code may need interrupt support for peripherals not available in the F401.

fughilli, 2021-12-18:
> @fughilli I have a fix for the map file ready to be commited, `GPIO_H` clock init is done in `MX_GPIO_Init()` so no need to add that to `main()`. FYI I have used the F401 startup code for all [...]

I didn't read that part of the datasheet carefully but I was under the impression that GPIO H clock init was needed prior to configuring HSE since the XIN/XOUT are on port H pins.

Just one of a number of things I tried before getting it to work and didn't test if it worked without--but as you point out, the existing code worked elsewhere :)

closed 2022-01-20 by terjeio