grblHAL/STM32F4xx #49: errors compiling STM32F4xx

acaballero6270, 2022-01-10:
Attempting to compile STM32F4xx for Bigtreeteck SKR2 I get the following errors:
Using platformio:
.pio/build/btt_skr_2_tmc5160/src/driver.o: In function `driver_init':
driver.c:(.text.driver_init+0xb6): undefined reference to `i2c_init'

Using STM32CubeIDE:
c:\st\stm32cubeide_1.8.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_2.0.0.202105311346\tools\arm-none-eabi\bin\ld.exe: ./Src/driver.o: in function `driver_init':
driver.c:(.text.driver_init+0xb6): undefined reference to `i2c_init'
I followed the instructions in here to the letter. AM I doing something wrong?

terjeio, 2022-01-10:
> AM I doing something wrong?

No, comment out or remove this line for now :

https://github.com/grblHAL/STM32F4xx/blob/4e9600e9159927723e6f7af58c1cb3fc43d035fc/Inc/btt_skr_2.0_map.h#L37

acaballero6270, 2022-01-10:
That seemed to work!!
Thank you very much. Specially for your super fast response!!

closed 2022-01-10 by acaballero6270