grblHAL/STM32F4xx #51: Compile time  errors for STM32F4XX build

jappiemike, 2022-01-12:
Hello, when I try to build the latest version I get these errors
Description	Resource	Path	Location	Type
./Src/driver.o: in function `driver_init':	GRBL Driver STM32F401		 	C/C++ Problem
make: *** [makefile:70: grblHAL F411 Nucleo.elf] Error 1	GRBL Driver STM32F401		 	C/C++ Problem
undefined reference to `i2c_eeprom_init'	driver.c	/GRBL Driver STM32F401/Src	line 1968	C/C++ Problem
 I also noticed there are no eeprom files in the GRBL folder. Am I doing something wrong (as usual)?
Cheers
Mike

rvalotta, 2022-01-12:
What board are you using?

terjeio, 2022-01-12:
This could be due to me messing up the code a bit yesterday. Fix for that just committed.

> I also noticed there are no eeprom files in the GRBL folder. 

They are found in the eeprom folder alongside the grbl folder. If not there something must have gone wrong when fetching the project. BTW this function is called when EEPROM_ENABLE is uncommented in _my_machine.h_ or a board map is selected that forces EEPROM_ENABLE  (e.g. BOARD_MORPHO_CNC does so).

jappiemike, 2022-01-12:
I've been using the ST-MORPHO board layout so that explains the EEPROM_ENABLE that I couldn't account  for in my build which was the 411 release version (sorry if I'm a bit vague I'm recovering from cataract surgery). I'll give it a try when I feel a bit more intelligent (could be decades from now at the present trajectory) .
Cheers
Mike

terjeio, 2022-01-12:
> I'll give it a try when I feel a bit more intelligent (could be decades from now at the present trajectory) .

Ouch - that does not sound any good.

You can disable EEPROM support beeing compiled in for the board if you remove these lines:

https://github.com/grblHAL/STM32F4xx/blob/7774e52513f6bed090a80de42dbaa29b8e5d617c/Inc/st_morpho_map.h#L36-L43

dresco, 2022-01-12:
> or a board map is selected that forces EEPROM_ENABLE (e.g. BOARD_MORPHO_CNC does so).

I use the Morpho option for my Nucleo dev board builds, and will confess I had never noticed this. From the report output, I see it ends up using emulated FRAM for storing settings, is there any downside to leaving it as is?

terjeio, 2022-01-12:
>  From the report output, I see it ends up using emulated FRAM for storing settings, is there any downside to leaving it as is?

If you do not have a FRAM chip connected the your settings/offsets will be lost across sessions. The star in front of FRAM means it is buffered - writes are delayed until the controller state is _Idle_ (or _Alarm_):
`[NVS STORAGE:*FRAM]`
I have not added any handling of write fails, maybe I should - by changing to pure emulation and reporting that? If read fails on startup error 7 is reported and settings reset to default.

closed 2022-01-18 by jappiemike