grblHAL/STM32F4xx #52: grbl has not finished booting: UGS Blackpill STM32F401CCU6

gxdeange, 2022-01-13:
Hi terjeio,

I have flashed the above board but then get the above error when attempting to connect to UGS.

I am not sure if there is anything specific to the flash process but there are no errors regarding USB connectivity and baud rate is 1152000 (Windows 10).

Cheers
George

terjeio, 2022-01-13:
If the reset (or e-stop) pin is not connected to ground the controller starts in _Alarm_ state by default, some senders does not handle that.
Check this [wiki page](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings) for more.

gxdeange, 2022-01-13:
I followed the instructions but the same issue persists.

Regards
George

terjeio, 2022-01-13:
Any response whe connecting with a teminal emulator such as PuTTY? You may also try with the [edge version](http://www.io-engineering.com/downloads) of [ioSender](https://github.com/terjeio/ioSender), it will give better feedback than most senders will.

gxdeange, 2022-01-13:
I could not get any response with PuTTY and have not had a chance to look at the ioSender option.

I enabled vebose logging and I am not sure if this provides any guidance on what might be happening.

[verbose] GrblFeedbackMessage{message='GrblHAL 1.1f ['$' or '$HELP' for help]', distanceMode='null', units='null'}
GrblHAL 1.1f ['$' or '$HELP' for help]

Cheers
George

terjeio, 2022-01-13:
> I could not get any response with PuTTY and have not had a chance to look at the ioSender option.

Did you try to request a real-time report by typing `?` - there are a couple of statuses that blocks all other commands except a soft reset.

> I enabled vebose logging and I am not sure if this provides any guidance on what might be happening.

Well, it _does_ seem like a response is coming from the controller - perhaps from UGS trying a soft reset. If so you should get a response by typing `?` or by trying a soft reset by sending \<Ctrl\>+X in PuTTY.

gxdeange, 2022-01-13:
I have not had any luck with PuTTY and when I send '?' via UGS I still get the 'grbl has not finished booting error message.

I have been looking at how the HSE / HSI is configured within the Cube IDE and was curious if there may be a clock issue? Is it possible for me to manually reconfigure a file that can change to the HSI ?

terjeio, 2022-01-14:
> I have been looking at how the HSE / HSI is configured within the Cube IDE and was curious if there may be a clock issue? Is it possible for me to manually reconfigure a file that can change to the HSI ?

If you can connect the USB clock is likely running at the correct speed. And you got a grblHAL response in the log you mentioned above - you would not get that if there was nothing coming from controller?

Did you try a soft reset from PuTTY?

gxdeange, 2022-01-14:
ok.

I am convinced I have not setup the Cube IDE 'Path and Symbols' properly for the STM32F401CCU6 as I am new to this IDE and am not great with IDE's in general.

I will have a go at add the correct info and see how that goes.

Regards and thanks for your support
George

gxdeange, 2022-01-14:
OK.

I think I have imported the project correctly but I get this error when compiling for 'Release F401 Blackpill'

[image]

terjeio, 2022-01-14:
Seems the latest commit did not update submodules as it should, even if the commit id's for the core and and driver grbl submodules match... Odd.

I have just pushed two new commits that finally seems to bring them in sync.

gxdeange, 2022-01-14:
I just downloaded the updated files and it throws the same error when compiling

terjeio, 2022-01-14:
Did you [update the submodules](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL#using-git-command-line)? The [missing definition](https://github.com/grblHAL/core/blob/9a904127b441ca3d08c38155190f1f235b8ef4dc/core_handlers.h#L99) is there now.

gxdeange, 2022-01-14:
My apologies.

Downloaed the latest Core and has compiled succefully.

Will flash and test.

CHeers
G

gxdeange, 2022-01-14:
Have flashed but I am getting the same error when connecting via UGS and cannot get any type of connection via PuTTY.

terjeio, 2022-01-14:
You cannot connect or not communicate? If connect fails with PuTTY which error message do you get?

gxdeange, 2022-01-14:
Cannot communicate. No error message at all.

gxdeange, 2022-01-14:
Tried different Port number and baud rates. All fail to setup any type of connection. Also tried two USB cables

gxdeange, 2022-01-14:
Is there anything specific with in the IDE that I need to reconfigure or enable? I am sure it is something I am doing wrong.

terjeio, 2022-01-14:
But you still get "GrblHAL..." in the UGS log?

> Tried different Port number and baud rates. All fail to setup any type of connection. Also tried two USB cables

You changed the port number in Device Manager? BTW baud rate usually does not matter, it will be as fast as the connection allows. Since you can connect this means the USB clock is correct and the controller has not hardfaulted.

Do you program via ST link and can set break points? BTW do you have a Nucleo board?

terjeio, 2022-01-14:
> Is there anything specific with in the IDE that I need to reconfigure or enable?

No.

Please tell me which board you have, If a Nucleo dev board the then USB port is not a native USB port, The processor communicates viat UART to the ST-Link processor that translates to/from USB. If a Blackpill do you have a link to where you bought it?

gxdeange, 2022-01-14:
This is all I get from UGS.
[image]

I tried various Port numbers in PuTTY with no success.
[image]

I programmed via a ST-Link and I have a Blackpill which I bought from a local seller on eBay.

Just as a note, I am not too bad with the hardware side but my knowledge around programming and software is very basic so I apologise if I do not get the right message across.

Cheers
G

gxdeange, 2022-01-14:
If I can get this to work I am also looking at building a custom PoC board that uses the 64 Pin variant of the STM32F401 so I can include features such as SD Card and AUX In / Out ports

terjeio, 2022-01-14:
> I tried various Port numbers in PuTTY with no success.

UGS connects to COM5 and get a response, use the same in PuTTY.
To be 100% sure that the logged message form UGS comes from the controller you may modify the welcome message [here](https://github.com/grblHAL/core/blob/9a904127b441ca3d08c38155190f1f235b8ef4dc/grbl.h#L32-L35), recompile and check that it changes. E.g change the version number to 1.1g. If it changes you should be able to get a response from PuTTY, at least from Ctrl+X (soft reset) or a status report by typing `?`.

Have you tried to connect with ioSender?  You may get more info from that.

gxdeange, 2022-01-14:
OK. I can verify that the change I made to the version# is showing in UGS.

[image]

I suspect that a driver is not running correctly as I get nothing more in PuTTY other than what you see in the previous image.

I have not tried ioSender as it seems a bit complicated and I will need more time to read

gxdeange, 2022-01-14:
If I type any commands in UGS I get the 'grbl has not finished booting error or if I press the $X button for instance I get the below error in UGS.
[image]

terjeio, 2022-01-14:
Replace [these lines](https://github.com/grblHAL/STM32F4xx/blob/49b1a712eec7e3caf5ca82ede694e870a10bdcbd/Src/driver.c#L941-L978) with 

`signals.mask = 0;`

and try again.

gxdeange, 2022-01-14:
Unfortunately no difference.

gxdeange, 2022-01-15:
OK. When I revert back to the original signals.mask settings I can see the grgl $ codes witin the serial monitor but then I get errors if I send commands such as $X or $H

gxdeange, 2022-01-15:
BTW, is there a pre compiled version of ioSender that I can download and run?

rvalotta, 2022-01-15:
If you would like I can send you a precompiled version for the STM32F4XX that I know to be working...   If you sent me your pin mapping I can compile a copy with those settings as I'm a little odd with how mine is configured.  

Also not sure if you've seen these but i found them wonderful while working out how i wanted to connect everything,  https://universal-solder.ca/product/stm32-screw-terminal-adapter-for-blue-pill-stm32f103-and-black-pill-stm32f4x1-diy-kit/

gxdeange, 2022-01-15:
Hi rvalotta,

Thanks for your response. I have managed to successfully compile with the update pin settings for the blackpill and I have a conceptual design for a pcb that I will get built as a PoC.

Once I am happy with that I plan to create a prototype board with the 64 pin version of this MCU and look to include SD Card, AUX in/out ports and also set up a ganged motor output.

I am hoping that using a customised my_machine_map file will allow me to do this.

@terjeio 

What do you think is the possibility of having this code setup to use the HSI and LSI clocks as opposed to the external clocks? This would simplify board design and reduce component count. I assume the HSI & LSI will be more than sufficient for the purpose of grbl?
Cheers
George

rvalotta, 2022-01-15:
Yup thats what i did to get the Blackpilll_map file added to the repo... Good luck with your board.. Seems you and I are on similar paths with out black pills

gxdeange, 2022-01-15:
I just want to also say a big thanks to terjeio as my knowledge of software and programming is really limited. Hopefully we may be able to collaborate on the custom design board sometime in the future.

Cheers
George

terjeio, 2022-01-15:
> What do you think is the possibility of having this code setup to use the HSI and LSI clocks as opposed to the external clocks? 

Clock setup [is done here](https://github.com/grblHAL/STM32F4xx/blob/49b1a712eec7e3caf5ca82ede694e870a10bdcbd/Src/main.c#L57-L185) for the different processors/configurations by having symbols defined in the build configurations. An example:

[image]

> Hopefully we may be able to collaborate on the custom design board sometime in the future.

You may get some ideas from my [Nucleo board](https://github.com/terjeio/CNC_Breakout_Nucleo64).

gxdeange, 2022-01-15:
Interesting. I am considering MCU (STM32F7...) options at the moment. What do you think is the minimum Flash size to support features such as Ethernet, WebUI, Bluetooth / WiFi?

terjeio, 2022-01-15:
> What do you think is the minimum Flash size to support features such as Ethernet, WebUI, Bluetooth / WiFi?

WiFi via SPI interface and an external module?

From a build with these parameters:

```
//#define VFD_ENABLE           1 // Set to 1 or 2 for Huanyang VFD spindle. More here https://github.com/grblHAL/Plugins_spindle
//#define DUAL_SPINDLE         1 // Uncomment for switching between VFD spindle and PWM output with $32
//#define MODBUS_ENABLE        1 // Set to 1 for auto direction, 2 for direction signal on auxillary output pin.
#define WEBUI_ENABLE         1 // Enable ESP3D-WEBUI plugin along with networking and SD card plugins.
#define WEBUI_INFLASH        1 // Store WebUI files in flash instead of on SD card.
#define ETHERNET_ENABLE      1 // Ethernet streaming. Uses networking plugin.
#define BLUETOOTH_ENABLE     1 // Set to 1 for HC-05 module. Uses Bluetooth plugin.
#define SDCARD_ENABLE        1 // Run gcode programs from SD card, uses sdcard plugin.
//#define KEYPAD_ENABLE        1 // I2C keypad for jogging etc., uses keypad plugin.
//#define ODOMETER_ENABLE      1 // Odometer plugin.
//#define PPI_ENABLE           1 // Laser PPI plugin. To be completed.
//#define LASER_COOLANT_ENABLE 1 // Laser coolant plugin. To be completed.
//#define OPENPNP_ENABLE 1
//#define FANS_ENABLE 1
//#define TRINAMIC_ENABLE   2130 // Trinamic TMC2130 stepper driver support. NOTE: work in progress.
//#define TRINAMIC_ENABLE   5160 // Trinamic TMC5160 stepper driver support. NOTE: work in progress.
//#define TRINAMIC_I2C         1 // Trinamic I2C - SPI bridge interface.
//#define TRINAMIC_DEV         1 // Development mode, adds a few M-codes to aid debugging. Do not enable in production code.
//#define EEPROM_ENABLE        2 // I2C EEPROM support. Set to 1 for 24LC16 (2K), 3 for 24C32 (4K - 32 byte page) and 2 for other sizes. Uses eeprom plugin.
//#define EEPROM_IS_FRAM       1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.
```

I have allocate quite a bit of RAM to the heap:
```
_Min_Heap_Size = 0x10000;	/* required amount of heap  */
_Min_Stack_Size = 0x3000;	/* required amount of stack */
```

[image]

gxdeange, 2022-01-15:
Could you potenially use something like the ESP 8266 WiFi module instead of HC-05?

JoachimF, 2022-01-16:
Hello,

If it can help, I compiled and tested on blackpill F401CC, the HSE was set to 8000000 and I changed to 25000000, all is OK for me, need now to wire on the CNC, but on iosender all is ok after inverted the limits, probe and buttons.
Here is my .bin

[grblHAL F401 Blackpill.zip](https://github.com/grblHAL/STM32F4xx/files/7877984/grblHAL.F401.Blackpill.zip)

gxdeange, 2022-01-17:
Hi JoachimF,

Thanks for your input. I am hoping to remove the requirement for external crystals altogether and use / configure the internal clocks on these MCU's. This will free up 4 IO's that can then be allocated to a ganged motor output, WiFi connectivity or Aux ports (dependant on overall pins available). 

@terjeio 
Looking at the above image can I assume that a MCU with at least 512KB Flash is the most practicle option for having these additional features.
I have also had a look at the HC-05 and please correct me if I am wrong but this only offers Bluetooth and not 802.1x WiFi? Can a WiFi adaptor such as the ESP-01 be used instead of the HC-05?

Cheers George

terjeio, 2022-01-17:
> Looking at the above image can I assume that a MCU with at least 512KB Flash is the most practicle option for having these additional features.

Yes, and by not having the WebUI in flash  quite a bit of flash is freed.

> I have also had a look at the HC-05 and please correct me if I am wrong but this only offers Bluetooth and not 802.1x WiFi? 

Correct.

> Can a WiFi adaptor such as the ESP-01 be used instead of the HC-05?

Yes - [check this discussion](https://github.com/terjeio/grblHAL/discussions/341#discussioncomment-1915659).

gxdeange, 2022-01-24:
Hi terjeio,

Quick question, if I was to comment out USB_SERIAL_CDC can I then communicate (via a USB to UART device) via one of the USART channels (and if yes, which USART channel)?

My intention is to make this a multi input channel in that I can have direct TX/RX or via an intermediate device (such as the CP2102 USB to UART).

Cheers
George

terjeio, 2022-01-24:
Quick question, if I was to comment out USB_SERIAL_CDC can I then communicate (via a USB to UART device) via one of the USART channels (and if yes, which USART channel)?

Yes, and you can have two channels to choose from depending on the configuration:

https://github.com/grblHAL/STM32F4xx/blob/c6603c717c0e32c989242e4abdacb61dc41e3e12/Src/serial.c#L42-L66

The first is always available, check the code or use the `$pins` command to see the pin mapping.

gxdeange, 2022-01-24:
Awesome. I will give it a go soon.

I have also been trying to bypass the HSE clock and use the HSI by configuring a 401 clock within the CubeIDE and then pasting the settings across into the main.c without success. I then attempted to copy / paste the clock settings for the Nucleo_401 without success either.

Is this possible to do within this file or would I need to change other settings?

Cheers
George

gxdeange, 2022-01-25:
Hi terjeio,

I have commented out '#define USB_SERIAL_CDC' within my_machine.h and reflashed the Blackpill. I have attempted to connect to a USB to UART adaptor to pin 9 (TX) and pin 10 (RX) but cannnot connect to the blackpill via a serial monitor or iosender. I do notice that the RX & TX leds on the adaptor are constantly on which would indicate that pins 9 and 10 are driving high.

I did also notice that within serial.h file the below is the default settings.

[image]

I have also used the same USB to UART adaptor on a Teensy4.1 setup in a similar configuration.

Regards
George

terjeio, 2022-01-26:
@gxdeange Did you delete your last two comments or has github lost them?

gxdeange, 2022-01-26:
Hi terjeio,

Yes I did as I realised what I did wrong and did not want to bother you with it.

I found that:

If I use the Release 401 Nulceo64 I can then use PA2 and PA3 for UART comms and I also noticed that the clock settings are for the HSI (which is what I was after).
I did however notice that the serial.h file allocates GPIOD and pins 8 & 9 so I changed them to GPIOA and pins 2 & 3.

All I need to do now is build a prototype with an embedded STM32F401RCT6 (64 pin version) and no external crystals and test.

Cheers and have a great day
George

gxdeange, 2022-01-29:
Hi terjeio,

I have been playing around with different plugins and config settings but have run into an issue when attempting to compile with SDCARD_ENABLE active. These are the errors I receive.

[image]

I do notice that even though SDCARD_ENABLE is active, within the map.h file I am using, the port / pin assignments are still greyed out.

[image]

I am compiling with Release F401 Nulceo64 which I suspect is coded to not allow SD Cards. Is there a way to bypass this restriction?

Cheers
George

terjeio, 2022-01-30:
Have you added your map file to [driver.h](https://github.com/grblHAL/STM32F4xx/blob/a09e339fb41c95995b87ccd19aa67500b132c509/Inc/driver.h#L106-L138)?

When I compile for BOARD_MORPHO_CNC both the SD card and M3 pin definitions are not greyed out (after an index rebuild that sometimes is required).

gxdeange, 2022-01-30:
I think it is because I am specifying different Ports for STEP and Direction as I was trying to get the best track runs for the PCB.

I guess I will need to go back to one Port for each Step, Direction and Control pins

//#define STEP_PORT               GPIOA
#define X_STEP_PORT				GPIOA
#define X_STEP_PIN              1
#define Y_STEP_PORT				GPIOB
#define Y_STEP_PIN              2
#define Z_STEP_PORT				GPIOA
#define Z_STEP_PIN              11
#define STEP_OUTMODE            GPIO_MAP
//#define STEP_PINMODE            PINMODE_OD // Uncomment for open drain outputs

// Define step direction output pins.
//#define DIRECTION_PORT          GPIOB
#define X_DIRECTION_PORT		GPIOA
#define X_DIRECTION_PIN         4
#define Y_DIRECTION_PORT		GPIOB
#define Y_DIRECTION_PIN         10
#define Z_DIRECTION_PORT		GPIOA
#define Z_DIRECTION_PIN         11
#define DIRECTION_OUTMODE       GPIO_MAP
//#define DIRECTION_PINMODE       PINMODE_OD // Uncomment for open drain outputs

// Define stepper driver enable/disable output pin.
#define STEPPERS_ENABLE_PORT    GPIOB
#define STEPPERS_ENABLE_PIN     1
//#define STEPPERS_ENABLE_PINMODE PINMODE_OD // Uncomment for open drain outputs

// Define homing/hard limit switch input pins.
#define LIMIT_PORT              GPIOB
#define X_LIMIT_PIN             12
#define Y_LIMIT_PIN             13
#define Z_LIMIT_PIN             14
#define LIMIT_INMODE            GPIO_SHIFT12

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS == 1
#define M3_AVAILABLE
#define M3_STEP_PORT            GPIOC
#define M3_STEP_PIN             8
#define M3_DIRECTION_PORT       GPIOC
#define M3_DIRECTION_PIN        9
#if N_AUTO_SQUARED
#define M3_LIMIT_PORT           LIMIT_PORT
#define M3_LIMIT_PIN            15
#endif
#endif

  // Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_PORT     GPIOA
#define SPINDLE_ENABLE_PIN      15
#define SPINDLE_DIRECTION_PORT  GPIOC
#define SPINDLE_DIRECTION_PIN   10

// Define spindle PWM output pin.
#define SPINDLE_PWM_PORT_BASE   GPIOA_BASE
#define SPINDLE_PWM_PIN         8

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_PORT      GPIOC
#define COOLANT_FLOOD_PIN       12
#define COOLANT_MIST_PORT       GPIOC
#define COOLANT_MIST_PIN        11

// Define user-control controls (cycle start, reset, feed hold) input pins.
//#define CONTROL_PORT            GPIOB
#define RESET_PORT				GPIOD
#define RESET_PIN               2
#define FEED_HOLD_PORT			GPIOB
#define FEED_HOLD_PIN           3
#define CYCLE_START_PORT		GPIOB
#define CYCLE_START_PIN         4
#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PORT		GPIOB
#define SAFETY_DOOR_PIN         5
#endif
#define CONTROL_INMODE GPIO_SHIFT5

// Define probe switch input pin.
#define PROBE_PORT              GPIOB
#define PROBE_PIN               8

#if I2C_STROBE_ENABLE
#define I2C_STROBE_PORT         GPIOB
#define I2C_STROBE_PIN          9
#endif

#if SDCARD_ENABLE
#define SD_CS_PORT                  GPIOC
#define SD_CS_PIN                   2
#define SPI_PORT                    1 // GPIOA, SCK_PIN = 5, MISO_PIN = 6, MOSI_PIN = 7
#endif

JoachimF, 2022-01-30:
Hello Terjeio,

What do you think about this board?
https://a.aliexpress.com/_vUFZ4w
All ports will be usable?

dresco, 2022-01-30:
Perhaps also worth noting that WeAct have a couple of STM32H7 boards..

https://www.aliexpress.com/item/1005001700815862.html

terjeio, 2022-01-30:
@gxdeange Here is a map file that compiles with both SD_CARD_ENABLE and M3 definitions ungreyed after an index rebuild:

```
#define IS_NUCLEO_BOB

//#define STEP_PORT GPIOA
#define X_STEP_PORT GPIOA
#define X_STEP_PIN 1
#define Y_STEP_PORT GPIOB
#define Y_STEP_PIN 2
#define Z_STEP_PORT GPIOA
#define Z_STEP_PIN 11
#define STEP_OUTMODE GPIO_BITBAND
//#define STEP_PINMODE PINMODE_OD // Uncomment for open drain outputs

// Define step direction output pins.
//#define DIRECTION_PORT GPIOB
#define X_DIRECTION_PORT GPIOA
#define X_DIRECTION_PIN 4
#define Y_DIRECTION_PORT GPIOB
#define Y_DIRECTION_PIN 10
#define Z_DIRECTION_PORT GPIOA
#define Z_DIRECTION_PIN 11
#define DIRECTION_OUTMODE GPIO_BITBAND
//#define DIRECTION_PINMODE PINMODE_OD // Uncomment for open drain outputs

// Define stepper driver enable/disable output pin.
#define STEPPERS_ENABLE_PORT GPIOB
#define STEPPERS_ENABLE_PIN 1
//#define STEPPERS_ENABLE_PINMODE PINMODE_OD // Uncomment for open drain outputs

// Define homing/hard limit switch input pins.
#define LIMIT_PORT GPIOB
#define X_LIMIT_PIN 12
#define Y_LIMIT_PIN 13
#define Z_LIMIT_PIN 14
#define LIMIT_INMODE GPIO_SHIFT12

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS == 1
#define M3_AVAILABLE
#define M3_STEP_PORT GPIOC
#define M3_STEP_PIN 8
#define M3_DIRECTION_PORT GPIOC
#define M3_DIRECTION_PIN 9
#if N_AUTO_SQUARED
#define M3_LIMIT_PORT LIMIT_PORT
#define M3_LIMIT_PIN 15
#endif
#endif

// Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_PORT GPIOA
#define SPINDLE_ENABLE_PIN 15
#define SPINDLE_DIRECTION_PORT GPIOC
#define SPINDLE_DIRECTION_PIN 10

// Define spindle PWM output pin.
#define SPINDLE_PWM_PORT_BASE GPIOA_BASE
#define SPINDLE_PWM_PIN 8

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_PORT GPIOC
#define COOLANT_FLOOD_PIN 12
#define COOLANT_MIST_PORT GPIOC
#define COOLANT_MIST_PIN 11

// Define user-control controls (cycle start, reset, feed hold) input pins.
//#define CONTROL_PORT GPIOB
#define RESET_PORT GPIOD
#define RESET_PIN 2
#define FEED_HOLD_PORT GPIOB
#define FEED_HOLD_PIN 3
#define CYCLE_START_PORT GPIOB
#define CYCLE_START_PIN 4
#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PORT GPIOB
#define SAFETY_DOOR_PIN 5
#endif
#define CONTROL_INMODE GPIO_BITBAND

// Define probe switch input pin.
#define PROBE_PORT GPIOB
#define PROBE_PIN 8

#if I2C_STROBE_ENABLE
#define I2C_STROBE_PORT GPIOB
#define I2C_STROBE_PIN 9
#endif

#if SDCARD_ENABLE
#define SD_CS_PORT GPIOC
#define SD_CS_PIN 2
#define SPI_PORT 1 // GPIOA, SCK_PIN = 5, MISO_PIN = 6, MOSI_PIN = 7
#endif
```

A few remarks:

`GPIO_BITBAND` for access is always safe as each pin is read separately and _must_ be used if pins in a pin group is spread over several ports.

` GPIO_MAP` and `GPIO_SHIFTn` alternatives can only be used if all pins in the group are on the same port. These are the fastet alternatives as all pins in the group are written or read at the same time.

` GPIO_MAP` can only be used for step and direction signals.

`GPIO_SHIFTn` can only be used for limit and control signal when the pins are consecutive, the lowest pin number is the shift value.

terjeio, 2022-01-30:
> What do you think about this board?
All ports will be usable?

Yes, but you have to write plugin code for the ports that is not used by any of the exsiting board maps.

> Perhaps also worth noting that WeAct have a couple of STM32H7 boards..

This will require a new driver as they have a M7 processor, same as the iMXRT1062. It is likely the F4xx or F7xx driver can be used as a starting point if the peripherals are similar. A challenge for you @dresco?

dresco, 2022-01-31:
> This will require a new driver as they have a M7 processor, same as the iMXRT1062. It is likely the F4xx or F7xx driver can be used as a starting point if the peripherals are similar. A challenge for you @dresco?

You never know, I've been tempted to buy one to play with.. Although they're not far off the price of a Nucleo dev board - at least they're available this year! ;)

dresco, 2022-02-01:
I've gone ahead & ordered a WeAct STM32H7 board to play with, will follow up when it arrives..

closed 2022-02-23 by gxdeange