grblHAL/STM32F4xx #54: Build errors with BTT SKR 2

fitch22, 2022-01-16:
A change in driver.c introduced a call to grbl.on_execute_delay(get_state()), but there is no such element in grbl_t struct.

A change in serial.c introduced PinGroup_UART1 but this is not defined anywhere.  Maybe it should be PinGroup_UART instead?

fitch22, 2022-01-16:
One more: A change to serial.c changed initialization in stream from ".connected = true" to ".state.connected = true".  At present there is no member named "state" in io_stream_t.

Also, ".state.is_usb = On" results in similar error.

Seems like a change in io_stream_t to include member "state" did not make it in.

terjeio, 2022-01-16:
> A change in driver.c introduced a call to grbl.on_execute_delay(get_state()), but there is [no such element in grbl_t struct](https://github.com/grblHAL/core/blob/9a904127b441ca3d08c38155190f1f235b8ef4dc/core_handlers.h#L99).

Have you [updated the submodules](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL#using-git-command-line)?

A commit went bad and was fixed 2 days ago.

fitch22, 2022-01-16:
I thought I had already, but just in case I did a "git pull --recurse-submodules" and everything was up to date.  Tried another build and got same results.

terjeio, 2022-01-16:
If you compare your _core_handlers.h_ with the one I linked to above is `on_execute_delay` still missing in yours?

fitch22, 2022-01-16:
Terje,
Something must be broken on my end.  I looked at the file in your github and it has on_execute_delay.  If I look at my repository, which is a clone of grblHAL/STM32F4xx and follow the link into the grbl submodule, the file is not the same.  I just did a fetch and merge today, so it should be current.  Could the link be pointing somewhere it shouldn't?  My grbl shows last update 2 months ago.

Not sure what to do now.

terjeio, 2022-01-17:
> Not sure what to do now.

Could it be related to the PR you made? It has many changes that I cannot merge without creating a mess. My suggestion would be to save the three files actually changed and start from scratch (rebase?). Restore the changed files and try again.

[image]

fitch22, 2022-01-17:
I was thinking the same thing.  I will cancel the PR and try to fix things.  Sorry for the hassle.

closed 2023-08-26 by terjeio