grblHAL/STM32F4xx #58: What are the requirements to support ganged motors with Trinamic drivers

jevermeister, 2022-01-31:
I am currently porting grblHAL to a BTT Rumba32 board with UART mode for the TMC2209 I'll be using.

I read in the SKR 2.0 driver file 

#if TRINAMIC_ENABLE && N_GANGED > 0
#error "BTT SKR-2 does not support ganged motors with Trinamic drivers."
#endif

however the SKR 1.1 driver doesn't have such a limitation although both boards support UART mode.

Could you tell the reason that the SKR 2.0 doesn't support ganged motors with Trinamic drivers so I know before trying this on my Rumba32 port?

terjeio, 2022-01-31:
> Could you tell the reason that the SKR 2.0 doesn't support ganged motors with Trinamic drivers so I know before trying this on my Rumba32 port?

Most likely th reason is that the test was not removed when ganged support was added to the motors plugin. Try with the lines removed.

WellDone2094, 2022-02-02:
Hi @jevermeister, did you manage to get ganged motors working on skr 2?

jevermeister, 2022-02-02:
> Hi @jevermeister, did you manage to get ganged motors working on skr 2?

Hi,

I don't have a SKR 2 but a Rumba32. As both use UART mode for the trinamic drivers I assume that in case it works, it will work for both boards. However currently I am still building my MPCNC and did not switch to grblHAL yet. Therefore I don't know, sorry.

terjeio, 2022-02-03:
@WellDone2094 Did it fail for you after removing the lines I mentioned?

WellDone2094, 2022-02-03:
@terjeio compilation was fine without those line, but i'm still waiting for the board to be delivered to test it.
i used N_GANGED=2 and N_ABC_MOTORS=2, is that correct if i want to have 2 X motors and 2 Y motor and a limit switch for each? i would like to have auto squaring on both axis.

terjeio, 2022-02-03:
> i used N_GANGED=2 and N_ABC_MOTORS=2, is that correct if i want to have 2 X motors and 2 Y motor and a limit switch for each?

No - do not change those symbols - configuration should be done in [my_machine.h](https://github.com/grblHAL/STM32F4xx/blob/1551f39107291ab782659b9a659223a16e2313de/Inc/my_machine.h#L77-L80), the preprocessor sorts out the rest.

WellDone2094, 2022-02-03:
oh, i didn't see that file, that's much easier, i'll try it. Thanks

WellDone2094, 2022-02-03:
i can confirm it compiles, i should be able to test it on the board this weekend.

WellDone2094, 2022-02-04:
hi @terjeio,
i commented the line  `-D OVERRIDE_MY_MACHINE` inside platformio.ini and set `#define TRINAMIC_ENABLE   2209 // Trinamic TMC2209 stepper driver support. NOTE: work in progress.` inside my_machine.h

i then built the frimware with the command `pio run --environment btt_skr_2_tmc2130.
The firmware build without issues and it get flashed succesfully, but after flashing it my pc doesn't see the board anymore.

Any suggestions?

terjeio, 2022-02-04:
You flash with a SD card (bootloader)? If so [check this comment](https://github.com/grblHAL/STM32F4xx/issues/60#issuecomment-1028396822).

WellDone2094, 2022-02-06:
thanks that worked! and the ganged + auto-squaring worked!

WellDone2094, 2022-02-07:
after flashing it again, it seems way more stable.
The only problem now is that setting the correct microstep and set/mm seems to move too much (about two times).
I have a gt2 belt, 1.8 stepper driver, 16 microsteps, 16T pulley. the correct value should be 100 but i have to set 50 if i want to move the correct distance

WellDone2094, 2022-02-07:
i tired to connect to the board using putty after iosender freezed but it doesn't connect. I also tried to unplug and replug the board to make sure nothing was using the port but when i plug it in the pc doesn't recognize it. After a reset everything works again and i can connect to it with putty without issues. is there something i can do to debug it? debug mode, connect in a different way than usb or anything else?

WellDone2094, 2022-02-08:
about the step/mm i'm using a 1.8 deg motor, with a 16T pully and a gt2 belt, this shoud have a 100 step/mm but i need to use 50  if i want to move correctly. skr 2.0 with tmc 2209

terjeio, 2022-02-08:
> I also tried to unplug and replug the board to make sure nothing was using the port but when i plug it in the pc doesn't recognize it.

OK, then it is likely the controller has crashed for some reason, a bug or electrical interference? FYI I ran a 3hr test yesterday with a F446 board without issues - will run a longer one later today.

> is there something i can do to debug it? debug mode, connect in a different way than usb or anything else?

Without a debugger your options is to try a different USB cable, bypass any USB hub or bypass board USB by connecting via a UART <> breakout board. Does it take long for the hang to happen? Does it happen when sitting idle? When the spindle starts? ...?

> about the step/mm i'm using a 1.8 deg motor, with a 16T pully and a gt2 belt, this shoud have a 100 step/mm but i need to use 50 if i want to move correctly. skr 2.0 with tmc 2209

Set steps/mm to 200, microstepping to 16 and move the axis 16mm - does the motor turn one revolution? I did for me.

WellDone2094, 2022-02-08:
> Set steps/mm to 200, microstepping to 16 and move the axis 16mm - does the motor turn one revolution? I did for me.
i tried and the stepper turn two revolutions
here you can see my grblahl settings [https://ibb.co/mRHdhfk](settings)

> Does it take long for the hang to happen? Does it happen when sitting idle? When the spindle starts? ...?
it seems to be random, sometimes it take 2 minutes to crash sometime after 30 minute it's still working. 
often it happens when i try to move it, or while changing settings.
I don't have a spindle attached yet

Today i will use a different cable and see if that helps.

Thanks for your support

WellDone2094, 2022-02-09:
i will test it later today, i got an st-link v2 if that can help identifying the problem

acaballero6270, 2022-02-09:
git clone https://github.com/grblHAL/STM32F4xx.git is not downloading all the files. At least not the grbl files.

WellDone2094, 2022-02-10:
i flashe the new version and run M122 again, this time it worked and i got this output
```
M122
[TRINAMIC]
                      X       Y       Z      X2      Y2
Driver          TMC2209 TMC2209 TMC2209 TMC2209 TMC2209
Set current         700     700     500     700     700
RMS current         673     673     489     673     673
Peak current        951     951     691     951     951
Run current       21/31   21/31   15/31   21/31   21/31
Hold current      10/31   10/31    7/31   10/31   10/31
CS actual         10/31   10/31   10/31   10/31   10/31
PWM scale            10      10      10      10      10
vsense          1=0.180 1=0.180 1=0.180 1=0.180 1=0.180
stealthChop        true    true    true    true    true
msteps               16      16      16      16      16
tstep           1048575 1048575 1048575 1048575 1048575
pwm
threshold             0       0       0       0       0
[mm/s]                0       0       0       0       0
OT prewarn        false   false   false   false   false
OT prewarn has
been triggered    false   false   false   false   false
off time              3       3       3       3       3
blank time            1       1       1       1       1
hysteresis
-end                 -1      -1      -1      -1      -1
-start                1       1       1       1       1
Stallguard thrs       0       0       0       0       0
DRIVER STATUS:
stallguard     
sg_result             0       0       0       0       0
fsactive                                               
stst                  *       *       *       *       *
olb                                                    
ola                                                    
s2gb                                                   
s2ga                                                   
otpw                                                   
ot                                                     
STATUS REGISTERS:
 X = 0xC0:0A:00:00
 Y = 0xC0:0A:00:00
 Z = 0xC0:0A:00:00
 X2 = 0xC0:0A:00:00
 Y2 = 0xC0:0A:00:00
ok
```
i tried to set step/mm=200 and move 16mm but the stepper still turn 2 revolution

acaballero6270, 2022-02-10:
Make sure you remove the jumpers on MS1 and MS2. There should be only one on the UART pin of the TMC2209.

WellDone2094, 2022-02-10:
under the driver there is only one jumper like [this](https://3dwork.io/wp-content/uploads/2020/06/Modo_UART_TMC2209_BIGTREETECH.jpg)

terjeio, 2022-02-10:
> i tried to set step/mm=200 and move 16mm but the stepper still turn 2 revolution

What happens if you change the [microsteps setting](https://github.com/grblHAL/Plugins_motor#15x---motor-microsteps-x--0-for-x-axis-1-for-y-axis-etc)?
I can see that the Z driver has a different [current setting](https://github.com/grblHAL/Plugins_motor#14x---stepper-current-x--0-for-x-axis-1-for-y-axis-etc) from the other drivers in the report above, can you try to set each differently and see if reported correctly?

WellDone2094, 2022-02-10:
I set each axis to a different current and that seems to be working, but when i set Y-axis to 8 microsteps all the driers got changed to 8.  Moving 16mm with 8 or 16 microstpes keeping 50 steps/mm produced the same results. The stepper rotated the same amount.

```
M122
[TRINAMIC]
                      X       Y       Z      X2      Y2
Driver          TMC2209 TMC2209 TMC2209 TMC2209 TMC2209
Set current         700     600     500     700     600
RMS current         673     581     489     673     581
Peak current        951     821     691     951     821
Run current       21/31   18/31   15/31   21/31   18/31
Hold current      10/31    9/31    7/31   10/31    9/31
CS actual         10/31   10/31   10/31   10/31   10/31
PWM scale            10      10      10      10      10
vsense          1=0.180 1=0.180 1=0.180 1=0.180 1=0.180
stealthChop        true    true    true    true    true
msteps                8       8       8       8       8
tstep           1048575 1048575 1048575 1048575 1048575
pwm
threshold             0       0       0       0       0
[mm/s]                0       0       0       0       0
OT prewarn        false   false   false   false   false
OT prewarn has
been triggered    false   false   false   false   false
blank time            1       1       1       1       1
hysteresis
-end                 -1      -1      -1      -1      -1
-start                1       1       1       1       1
Stallguard thrs       0       0       0       0       0
DRIVER STATUS:
stallguard     
sg_result             0       0       0       0       0
fsactive                                               
stst                  *       *       *       *       *
s2gb                                                   
s2ga                                                   
STATUS REGISTERS:
 X = 0xC0:0A:00:00
 Y = 0xC0:0A:00:00
 Z = 0xC0:0A:00:00
 X2 = 0xC0:0A:00:00
 Y2 = 0xC0:0A:00:00
```

After resetting the board all driver go back to 16 microsteps even if grblhal shows 8 for the Y axis

terjeio, 2022-02-10:
Do you have an oscilloscope? I wonder if this somehow is related to the motor power circuitry on this board.
If I remove motor power from the stepper driver and reconnect it msteps goes to 8 which I assume is the default value in the driver. Some data presented in the `M122` report are read from the driver and some from "shadow registers" kept in MCU RAM only - since some changes and others not suggests that the drivers somehow are reset when they should not be.

In my test setup a `M122I` restores msteps to the configured value after reconnecting motor power.

WellDone2094, 2022-02-10:
unfortunately I don't have an oscilloscope, from my tests I can see two issues
- changing microsteps doesn't seem to change the stepper/driver behavior
- changing the microstep on one of the axes affects all the drivers, I also noticed that after the reset all the drivers have the same microstep settings as the X-axis, this lead me to think that the X-axis is the last one to be confgured and since the microsteps settings is applied to all the driver this affect all other axis as well

WellDone2094, 2022-02-10:
I have never worked with stm32 and i don't have much experience with microcontrollers, I'll see if I manage to understand the code and debug it. Could you point me to the part of the code that manages all this?

terjeio, 2022-02-10:
There are basically four layers of code:

[trinamic.c](https://github.com/grblHAL/Plugins_motor/blob/108431e3b8e9db44b75e2beaf101714587a9df2e/trinamic.c) - which extends the grblHAL core with Trinamic specific settings, M-codes, orchestrates driver initialization etc. Most of this code is not Trinamic driver specific.

[tmc2209_hal.c](https://github.com/terjeio/Trinamic-library/blob/1fe15f73a1235326a561305407bce30d53906034/tmc2209hal.c) which is a relatively thin layer abstracting the different Trinamic driver implementations into an common API for use by _trinamic.c_. This is the TMC2209 version of that API.

[tmc_2209.c](https://github.com/terjeio/Trinamic-library/blob/1fe15f73a1235326a561305407bce30d53906034/tmc2209.c) which is contains data structures and code that matches the TMC2209 registers, handles driver initialization and calls the low level code that performs the actual data transfers.

[tmc_uart.c](https://github.com/grblHAL/STM32F4xx/blob/master/Src/tmc_uart.c) is the low level code which handles the data transfers for UART mode drivers when using separate software UARTs for each driver.
[st_morpho.c](https://github.com/grblHAL/STM32F4xx/blob/master/Src/st_morpho.c) is an alternative low level implementation for hardware UART communication where the driver is addressed in the message itself. This is not used by the SKR 2.0 board and is a lot simpler...

Only the low level code differs between the different boards and processors there are grblHAL drivers for. The top three layers are shared between them.

WellDone2094, 2022-02-11:
I did not, I'm going to update it and test it again, thanks you for pointing it out

WellDone2094, 2022-02-12:
I have been testing it for the last two days and i have had quite few crashes where the board is not recognized by the pc anymore, before the last update it looked more stable, i will have to figure out how to connet the st-link to the board to get some debug info. Do i need to use STMcube to debug it?

WellDone2094, 2022-02-13:
>A crash always happens during motion? Never when left idle?

i test this and it crash also in idle, i turn it on, didn't move and after a few minutes it was not responding

terjeio, 2022-02-13:
> i test this and it crash also in idle, i turn it on, didn't move and after a few minutes it was not responding

Can you try with a different computer?
Assuming the MCU not is powered by USB, can you connect after disconnecting/reconnecting the USB cable?

@acaballero6270 Is the connection stable for you?

WellDone2094, 2022-02-13:
I tried to use a different pc and I get the same behavior, I tried to unplug and replug after the crash and the USB device doesn't get recognized, but as soon as I reset the board the pc recognize the device. I also tried to change pc after the crash but it didn't help. The new pc doesn't recognize the device until I reset it

terjeio, 2022-02-14:
I only have a F411 Blackpill with native USB to test with, it has now run for 24 hours in idle mode with the console displaying the real time reports.
Do you have any plugins enabled other than the Trinamic enabled?
Do you have a (FTDI) UART <> USB breakout so we can test without having the MCU USB stack involved?

WellDone2094, 2022-02-14:
yes i ave an FTDI and i just got a logic analyzer if that can help

WellDone2094, 2022-02-14:
i think it's working find for you, on my board it takes less than 10 minutes from boot to crash, i'll try to debug as you suggested and get back to you

WellDone2094, 2022-02-14:
meanwhile i will run a longer program with FTDI

WellDone2094, 2022-02-14:
after a longer test it crashed using FTDI too, I tried to send commands using putty but it's not responding, as soon as I rest the board it starts working again.

acaballero6270, 2022-02-14:
Connection is stable for me

terjeio, 2022-02-14:
> Connection is stable for me
@acaballero6270 Thanks for this info.

Do we have some kind of wiring issue at hand? Ground loops? Interference? Bad board? Bad power supply?
Is Marlin stable?

WellDone2094, 2022-02-14:
is @acaballero6270 using an skr 2.0 with tmc 2209? 

>> Is Marlin stable?

i haven't tested it

>> Do we have some kind of wiring issue at hand? Ground loops? Interference? Bad board? Bad power supply?

i'm not sure it's wiring issues, before the last commit where you fixed the connection to tmc 2209 the system was using the wrong microstep but it seemd very stable.

I have been trying to use stlink utility to see some logs but  i have never used it before so i did connect to the board but I cannot get the swv working

acaballero6270, 2022-02-15:
Yes. SKR 2.0 with TMC2209's. Both Marlin & GrblHal work fine so far. Sensorless homing works also.

terjeio, 2022-02-15:
> i'm not sure it's wiring issues, before the last commit where you fixed the connection to tmc 2209 the system was using the wrong microstep but it seemd very stable.

I believe "seemed" is the right term, the Trinamic code is only executed during startup and when settings are changed, sensorless homing is performed or M122 reports are requested. The execution path trough the code when in `Idle` mode is far simpler than during motion, and even simpler if no real time reports are continually requested. Since the code crashes for you even when in `Idle` mode suggest to me that there is something very basic going wrong.  BTW does it happen in `Idle` mode when connecting via the FTDI breakout as well, both with a sender requesting real time reports or with a terminal such a PuTTY doing nothing - and then after some time requesting a report manually?

terjeio, 2022-02-15:
@acaballero6270 Dou you still have a the Marlin binary and can you add it to a comment for @WellDone2094 to try with?

WellDone2094, 2022-02-15:
i spent few hours trying to get debug working, doing so i fucked up everything so i had to reflash the bootloader and everything else.
Now, without making any changes, everything has been running without issues for 30 minutes, i will run a long program all afternoon to test the stability.

I have no idea what was wrong before, but somehow flashing everything including the bootloader fixed it.

WellDone2094, 2022-02-16:
after flashing everything I haven't had a single crash in 5 hours. I think the problem is solved. Thanks everyone for your patience. 
Is there a donation page or something like that so I can give something back to this amazing project?

closed 2023-08-26 by terjeio