grblHAL/STM32F4xx #59: Could not communicate with stepper driver!

acaballero6270, 2022-01-31:
I enabled trinamic drivers on my bigtreetech SKR2 board with TRINAMIC_ENABLE 2209. It seems to work as I get more settings like $140 $150. However when I try M122 I get the message: "Could not communicate with stepper driver!". I only want to set the microsteps. If I read $150 I get 16. However, the motor moves as if it has 8 microsteps. Trying to set $150 to another value does not work. I keep getting 16. I only have one jumper on UART signal as described in the user guide for SKR2. If I connect my scope to pin 4 on the driver (UART) I don't see any signal. Is there some other symbol I have to define?

terjeio, 2022-01-31:
> Is there some other symbol I have to define?

[$338](https://github.com/grblHAL/Plugins_motor#338---driver-enable)?

acaballero6270, 2022-01-31:
I set $338 to 7. Still the same.

terjeio, 2022-02-01:
Do you program with a ST-Link and can set breakpoints?

acaballero6270, 2022-02-01:
Yes I do. However, I'm not that familiar with debugging. I can give it a try.

terjeio, 2022-02-01:
Ok, good. You should use a debug build for debugging though as optimizations is likely make a mess of the code by rearranging it.

In [btt_skr_2.0.c](https://github.com/grblHAL/STM32F4xx/blob/master/Src/btt_skr_2.0.c) set a breakpoint at line 518, that should be hit several times. Then try one at line 255.

You may also try to switch to the btt_skr_1.1.c board support code, but that requires a change in the map file as SPI and UART pins are defined differently.

acaballero6270, 2022-02-01:
I did the breakpoints on both lines. It does hit them. I tried the btt_skr_1.1.c but I kept getting the same results.
However, I loaded the original factory Marlin firmware, and while it seems to accept the M122 command. I noticed that at the end it gives a "TMC CONNECTION ERROR" message. In addition, even though it shows the microsteps at 16. The motors behave as 8 microsteps. Also, It wont let me change the microsteps it keeps reading 16.
So, if the original firmware doesn't work either perhaps the board itself has a problem.
On the other hand, using the jumpers for MS1 and MS2 I'm able to configure the microsteps. So, I can work like that.
If you want me to try something else please let me know and I'll be happy to do it.
Regards

terjeio, 2022-02-01:
What kind of make of TMC2209 drivers do you have? Fystec or Watterott?
I had to solder a bridge on the Watterot drivers I use for my breakout board...

acaballero6270, 2022-02-01:
It's the Bigtreetech TMC2209-V1.2 driver board. It has Trinamic TMC2209 chip.

acaballero6270, 2022-02-01:
They came with a jumper already soldered to pin 4. Which is where the  RX pin of the UART on the uC is. The strange thing is that there is only one pin (PE0 for X motor). While the datasheet for the 2209 shows 2 pins one RX and one TX connected together through a resistor and then the RX signal directly to the uC pin.

acaballero6270, 2022-02-03:
Since there is only one wire. It must require half duplex communication.

terjeio, 2022-02-03:
>  It must require half duplex communication.

It is, and I find it a bit strange that there is no signal on the UART pin since the breakpoint was triggered. It could be due to the motors beein initialized from the highest numbered (usually Z) - if that fails the rest is skipped. Best is to enable only X by setting $338 to 1 and checking the X driver UART pin.

acaballero6270, 2022-02-03:
Tried $338=1. Nothing changes.  In fact when I do I get the message "Could not communicate with stepper driver!". I also noticed that if I reset the board $338 is back to 7.

terjeio, 2022-02-04:
I have verified driver communication with a F446 and a single TMC2209 with the _btt_skr_2.0.c_ code so I assume the code is good. The F429 is seems to be similar enough to the F407 so that should not be an issue.

A bit strange that the $338 setting reverts to 7 on a reset, if you change another does, say $0=5, does that revert back as well?

It can be noted that I had to add a delay (100ms for TMC2130) to the LPC176x driver before initializing the TMC drivers, wonder if that could be an issue here too since the motor power is switched on by the MCU. If so a `M122I` command attempts to reinitialize the drivers. FYI you can use this command when checking the UART signal as well.

acaballero6270, 2022-02-05:
Something weird is going on. I never noticed this before. Whenever I change any setting they revert back after resetting. I reloaded the firmware and then used LaserGrbl to import all the settings I had saved sometime ago and the settings changed just fine and remained even after a reset. If I try to change them afterwards either with an import or a manual change again they revert after a reset.
The M122I command also gives an error.

terjeio, 2022-02-05:
> If I try to change them afterwards either with an import or a manual change again they revert after a reset.

Does this apply to all settings? $0 too as I asked about above?

> The M122I command also gives an error.

You have to find out why there is no signal on the UART pin(s) even as the code is in fact called. If you [set a breakpoint here](https://github.com/grblHAL/STM32F4xx/blob/6f5f5e32c573b6e08e647385da19ef3aff80bb6c/Src/btt_skr_2.0.c#L232) you can single step through each bit sent and measure the output, even with a voltmeter. If nothing is reaching the driver there is either someting wrong with the code (odd since it works form me, but for the record I changed the X pin as there is no E port on the MCU I tested with) or the pin mapping is not correct.

A potential issue with the code is the [empty while loops](https://github.com/grblHAL/STM32F4xx/blob/6f5f5e32c573b6e08e647385da19ef3aff80bb6c/Src/btt_skr_2.0.c#L203), they should execute at least a NOP instruction - some ARM processors seems not to mind others does.

acaballero6270, 2022-02-06:
OK. By going through the breakpoint you indicated I noticed it was talking to axis Z as you suspected. I believe the reason I couldn't change the settings is because of Lasergrbl import. Right after reloading the firmware I'm able to change the settings normally. I then set $338=1 and now I see a signal on axis X upon resetting. But, shouldn't I see a signal with $338=7 too?

terjeio, 2022-02-06:
> But, shouldn't I see a signal with $338=7 too?

If the Z or Y drivers are not responding then no, the initialization sequence is aborted if a driver does not respond.  Start with the X-axis and when `M122` responds with driver information you can add more drivers, one at a time.

You may check that the code is reading data by unplugging a driver and connect the UART pin to GND and 3V3 and check that the [read data](https://github.com/grblHAL/STM32F4xx/blob/6f5f5e32c573b6e08e647385da19ef3aff80bb6c/Src/btt_skr_2.0.c#L315) (wdgr->data array) is either all 0x00 (GND) or all 0xFF (3V3). If this is ok then proceed to find out why drivers are not responding. Bad or inconsistent baud rate? Drivers without motor power (they do not respond if not powered)...

terjeio, 2022-02-06:
Looks like the [code is working](https://github.com/grblHAL/STM32F4xx/issues/58#issuecomment-1030812022), do you have bad drivers or is there something about the F429 that has to be coded for differently?

acaballero6270, 2022-02-07:
I compiled another version of Marlin. It works flawlessly. Communicates with the drivers. There is signal going back and forth. Looking at the signals on the drivers it seems to be at 9600 baud. While the signal sent by GrblHal looks at least twice as fast. And all I can see is a signal going but nothing coming back. Is the baud rate specified somewhere in the code?

terjeio, 2022-02-09:
[New version is up](https://github.com/grblHAL/STM32F4xx/issues/58#issuecomment-1033769834) - there were several isues with the code that I hope is now fixed. The baud rate output from F466 I tested with is ~100000.

acaballero6270, 2022-02-09:
I tried the new version but it doesn't communicate with the drivers either. I looked into btt_skr_2.0.c and the section on TRINAMIC_ENABLE == 2209 is mostly gone. So I didn't try anymore.
Using the previous version I managed to get the X driver working by changing SWS_BAUDRATE in btt_skr_2.0.c line 135 to 25000. 50000 works also. Also had to set $338=1. Other values do not work. Where is the number of axes defined?

terjeio, 2022-02-10:
The two versions of the UART code has been consolidated into [tmc_uart.c](https://github.com/grblHAL/STM32F4xx/blob/master/Src/tmc_uart.c) and [TRINAMIC_ENABLE == 2209 is used to set a new symbol TMC_UART_ENABLE](https://github.com/grblHAL/core/blob/0fd41b49bfbba50f3c56fb88f848227d729da647/driver_opts.h#L131-L135) that is used to guard the code. This to make it easier to add new driver variants without editing the low level data exchange code of which there are currently only two versions, SPI and UART.

> Where is the number of axes defined?

In [grbl/config.h](https://github.com/grblHAL/core/blob/0fd41b49bfbba50f3c56fb88f848227d729da647/config.h#L40).

I find it a bit strange that @WellDone2094 now seems to be able to [communicate with the drivers](https://github.com/grblHAL/STM32F4xx/issues/58#issuecomment-1034359595) and you are not, is it due to he having the F407 variant and the F407 and F427 processors clocking the UART timer differently? FYI the F446 I use for testing outputs a bit every 10 microsconds.

acaballero6270, 2022-02-10:
OK. before I did your suggestions I downloaded again the entire updated modules & submodules. Now the situation is different. Using all defaults. I still get the Could not communicate with driver error. With $338=3 M122 gives me:
M122
[TRINAMIC]
                      X       Y
Driver          TMC2209 TMC2209
Set current         500     500
RMS current         489     489
Peak current        691     691
Run current       15/31   15/31
Hold current       7/31    7/31
CS actual          7/31    7/31
PWM scale             9       9
vsense          1=0.180 1=0.180
stealthChop        true    true
msteps               16      16
tstep           1048575 1048575
pwm
threshold             0       0
[mm/s]                0       0
OT prewarn        false   false
OT prewarn has
been triggered    false   false
off time              3       3
blank time            1       1
hysteresis
-end                 -1      -1
-start                1       1
Stallguard thrs       0       0
DRIVER STATUS:
stallguard
sg_result             2       0
fsactive
stst                  *       *
olb
ola
s2gb
s2ga
otpw
ot
STATUS REGISTERS:
 X = 0xC0:07:00:00
 Y = 0xC0:07:00:00
ok

With $338=7 I get the Could not communicate error. So, I think it is able to communicate with 2 drivers only.

terjeio, 2022-02-11:
With $338=4 do you get any signal on the Z UART pin?

acaballero6270, 2022-02-11:
The instant I set $338=4 I see a short signal on Z and I get the Could not communicate with driver error. M122 gives the same error but no signal at all on any driver.

terjeio, 2022-02-11:
Here is what I get from the first message, the first 0.4 ms is the request, the rest is the response. You have to use `M122I` to attempt a reinit of the driver, `M122` will just repeat the error message without any transfers taking place. And with `$338=4` only the Z driver is initialized.

[image]

acaballero6270, 2022-02-11:
With M122 I I get the same short signal (about 448 usec with 100K baudrate). No response from the driver. Also it seems to hang up with no OK response.

acaballero6270, 2022-02-11:
Picture attached
[image]

terjeio, 2022-02-11:
> Also it seems to hang up with no OK response.

Ouch - that is a bug showing up when the X-axis is not enabled, it crashes the controller. I'll fix that.

Your signal is odd, it is pulled down to about 1V after the request is sent - by what? If you unplug the driver do you get the same? And if you swap in another driver the same as well? If I unplug the driver I only get a slight dip in the voltage from 3.3V - 100mV or so.

terjeio, 2022-02-11:
Ok, I get the same as you when motor power is not applied. Trinamic drivers do not respond without motor power!

acaballero6270, 2022-02-11:
Power is applied

terjeio, 2022-02-11:
OK, is the driver defective then? Swap drivers or test without it beeing inserted.

acaballero6270, 2022-02-11:
That seemingly odd voltage level normally happens with an output that is set to open drain. With no pullup or pulldown. It stays like that for a short time right after the transmit signal. I have swapped drivers

acaballero6270, 2022-02-11:
And your suspicion was correct ! I changed the driver again and now everything works!!! Bad driver even though it was  new right out of the box.

acaballero6270, 2022-02-11:
I believe that solves my driver communication problem. Thank you very much for everything.

acaballero6270, 2022-02-11:
Thanks for you continued support. I'm sure GrblHal will be a success.

closed 2022-02-11 by acaballero6270