grblHAL/STM32F4xx #60: Testing on BTT SKR PRO 1.2

gambrose, 2022-02-02:
I have a BIGTREETECH SKR PRO V1.2 I am trying to get grbl running on.

I have tried to build using the instructions on platform.io. All the configured environments build expect for the `btt_skr_pro_1_1` one. Which annoying is the one I need.

Building from master checked out with `--recurse-submodules` I get the following build error

```bash
$ pio run --environment btt_skr_pro_1_1
Processing btt_skr_pro_1_1 (board: genericSTM32F407VGT6; platform: ststm32; framework: stm32cube)
---------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VGT6.html
PLATFORM: ST STM32 (15.2.0) > STM32F407VG (128k RAM. 1024k Flash)
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES:
 - framework-stm32cubef4 1.26.2
 - tool-ldscripts-ststm32 0.2.0
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 80 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <bluetooth>
|-- <grbl>
|-- <keypad>
|-- <laser>
|-- <motors>
|-- <odometer>
|-- <spindle>
|-- <Core>
|-- <Class>
|-- <App>
|-- <Target>
|-- <eeprom>
Building in release mode
Compiling .pio/build/btt_skr_pro_1_1/FrameworkHALDriver/Src/stm32f4xx_hal.o
Compiling .pio/build/btt_skr_pro_1_1/FrameworkHALDriver/Src/stm32f4xx_hal_adc.o
Compiling .pio/build/btt_skr_pro_1_1/FrameworkHALDriver/Src/stm32f4xx_hal_adc_ex.o
Compiling .pio/build/btt_skr_pro_1_1/FrameworkHALDriver/Src/stm32f4xx_hal_can.o
... removed from brevity ... 
Compiling .pio/build/btt_skr_pro_1_1/lib715/grbl/wall_plotter.o
Compiling .pio/build/btt_skr_pro_1_1/lib7fe/keypad/keypad.o
Compiling .pio/build/btt_skr_pro_1_1/lib256/laser/coolant.o
Compiling .pio/build/btt_skr_pro_1_1/lib256/laser/ppi.o
Compiling .pio/build/btt_skr_pro_1_1/liba00/motors/trinamic.o
In file included from Inc/driver.h:119:0,
                 from keypad/keypad.c:26:
Inc/btt_skr_pro_v1_1_map.h:25:2: error: #error "This board has STM32F407 processor with a 8MHz crystal, select a corresponding build!"
 #error "This board has STM32F407 processor with a 8MHz crystal, select a corresponding build!"
  ^~~~~
*** [.pio/build/btt_skr_pro_1_1/lib7fe/keypad/keypad.o] Error 1
In file included from Inc/driver.h:119:0,
                 from laser/ppi.c:24:
Inc/btt_skr_pro_v1_1_map.h:25:2: error: #error "This board has STM32F407 processor with a 8MHz crystal, select a corresponding build!"
 #error "This board has STM32F407 processor with a 8MHz crystal, select a corresponding build!"
  ^~~~~
*** [.pio/build/btt_skr_pro_1_1/lib256/laser/ppi.o] Error 1
In file included from Inc/driver.h:119:0,
                 from laser/coolant.c:24:
Inc/btt_skr_pro_v1_1_map.h:25:2: error: #error "This board has STM32F407 processor with a 8MHz crystal, select a corresponding build!"
 #error "This board has STM32F407 processor with a 8MHz crystal, select a corresponding build!"
  ^~~~~
*** [.pio/build/btt_skr_pro_1_1/lib256/laser/coolant.o] Error 1
In file included from Inc/driver.h:119:0,
                 from motors/trinamic.c:25:
Inc/btt_skr_pro_v1_1_map.h:25:2: error: #error "This board has STM32F407 processor with a 8MHz crystal, select a corresponding build!"
 #error "This board has STM32F407 processor with a 8MHz crystal, select a corresponding build!"
  ^~~~~
*** [.pio/build/btt_skr_pro_1_1/liba00/motors/trinamic.o] Error 1
============================================ [FAILED] Took 20.85 seconds ============================================

Environment      Status    Duration
---------------  --------  ------------
btt_skr_pro_1_1  FAILED    00:00:20.853
======================================= 1 failed, 0 succeeded in 00:00:20.853 =======================================
```

Changing `-D HSE_VALUE=25000000` to `-D HSE_VALUE=8000000` does enable it to build.
8000000 seams more in keeping with the 8MHz crystal comment.

I have copied the firmware.bin to an sd card and flashed the board but it does not appear in windows as a USB serial port.

Could you advise on the next steps to try?

gambrose, 2022-02-02:
OK, so I have made some more progress.

following some comments from #46 
I have also changed `board_build.ldscript = STM32F407VGTX_FLASH.ld` to `board_build.ldscript = STM32F407VGTX_BL32K_FLASH.ld`
and replaced `-D VECT_TAB_OFFSET=0x8000` with `-D HAS_BOOTLOADER`

I have also set  `-D USB_SERIAL_CDC=1`

This has resulted in me successful connecting to the board over USB serial.

gambrose, 2022-02-05:
`btt_skr_pro_v1_1_map.h` has the [RESET_PIN defined as GPIOG4](https://github.com/grblHAL/STM32F4xx/blob/6f5f5e32c573b6e08e647385da19ef3aff80bb6c/Inc/btt_skr_pro_v1_1_map.h#L135) commented as being the E0 Limit.
This is not the E0 limit pin.  That is [GPIOE15](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/manual/SKR-PRO-V1.1-Pin.pdf)
G4 is in the EXP1 socket.

The feed hold and cycle start pins are also not mapped as commented to e1 and e2 limits.

@terjeio would you accept a PR to fix these mappings to match the comments? i.e. E0, E1, and E2 pins.

terjeio, 2022-02-06:
> would you accept a PR to fix these mappings

Yes, please. I rely on community feedback for many boards since I am not able to verify the maps myself - and I am not going to buy them for verification only.

gambrose, 2022-02-06:
I have been testing out updating the mappings. Updating the reset and cycle start pins to E0 and E2 limit pins worked a treat.

But updating the feed hold pin to GPIOE10 (E1 limit) gave me a compilation error.
```
Src/driver.c:109:2: error: #error Interrupt enabled input pins must have unique pin numbers!
 #error Interrupt enabled input pins must have unique pin numbers!
  ^~~~~
*** [.pio/build/btt_skr_pro_1_1/src/driver.o] Error 1
```

The error does not give any clue as to what is using the same pin so I am a bit at a loss as to how to fix this.

This pin is mapped in `btt_skr_pro_v1_1_map.h` if N_ABC_MOTORS is greater than 0 but this is not the case in the `btt_skr_2_tmc2130` build config. I check this was not causing this issue by trying to build with this section removed.

So my questions are:
1) How can I resolve the pin conflict?
2) Is mapping reset, feed hold and cycle start to the E0,1,2 limit switches the correct thing to do if with will conflict if N_ABC_MOTORS > 0 will confit with it.
3) Being a CNC newbie is there any good references on what the reset, feed hold and cycle start pins actually do/are used for.

terjeio, 2022-02-07:
> The error does not give any clue as to what is using the same pin so I am a bit at a loss as to how to fix this.

You have to check the map file to find out, search for 10 - it is a different pin that is mapped to an interrupt interrupt enabled pin function.

> How can I resolve the pin conflict?

By not having two pins with the same pin number mapped to a pin function that require interrupts enabled. This is a limitation of the STM32 architecture. An option is to constantly poll the conflicting pin  - but I do not like that option.

> Is mapping reset, feed hold and cycle start to the E0,1,2 limit switches the correct thing to do if with will conflict if N_ABC_MOTORS > 0 will confit with it.

If this is about mapping any given pin to two functions then no, that is not the correct thing to do. 

> Being a CNC newbie is there any good references on what the reset, feed hold and cycle start pins actually do/are used for.

[Feed hold and cycle start](https://duckduckgo.com/?t=ffsb&q=cnc+feed+hold&ia=web).

> _Reset_ resets the controller in a little less hard manner than a cold start or MCU reset does. Motion, spindle and coolant is stopped on a reset - and some data might be kept intact such as position and homing status depending on what was going on when reset was asserted. grblHAL has a "softer" version than legacy grbl has that keeps more data. This is called _Stop_ and the reset input can be "wired" as either the internal reset signal or the stop signal.

gambrose, 2022-02-07:
If I am understanding you correctly you are saying that pins must be unique irrespective of the port.

So for the SKR PRO board which has the X limit switch on PB10 and the E1 limit switch on PE10. This is not supported and so one physical pin would need to remain unused.

terjeio, 2022-02-08:
> If I am understanding you correctly you are saying that pins must be unique irrespective of the port.

Yes, it is due how the MCU is made (the EXTI controller).

> So for the SKR PRO board which has the X limit switch on PB10 and the E1 limit switch on PE10. This is not supported and so one physical pin would need to remain unused.

Only one can be interrupt enabled and thus used for hard limit detection, homing is ok. Code can be added to poll the input though. I call it a design error, and the LPC176x 3D printer boads are even worse - maybe because Marlin does not support hard limits?

gambrose, 2022-02-08:
I have created #61 which maps the control signals to pins in EXP2 port.

These are not so easy to jumper to the ground as the limit pins so may require [the control signals to be inverted](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings#input-controls) to run.

CobraPi, 2022-03-03:
Are you using the SD card firmware upload method? I'm using an STlink. Was able to compile and flash grbl-Hal, but serial not detected on computer. Am I missing a step?

gambrose, 2022-03-03:
Yes, I used the SD card to upload the firmware.

In addition to the changes to get the firmware offset right for the SD upload, I also had to define `USB_SERIAL_CDC` to make serial over USB work.

CobraPi, 2022-03-03:
Ahh I see. I'm assuming the offset is different when using an stlink? I'm trying to figure out if my problem is related to the serial communication or the flashing process. I've already included that USB_SERIAL_CDC parameter in my testing.

terjeio, 2022-03-03:
> Ahh I see. I'm assuming the offset is different when using an stlink?

Yes, when flashing with ST-link you are overwriting the bootloader and there is no offset.

> I'm trying to figure out if my problem is related to the serial communication or the flashing process.

If you are flashing with the wrong offset the code will not run. Are you compiling with STM32CubeIDE? If so you have to select the correct build from the build tool (the hammer icon) - "Release F407 8Mhz" (or the corresponding debug build).

CobraPi, 2022-03-03:
I actually used platformio to flash the board with the stlink. Everything compiled and uploaded successfully. 

Is there some feature that requires the use of the STM32CubeIDE?

Joey-Orduna, 2022-03-22:
Hello again. I've since switched back to the SD card upload method but still can't get it to  communicate over USB. It compiles and I've verified that `firmware.bin` was uploaded correctly. I've followed your steps to the T but am still having trouble. I've also compiled it on the STM32CubeIDE and get the same results... which leaves me to believe that I'm missing a step. This is how the environment is setup in my .ini file:
```
[env:btt_skr_pro_1_1]
board = genericSTM32F407VGT6
board_build.ldscript = STM32F407VGTX_BL32K_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_BTT_SKR_PRO_1_2=
  # 8MHz crystal
  -D HSE_VALUE=8000000
  -D HAS_BOOTLOADER
  -D USB_SERIAL_CDC=1
lib_deps = ${common.lib_deps}
  eeprom
lib_extra_dirs = ${common.lib_extra_dirs}
```

@gambrose, I've checked out your fork and don't see any differences with our setups. Is there anything else I'm missing?

CobraPi, 2022-03-22:
Hello again. I've since switched back to the SD card upload method but still can't get it to  communicate over USB. It compiles and I've verified that `firmware.bin` was uploaded correctly. I've followed your steps to the T but am still having trouble. I've also compiled it on the STM32CubeIDE and get the same results... which leaves me to believe that I'm missing a step. This is how the environment is setup in my .ini file:
```
[env:btt_skr_pro_1_1]
board = genericSTM32F407VGT6
board_build.ldscript = STM32F407VGTX_BL32K_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_BTT_SKR_PRO_1_2=
  # 8MHz crystal
  -D HSE_VALUE=8000000
  -D HAS_BOOTLOADER
  -D USB_SERIAL_CDC=1
lib_deps = ${common.lib_deps}
  eeprom
lib_extra_dirs = ${common.lib_extra_dirs}
```

@gambrose, I've checked out your fork and don't see any differences with our setups. Is there anything else I'm missing?

gambrose, 2022-03-22:
@CobraPi 
As you have previously been flashing the board using an stink you may have wiped the standard bootloader.

CobraPi, 2022-03-22:
@gambrose I've found the original bootloader online and have flashed it with the STLink. I've verified it by flashing the Klipper firmware to the board via the SD card with no issues. Were there any other changes you've made to the driver file or map files?

gambrose, 2022-03-22:
@CobraPi I'm afraid not, you look to have all the changes I made.

CobraPi, 2022-03-22:
@gambrose okay, I'll keep troubleshooting in that case. Is there any difference with how you connect to the serial interface as compared to say an Arduino?

How do you connect to the board?

terjeio, 2022-03-22:
>  Is there any difference with how you connect to the serial interface as compared to say an Arduino?

No, the board should show up as com port just like an Arduino board.

Are you on Windows? If so is there a difference in available com ports when you connect the board (check in the Device Manager)? If not then either add some code, [such as this](https://github.com/grblHAL/Templates/blob/master/my_plugin/stm32F411blinky/my_plugin.c), to see if the board is alive or use Eclipse/ST-Link to debug.

CobraPi, 2022-03-24:
I was missing the usb jumpers... such a silly mistake. I'm really enjoying the extended functionality! Was able to get it to work with the ST-link as well. Thank you guys for your help.

Joey-Orduna, 2022-04-12:
I'm getting a strange error after uploading the firmware to the board. I'm using platformio and the SD card upload method. I've also tried using an ST link but get the same error. The strange thing was that it was working before. This is the error I get:
```
GrblHAL 1.1f ['$' or '$HELP' for help]
ALARM:16
[MSG:Fatal: Incompatible driver (4)]
```

What is the source of this error? Could this be caused by a hardware problem?

terjeio, 2022-04-12:
> What is the source of this error?

My best guess is you have updated to a later version without [pulling in the submodules](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL#using-git-command-line).

closed 2022-09-27 by terjeio