grblHAL/STM32F4xx #63: No comms from USB port to host using default program load

itchytweed, 2022-02-26:
Software: This repository - cloned locally.
Board: SKR-PRO-V1.2
my_machine.h: #define BOARD_BTT_SKR_PRO_1_2 // F407 based 3D Printer board
No other changes. 

Compiles using STM32CubeIDE using "Release F407 8MHz" as the rock on the board show 8 MHz on its tag.
Firmware loads from sdcard under "firmware.bin". Reboot board.

No communications out of USB port. Win10 does not create a comm port instance in Device Manager. Reloading Marlin for the board has comms out the USB port so the unit is not bricked.

Checking software code, it appears that comms is being routed to PD8/MCU_TX3 and PD9/MCU_RX3 which goes to connector block UART3 on the board. Is this the normal configuration for this package configuration? The main USB port is located on PA12/MCU_DP+ and PA11/MCU_DM-.

Thank you in advance.

terjeio, 2022-02-26:
You have to build the _Release F407 8MHz 32K Bootloader_ version when uploading via a SD card.

> Checking software code, it appears that comms is being routed to PD8/MCU_TX3 and PD9/MCU_RX3 which goes to connector block UART3 on the board.

Where did you find that code?

AFAIKT he USB port is routed [here](https://github.com/grblHAL/STM32F4xx/blob/1c8cfaccd3cb45752a5233028086adab79f55178/USB_DEVICE/Target/usbd_conf.c#L77-L87).

itchytweed, 2022-02-26:
Terje - 

Thank you for the prompt reply. 

First off, thank you for the pointer in the right direction of using the 32K bootloader option. I have updated my notebook with that change. I do have the unit talking over the normal USB port but Win10 has given me some fits about not liking the port descriptor. That took a computer reboot to bring it back to its senses so I do not believe there to be an error in the software or the board.  

I am currently walking in the changes for my particular situation.  I will have to go back through my notes to find what led me to the PD8/PD9 conclusion and I am leaning to some confusion on my part. 

Kind Regards,

Richard

itchytweed, 2022-02-28:
Terje,

Found the mistake. I got two flows crossed on paper. All is working fine right now. Thank you for the assistance

closed 2022-02-28 by itchytweed