grblHAL/STM32F4xx #64: Enabling modbus gives hardfault

andrewmarles, 2022-03-18:
I am still trying to drill down into this, but wanted to see if anyone else has seen this.  I get a hard fault when I enable modbus with STM32F446.  This is a custom board, and the RS485 is USART3 (PORTC, pins 5 and 10).  I have confirmed that I can successfully use this port and pins with both the main system uart and also with the MPG plugin.  Normally the board is using USB_CDC for sender comms.

But when I enable the modbus plugin I get a hard fault as soon as the system tries to output anything to the main stream.  Debugging it really looks to me like there is no pointer assigned to the output stream, but this might be a red herring.  Any insights or suggestions would be greatly appreciated.

terjeio, 2022-03-18:
> This is a custom board, ...

Did you modify the original _serial.c_ or do you have a board specific version for USART3? Link to code?

andrewmarles, 2022-03-18:
At this time I have just modified serial.c to try to get stuff working.  I did have to add some additional init code generated from stmcube to get USART3 to send bytes.  My testing tree is here:
[https://github.com/Expatria-Technologies/STM32F4xx/tree/F446_Flexi_Dev](https://github.com/Expatria-Technologies/STM32F4xx/tree/F446_Flexi_Dev)

terjeio, 2022-03-20:
There is a a bug here:

https://github.com/grblHAL/Plugins_spindle/blob/ebb158fa23d5619c43960c6338b005af65d5dc97/modbus.c#L489-L507

The correct code is:

```

    if(sstream->type == StreamType_Serial && sstream->flags.modbus_ready && !sstream->flags.claimed) {
#endif
        if((claimed = sstream->claim(baud[DEFAULT_BAUDRATE])) && stream_is_valid(claimed)) {

            claimed->set_enqueue_rt_handler(stream_buffer_all);

            stream.set_baud_rate = claimed->set_baud_rate;
            stream.get_tx_buffer_count = claimed->get_tx_buffer_count;
            stream.get_rx_buffer_count = claimed->get_rx_buffer_count;
            stream.write = claimed->write_n;
            stream.read = claimed->read;
            stream.flush_tx_buffer = claimed->reset_write_buffer;
            stream.flush_rx_buffer = claimed->reset_read_buffer;
#if MODBUS_ENABLE == 2
            stream.set_direction = modbus_set_direction;
#endif
            if(hal.periph_port.set_pin_description) {
                hal.periph_port.set_pin_description(Output_TX, (pin_group_t)(PinGroup_UART + claimed->instance), "Modbus");
                hal.periph_port.set_pin_description(Input_RX, (pin_group_t)(PinGroup_UART + claimed->instance), "Modbus");
            }
        } else
            claimed = NULL;
    }
```

You may force selection of the correct port by defining MODBUS_SERIAL_PORT to the port instance number, if not defined the first free usable port will be used. Add `#define MODBUS_SERIAL_PORT 1` to _my_machine.h_ to force it.

I do not have hardware to test USART3/STM32F446 but have tested with the F756 driver using USART6 mapped to PC6/7 for Modbus and USB CDC for the main channel.

andrewmarles, 2022-03-20:
No problem, will test and report back.

andrewmarles, 2022-03-21:
Ok, so the above didn't fix my problem, in the end it was unrelated to Modbus.  What was happening was that I was setting SPINDLE_RPM_CONTROLLED in my_machine.h.  I had been doing this in the IMX driver and so just copied over to the STM32 driver, but it looks like for some reason the defines are not propagating completely - I noticed that SPINDLE_PWM_DIRECT was not being evaluated correctly at a few points in stepper.c.  I think that this was corrupting the function pointers so that when the system tried to write to the output it faulted.

I did try putting SPINDLE_RPM_CONTROLLED into platformio.ini, but this didn't work either, so I will just make sure to set it in config.h. When I set it in config.h the hard fault is resolved.

Thank you so much for your help on this one.

closed 2022-03-21 by andrewmarles