grblHAL/STM32F4xx #65: F411 Blackpill Homing

vvv1954, 2022-03-30:
I have limit switches for 3 axes and they are working well as hard limits. But when I am trying to make homing it just starts with Z-axis, moves for a little (DRO shows DEFAULT_HOMING_PULLOFF value) and stops with ALARM:8 and error:9. 
I tryed homing with NO and NC switches (changing the wiring and settings) - the same results. When I am pressing a switch during the movement, the state of corresponding pin is changed but the controller does not feel this action.

vvv1954, 2022-03-30:
A couple of pictures
Initial state
[image]
After homing
[image]

terjeio, 2022-03-30:
I have checked with a F411 Blackpill and that works as well.
If you set $45 to 0 and $44 to 1 to enable homing for X-axis only does that work?

vvv1954, 2022-03-30:
No, it is not working.
What kind of F411 Blackpill do you have, is it original from WeAct? I have this clon of WeAct from Ali
https://www.aliexpress.com/item/1005001621784672.html?spm=a2g0o.order_list.0.0.21ef18024mG9wb

vvv1954, 2022-03-30:
I have this wiring that works with STM32F407ZGT6.  
[image]

terjeio, 2022-03-31:
> What kind of F411 Blackpill do you have, is it original from WeAct? 

I believe it is a clone as it is unbranded. The link I used to buy it from is dead.

> I have this wiring that works with STM32F407ZGT6.

I would move the 3K3 resistor to the collector side. Or at least disable the MCU pullup resistors with $18 as the 3K3 resistor and the pullup will form a voltage divider when the transistor is off...

Since the codebase is shared by the F407 and the F411 I believe your issue is either electrical or related to the configuration.

vvv1954, 2022-03-31:
I have changed the wiring as you recommended but it also does not work.  Now I want to check where is the problem: my setup or some defect of MSU. Could you attach please your firmware, wiring and $$ settings for F411 Blackpill?

terjeio, 2022-04-01:
[GRBL Driver STM32F401.zip](https://github.com/grblHAL/STM32F4xx/files/8394956/GRBL.Driver.STM32F401.zip)

Wiring is direct to my CNC machine simulator. Note that the binary is a test build that has a lot of changes to spindle handling so may no be used for production.

vvv1954, 2022-04-01:
Many thanks.
As far as I understood from settings, your wiring is something like this?
[image]

vvv1954, 2022-04-01:
I checked homing with your firmware. It works perfectly well with both wiring above.
[image]

And I understood what was wrong. I re-assigned limits pins to B3, B4, and B5. But for some reasons the homing did not want to work with these pins. So I returned limit pins to their original assignments (B12, B13, B14) and now there is no problem with homing.

Thank you very much for your help and for excellent codes!

terjeio, 2022-04-02:
You have to change the LIMIT_INMODE to GPIO_SHIFT3 to use B3-5:

```
// Define homing/hard limit switch input pins.
#define LIMIT_PORT              GPIOB
#define X_LIMIT_PIN             3
#define Y_LIMIT_PIN             4
#define Z_LIMIT_PIN             5
#define LIMIT_INMODE            GPIO_SHIFT3
```

When I did that homing works for those pins as well.

vvv1954, 2022-04-02:
My fault, I did not pay attention to shift. Thank you, now it works.

vvv1954, 2022-04-02:
Last question (I hope). How to show work and machine coordinates during homing?

terjeio, 2022-04-02:
> How to show work and machine coordinates during homing?

Not possible without [changes to the core and drivers](https://github.com/grblHAL/core/blob/dae7d56ab5ed6fcaa787aa181ef1ae331a03c9b1/limits.c#L228-L233). Limit switches are polled during homing, and only the abort realtime command is responded to to keep switch response time low.

vvv1954, 2022-04-09:
I made a PCB for my CNC 3018, works perfectly well, thank you for excellent codes.
[image]

closed 2023-08-26 by terjeio

BasisSun, 2024-07-08:
> You have to change the LIMIT_INMODE to GPIO_SHIFT3 to use B3-5:
>
> ```
> // Define homing/hard limit switch input pins.
> #define LIMIT_PORT              GPIOB
> #define X_LIMIT_PIN             3
> #define [...]

What exactly does GPIO_SHIFT3 stand for?

terjeio, 2024-07-08:
> What exactly does GPIO_SHIFT3 stand for?

Shift the contents read from the GPIO input register right 3 bits before storing the result. Requires the bits in the GPIO register to be consecutive and in the axis order. [Other options](https://github.com/grblHAL/STM32F4xx/blob/7d0132aeecd24ab9ae9633db3cd79fbe99bd3576/Inc/driver.h#L97-L112).