grblHAL/STM32F4xx #66: using pb9 as pwm

customsolutions, 2022-04-05:
Do to design requirements I need to but am having difficulties reassigning pb9 working.... I did it on a blue pill. I created my_machine_map....reassigned pins as needed and all worked for not Pwm pins, and have been trying to figure out where or what needs to be changed for timers etc...... on the blue pill I had to go into drivers.h and driver.c  and change various items........ These same files in the stm32f4 are organized a bit diff.... any assistance would be appreciated

terjeio, 2022-04-15:
Timer assignment is done [here](https://github.com/grblHAL/STM32F4xx/blob/2f2bad61c8df2575461c8716f2e5c46b16ed5cf1/Inc/driver.h#L162-L223), PB9 is currently not supported so has to be added (if possible),

customsolutions, 2022-04-15:
Thank you for your response. I know it is possible, and i did make changes
there. However it apeared that other supporting code in other files needed
to be changed. For example on the stm32f103(blue pill) the code was a
little more obvious and i changed it and got it working.

Thanks
Jason Moore

terjeio, 2022-04-15:
Adding this code to the definitions:

```
  #elif SPINDLE_PWM_PIN == 9 // PB9 - TIM11_CH1
    #define SPINDLE_PWM_TIMER_N     11
    #define SPINDLE_PWM_TIMER_CH    1
    #define SPINDLE_PWM_TIMER_INV   0
    #define SPINDLE_PWM_TIMER_AF    3
  #endif
```
and enabling timer 11 clock [here](https://github.com/grblHAL/STM32F4xx/blob/2f2bad61c8df2575461c8716f2e5c46b16ed5cf1/Src/driver.c#L1742) works for me.

customsolutions, 2022-04-15:
Maybe i did it wrong.... ill give your code a go....
Thanks again
Jason Moore

customsolutions, 2022-04-16:
Well I took a look at my code and I was for some reason using the incorrect
timer, hower while your code builds, i seem to have lost any output on
gpiob... Starting to suspect my board, however as of not the only things
connected to my board is leds and ressiters so I can verify function.... I
even completely erased and reloaded my project in cube redid the mods.....
no dice..... Btw I am connecting to light burn to verify movements and
laser on/off and pwm.... Any other ideas?

Thanks again
Jason Moore

terjeio, 2022-04-16:
All pins on port B are dead?
When I tested I initially got 3.3V on PB9 due to PB9 beeing configured as an aux input later in the setup sequence...

BTW I have just commited an update where option for PB9 for PWM has been added.

closed 2023-08-26 by terjeio