grblHAL/STM32F4xx #68: Is TMC2209 enabled?

mink007, 2022-04-12:
I was going through the code and read somewhere (in my_machine.h) that it is work in progress. Also generic_map.h, i see that if TRINAMIC_ENABLE is present then error will be generated.

#if TRINAMIC_ENABLE
#error "Trinamic plugin not supported!"
#endif

I am trying to compile grbl with STM32CubeIDE for STM32F407. Since my board is using TMC2209 so i wanted to know if the code is ready for TMC2209 in UART mode?

I also wanted to know is there a pin configuration file (like generic_map.h) where I can see which pin/port is assigned to which name.

Thank you

terjeio, 2022-04-12:
> I am trying to compile grbl with STM32CubeIDE for STM32F407. Since my board is using TMC2209 so i wanted to know if the code is ready for TMC2209 in UART mode?

It is but not for all boards supported.

> I also wanted to know is there a pin configuration file (like generic_map.h) where I can see which pin/port is assigned to which name.

There are many in the _Inc_ folder with names ending in __map.h_, which  to use is configured in [my_machine.h](https://github.com/grblHAL/STM32F4xx/blob/2f2bad61c8df2575461c8716f2e5c46b16ed5cf1/Inc/my_machine.h#L22-L36).

mink007, 2022-04-13:
Thank you for your response.

Is TMC2209 UART mode enabled for STM32F407 based boards?

terjeio, 2022-04-13:
There is support in the driver for both soft and hard UART modes, if the board in question supports TMC2209 drivers either there is support or it may easily be added. Which board do you have?

mink007, 2022-04-13:
I have mks eagle

terjeio, 2022-04-14:
Ok, then a board map has to be created - you can do it?
Easiest is to make a copy of _btt_skr_pro_v1_1_map.h_ and name it _my_machine_map.h_, then modify that to [match the board](https://github.com/makerbase-mks/MKS-EAGLE/tree/main/hardware/MKS%20Eagle%20V1.0_001).
Enabling the use of  _my_machine_map.h_ is done by uncommenting [this line](https://github.com/grblHAL/STM32F4xx/blob/2f2bad61c8df2575461c8716f2e5c46b16ed5cf1/Inc/my_machine.h#L36).

When you have a working board map it can then be added with a proper name (after a few additions to _my_machine.h_ and _driver.h_), either via a PR or by posting the map file in a comment.

mink007, 2022-04-14:
Hi Terjeio,
   I am specifically looking to integrate TMC2209 into eagle board, that way I can drive the stepper motors. I don't know much about grbl though.

My initial  goal is to get the code talk to TMC2209 driver chip though initialization and reading the registers & sending commands to it. 

Once I reach that point, i will share the map file here. 
On a side note, is there some command(s) that I can use in "grbl_enter()" function that initiates TMC2209? Because I see that when I enter in debug mode (in STM32CubeIDE after compiling) the code enters protocol loop and from there I cannot call TMC2209. May be it is intended behavior of grbl code. I am asking just from the standalone testing of TMC2209 driver communication perspective.

Thank you.