grblHAL/STM32F4xx #69: STM32F401RCT6 - 6 Axis Configuration - not showing Axis A, B or C

gxdeange, 2022-04-13:
Hi terjeio,

I hope you are well.

I have designed and built a 6 Axis board using the STM32F401RCT6 MCU and have modified the firmware for M3, M4 and M5 [no ganged outputs at this point in time] which has compiled without error.

When I launch IOSender I can only see 3 Axis and was wondering if there is any limitation on either IOSender or the core firmware to allow the 6 independent axis.

Cheers and have a great day
George

andrewmarles, 2022-04-14:
Did you change N_AXIS in config.h?

gxdeange, 2022-04-14:
Hi andrewmarles,

I had not thought of that but I just changed it to 6 and I am getting a 'This configuration is not supported'. I suspect that because I have built a custom board it is not linked with other boards and is restricted from building out 6 independant axis.

CHeers
George

gxdeange, 2022-04-14:
I have commented out the configuration error lines but am now getting this error:

**#error Interrupt enabled input pins must have unique pin numbers!**

I am hoping there is a way to not be restricted to specific boards

andrewmarles, 2022-04-14:
It's not about the board.  It is an STM32 thing, there are only 16 interrupts and they each need a unique number.  IE PA2 and PB2 cannot BOTH be interrupt pins.

gxdeange, 2022-04-14:
Thanks for the update.

I am not very good at this but does this mean that an input pin (such as limit, probe, hold) could be on any port but must only use 0 through to 15 once (i.e. PC0, PB1, PC2, PC3, PA4, etc)?

Cheers
George

gxdeange, 2022-04-14:
Would you know of any specific logic rules I need to follow when assigning pins to functions (such as limit pins)?

Any advice would be greatly appreciated.

Cheers
George

andrewmarles, 2022-04-14:
> Thanks for the update.
> 
> I am not very good at this but does this mean that an input pin (such as limit, probe, hold) could be on any port but must only use 0 through to 15 once (i.e. PC0, PB1, PC2, PC3, PA4, etc)?
> 
> Cheers George

Yes, like this.

terjeio, 2022-04-14:
For this MCU (in fact all supported STM32 MCUs) a particular rule is

> only use 0 through to 15 once (i.e. PC0, PB1, PC2, PC3, PA4, etc)

for input pins (control & limits). Probe input is polled and can be on any pin. There is support for polling the Z limit input, but best to avoid that.

UART pins (for serial ports) are also limited, both by code and by the MCU. Same for the spindle PWM output.

FYI if you have M3-M5 defined in the map file ganged axes can be configured, and if these motors has a corresponding limit input auto squaring is also possible.

gxdeange, 2022-04-14:
Awesome and thanks for the feedback. I have tried a few variations within the IDE and the inital error is no longer present but I received this error:

invalid type argument of '->' (have 'uint32_t' {aka 'volatile long unsigned int'})	driver.h	/GRBL Driver STM32F401/Inc	line 46	C/C++ Problem

Cheers
George

terjeio, 2022-04-14:
> invalid type argument of '->' (have 'uint32_t' {aka 'volatile long unsigned int'}) driver.h /GRBL Driver STM32F401/Inc line 46 C/C++ Problem

This is likely due to a badly defined mapping in your board map file.

gxdeange, 2022-04-14:
OK Thanks.

I will try a few different variations.

Cheers
George

gxdeange, 2022-04-14:
Just quickly, if I want to run these as independant Axis (A, B & C), do they still need to be defined as M3, M4 & M5 respectively?

Cheers
G

terjeio, 2022-04-14:
> do they still need to be defined as M3, M4 & M5 respectively?

Yes.

gxdeange, 2022-04-14:
OK. 

I found that if I uncomment the ganged and auto square for an axis, the previous error dows not appear. I guess I am doing something wrong but I cannot seem to resolve the error if I want to run fully indepandant axis.

Cheers
George

terjeio, 2022-04-14:
Post your map file and I'll take a look.

gxdeange, 2022-04-14:
#define X_STEP_PORT				GPIOA
#define X_STEP_PIN              1
#define Y_STEP_PORT				GPIOD
#define Y_STEP_PIN              2
#define Z_STEP_PORT				GPIOA
#define Z_STEP_PIN              12
#define M3_STEP_PORT            GPIOC
#define M3_STEP_PIN             3
#define M4_STEP_PORT            GPIOC
#define M4_STEP_PIN             5
#define M5_STEP_PORT            GPIOC
#define M5_STEP_PIN             0
#define STEP_OUTMODE            GPIO_BITBAND
//#define STEP_PINMODE            PINMODE_OD // Uncomment for open drain outputs

// Define step direction output pins.
//#define DIRECTION_PORT          GPIOB
#define X_DIRECTION_PORT		GPIOA
#define X_DIRECTION_PIN         4
#define Y_DIRECTION_PORT		GPIOB
#define Y_DIRECTION_PIN         13
#define Z_DIRECTION_PORT		GPIOA
#define Z_DIRECTION_PIN         11
#define M3_DIRECTION_PORT       GPIOC
#define M3_DIRECTION_PIN        9
#define M4_DIRECTION_PORT       GPIOH
#define M4_DIRECTION_PIN        1
#define M5_DIRECTION_PORT       GPIOB
#define M5_DIRECTION_PIN        15
#define DIRECTION_OUTMODE       GPIO_BITBAND
//#define DIRECTION_PINMODE       PINMODE_OD // Uncomment for open drain outputs

// Define stepper driver enable/disable output pin.
//#define STEPPERS_ENABLE_PINMODE PINMODE_OD // Uncomment for open drain outputs

#define X_ENABLE_PORT				GPIOH
#define X_ENABLE_PIN                0
#define Y_ENABLE_PORT               GPIOB
#define Y_ENABLE_PIN                14
#define Z_ENABLE_PORT               GPIOC
#define Z_ENABLE_PIN                1
#define M3_ENABLE_PORT				GPIOC
#define M3_ENABLE_PIN				8
#define M4_ENABLE_PORT				GPIOC
#define M4_ENABLE_PIN				4
#define M5_ENABLE_PORT				GPIOB
#define M5_ENABLE_PIN				8
//#define ENABLE_OUTMODE				GPIO_BITBAND

// Define homing/hard limit switch input pins.
#define X_LIMIT_PORT			GPIOB
#define X_LIMIT_PIN             7
#define Y_LIMIT_PORT			GPIOB
#define Y_LIMIT_PIN             1
#define Z_LIMIT_PORT			GPIOB
#define Z_LIMIT_PIN             0
#define M3_LIMIT_PORT           GPIOB
#define M3_LIMIT_PIN            12
#define M4_LIMIT_PORT           GPIOB
#define M4_LIMIT_PIN            10
#define M5_LIMIT_PORT           GPIOB
#define M5_LIMIT_PIN            9
#define LIMIT_INMODE            GPIO_BITBAND

#define M3_AVAILABLE
#define M4_AVAILABLE
#define M5_AVAILABLE

// Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_PORT     GPIOA
#define SPINDLE_ENABLE_PIN      15
#define SPINDLE_DIRECTION_PORT  GPIOC
#define SPINDLE_DIRECTION_PIN   10

// Define spindle PWM output pin.
#define SPINDLE_PWM_PORT_BASE   GPIOA_BASE
#define SPINDLE_PWM_PIN         8

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_PORT      GPIOC
#define COOLANT_FLOOD_PIN       12
#define COOLANT_MIST_PORT       GPIOC
#define COOLANT_MIST_PIN        11

// Define user-control controls (cycle start, reset, feed hold) input pins.
//#define CONTROL_PORT            GPIOB
#define RESET_PORT				GPIOB
#define RESET_PIN               2
#define FEED_HOLD_PORT			GPIOB
#define FEED_HOLD_PIN           3
#define CYCLE_START_PORT		GPIOB
#define CYCLE_START_PIN         4
//#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PORT		GPIOB
#define SAFETY_DOOR_PIN         5
//#endif
#define CONTROL_INMODE			GPIO_BITBAND

// Define probe switch input pin.
#define PROBE_PORT              GPIOB
#define PROBE_PIN               6
//#if I2C_STROBE_ENABLE
//#define I2C_STROBE_PORT         GPIOB
//#define I2C_STROBE_PIN          9
//#endif

#if SDCARD_ENABLE
#define SD_CS_PORT                  GPIOC
#define SD_CS_PIN                   2
#define SPI_PORT                    1 // GPIOA, SCK_PIN = 5, MISO_PIN = 6, MOSI_PIN = 7
#endif

#define HAS_IOPORTS
// AUX Output ports
#define AUXOUTPUT0_PORT			GPIOC
#define AUXOUTPUT0_PIN			6
#define AUXOUTPUT1_PORT			GPIOC
#define AUXOUTPUT1_PIN			7
//#define AUX_OUTMODE				GPIO_BITBAND
// AUX Input ports
#define AUXINPUT0_PORT			GPIOC
#define AUXINPUT0_PIN			15
#define AUXINPUT1_PORT			GPIOC
#define AUXINPUT1_PIN			14
#define AUXINPUT2_PORT			GPIOC
#define AUXINPUT2_PIN			13

terjeio, 2022-04-14:
You should add definitions for M3-M5 in the same way as it is done for the other drivers such as this:

https://github.com/grblHAL/STM32F4xx/blob/2f2bad61c8df2575461c8716f2e5c46b16ed5cf1/Inc/btt_skr_pro_v1_1_map.h#L80-L117

This since there is some preprocessor magic going on behind the scenes to create the definitions suitable for _driver.c_ use.

gxdeange, 2022-04-14:
OK. Map File has been updated but the same above error exists.

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_STEP_PORT            GPIOC
#define M3_STEP_PIN             3
#define M3_DIRECTION_PORT       GPIOC
#define M3_DIRECTION_PIN        9
#define M3_ENABLE_PORT			GPIOC
#define M3_ENABLE_PIN			8
#define M3_LIMIT_PORT           GPIOB
#define M3_LIMIT_PIN            12
#endif

// Define ganged axis or B axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 1
#define M4_AVAILABLE
#define M4_STEP_PORT            GPIOC
#define M4_STEP_PIN             5
#define M4_DIRECTION_PORT       GPIOH
#define M4_DIRECTION_PIN        1
#define M4_ENABLE_PORT			GPIOC
#define M4_ENABLE_PIN			4
#define M4_LIMIT_PORT           GPIOB
#define M4_LIMIT_PIN            10
#endif

// Define ganged axis or C axis step pulse and step direction output pins.
#if N_ABC_MOTORS == 3
#define M5_AVAILABLE
#define M5_STEP_PORT            GPIOC
#define M5_STEP_PIN             0
#define M5_DIRECTION_PORT       GPIOB
#define M5_DIRECTION_PIN        15
#define M5_ENABLE_PORT			GPIOB
#define M5_ENABLE_PIN			8
#define M5_LIMIT_PORT           GPIOB
#define M5_LIMIT_PIN            9
#endif

gxdeange, 2022-04-14:
I just changed the 'N_AXIS 6' to 'N_AXIS 5' and the firware compiled without error but M3 and M4 were greyed out

terjeio, 2022-04-14:
There is a bug in _driver.c_ popping up when C axis is defined:

https://github.com/grblHAL/STM32F4xx/blob/2f2bad61c8df2575461c8716f2e5c46b16ed5cf1/Src/driver.c#L577

This is the correct code:

`    DIGITAL_OUT(C_STEP_PORT, C_STEP_PIN, step_outbits.c);`

Sorry about that.

> ... but M3 and M4 were greyed out

Sometimes (often?) you have to rebuild the index after changing the configuration, right click on the project and select _Index > Rebuild_ from the menu.

gxdeange, 2022-04-14:
Legand: Compiles without error.

I am potentially looking at commercialising this board and was wondering if it was possible (through spare pins on the MCU) to switch between 3, 4, 5, 6 Axis and / or converting normal axis to ganged axis without having to reflash the firmware.

With 3 spare pins it could be configured up to 8 different ways.

terjeio, 2022-04-14:
> I am potentially looking at commercialising this board and was wondering if it was possible (through spare pins on the MCU) to switch between 3, 4, 5, 6 Axis and / or converting normal axis to ganged axis without having to reflash the firmware.

No, not without a major rewrite of the core and all drivers.

gxdeange, 2022-04-14:
OK.

Thanks again. You have been a great help.

Cheers and have a great weekend
George

closed 2022-04-14 by gxdeange

hankecnc, 2022-06-16:
Oops, it's a bit late to see this question, I'm also making a grblhal board based on F401RCT6 or F402 for sale, because this is the only cheapest stm32 chip I can buy during the epidemic (of course there are many arm chips made in China, but