grblHAL/STM32F4xx #70: WebUI through ESP-01S on SKR Pro 1.2

olijouve, 2022-04-27:
Is there any change to operate on STM32F4XX driver code or in config to be able to use the WebUI on a SKR Pro 1.2 with ESP-01S module ?

terjeio, 2022-04-28:
This is not possible without modifying code as the [current backend](https://github.com/grblHAL/Plugin_WebUI) relies on direct access to Grbl data and a networking stack. You would have to split the backend in two parts - one running on the ESP-01S and one on the grblHAL controller? Or is there a backend available that communicates with the controller over serial comms only?

olijouve, 2022-04-28:
ESP3D sends gcode commands over serial comms.
On SKR Pro 1.2 there is  3( or 4?) available serial ports: USB, TFT screen, Wifi and maybe UART as a 4th one.

On marlin i need to enable this:
#define SERIAL_PORT_3 6

6 is meant for Wifi / Esp01-S module

olijouve, 2022-04-29:
In Marlin serial Pin allocations for the SKR pro are defined here:
https://github.com/MarlinFirmware/Marlin/blob/a185ce22cf6e4fb15250815c5c39318606a7e65a/buildroot/share/PlatformIO/variants/MARLIN_BIGTREE_SKR_PRO_11/variant.h

I saw STM32F3xx and STM32F1xx seems to have support for a third serial port (#define USART_IRQHandler USART3_IRQHandler), could it be reused for STM32F4xx ?

terjeio, 2022-04-30:
> I saw STM32F3xx and STM32F1xx seems to have support for a third serial port (#define USART_IRQHandler USART3_IRQHandler), could it be reused for STM32F4xx ?

You need a third serial port or want to assign the the first grblHAL port to USART3?
For the latter I recently did it like this for the LPC1769 driver:

https://github.com/grblHAL/LPC176x/blob/fc790f022d4cc4624d94fb3718e3bfa8df843372/src/btt_skr_1.4_turbo_map.h#L41-L43
and
https://github.com/grblHAL/LPC176x/blob/fc790f022d4cc4624d94fb3718e3bfa8df843372/src/serial.c#L36-L60

This driver can be changed in a similar way.

olijouve, 2022-05-01:
A third serial port would be better as i still need the usb port for LightBurn and also the TFT serial for a possible futur use.

terjeio, 2022-05-07:
I'll add the option of assigning the primary serial port to USART3 in the next release. Adding support for a third serial port as standard will have to wait - but can be added by yourself in a user plugin - which will be required anyway if you plan to add a TFT.

olijouve, 2022-05-07:
Thank you !
I understand for a third serial port as standard, nothing crucial here.
Saying third, does that mean you already support two serials in standard, i suppose the first one would be the USB serial port but which one is the second supported serial port ?
Could it be possible to choose between the primary or the secondary serial on USART3 assignement ?

About writing plugin, i checked at the bluetooth plugin as it seems close in term of need but unfortunately i'm not skilled enough for that :(

terjeio, 2022-05-08:
> Saying third, does that mean you already support two serials in standard

Yes, but not all drivers do. Plugin/user code can add more.

> i suppose the first one would be the USB serial port but 

No, it is not counted as a serial port (UART based). If the virtual USB serial port is used as the main communication channel then the UART ports are free for other uses such as MPG, VFD etc. I want to implement support for automatic switching between the primary UART port and the virtual USB port on USB client connect/disconnect (if possible), I'll have to dig into the lower level USB CDC implementation for that. Similar to the switching done on connect/disconnect for virtual network ports (TCP/WebSockets) - which is very convenient for determining which port controls or "owns" the machine.

> which one is the second supported serial port

Which UART? It is configuration dependent.

closed 2023-08-26 by terjeio