grblHAL/STM32F4xx #71: PPR Settings Blackpill F401

JoachimF, 2022-05-09:
Hello Terjeio,

I continue my work to create a mainboard from blackpill and I test all fonctions available on this target.
When I enable spindle_sync, all is OK, but when I set PPR to 600, because my encoder is 600PPR, the blackpill doesn't boot.
No serial port available in the Windows device manager, I have to reflash the stm to make it works again.

I set index and pulse to pins B3 and B4 to have 2 differents timers, have I made something wrong ?

JoachimF, 2022-05-10:
Unfortunately, on the F401CC,  port D is not available and the TIM3_ETR has no alternate pin. 
Should it be possible to swap RPM_TIMER and RPM_COUNTER to get the pulse on TIM2_ETR on PA0 or PA5 or PA15 ?

driver.h
`#define RPM_COUNTER_N               **2**
#define RPM_COUNTER                 timer(RPM_COUNTER_N)
#define RPM_COUNTER_IRQn            timerINT(RPM_COUNTER_N)
#define RPM_COUNTER_IRQHandler      timerHANDLER(RPM_COUNTER_N)
#define RPM_COUNTER_CLOCK_ENA       timerCLKENA(RPM_COUNTER_N)

#define RPM_TIMER_N                 **3**
#define RPM_TIMER                   timer(RPM_TIMER_N)
#define RPM_TIMER_IRQn              timerINT(RPM_TIMER_N)
#define RPM_TIMER_IRQHandler        timerHANDLER(RPM_TIMER_N)
#define RPM_TIMER_CLOCK_ENA         timerCLKENA(RPM_TIMER_N)`

driver.c
`    GPIO_Init.Mode = GPIO_MODE_AF_PP;
    GPIO_Init.Pin = SPINDLE_PULSE_BIT;
    GPIO_Init.Pull = GPIO_NOPULL;
    GPIO_Init.Speed = GPIO_SPEED_FREQ_LOW;
    GPIO_Init.Alternate = **GPIO_AF1_TIM2;**
    HAL_GPIO_Init(SPINDLE_PULSE_PORT, &GPIO_Init);
`

It is correct ?

JoachimF, 2022-05-11:
Dear Terjeio,

I know why the stm crash on start up, I enable the VFD plugin and the Spindle_sync, but when VFD is enable, the hal.spindle.set_state is not set !
Is there a reason to not set this pointer? For my project I don't need VFD and Spindle_Sync, but I we can correct this it will be fine.

#if VFD_SPINDLE != 1
....
#endif // SPINDLE_PWM_TIMER_N
            hal.spindle.set_state = spindleSetState;
#endif // VFD_SPINDLE

Best Regards,

Joachim

terjeio, 2022-05-11:
> Should it be possible to swap RPM_TIMER and RPM_COUNTER to get the pulse on TIM2_ETR on PA0 or PA5 or PA15 ?

It should be - you can try?

> The pointer is set after the call of grbl.on_spindle_select that is called after the spindle sync init.

It looks like your driver code is outdated - spindle handling was [changed recently](https://github.com/grblHAL/STM32F4xx/commit/86d992b6379987857e53a79155d483df884c7010) so perhaps ok if you update?

JoachimF, 2022-05-11:
> > Should it be possible to swap RPM_TIMER and RPM_COUNTER to get the pulse on TIM2_ETR on PA0 or PA5 or PA15 ?
> 
> It should be - you can try?

I'll try it and I'll make you a feedback, it would be a nice option to have for the low pins MCUs
> It looks like your driver code is outdated - spindle handling was [changed recently](https://github.com/grblHAL/STM32F4xx/commit/86d992b6379987857e53a79155d483df884c7010) so perhaps ok if you update?

Shame on me, I debug code that you have already done... The lack of time is faulty ☺

Where can I send you a tip to thanks your amazing work?

Juank23, 2022-07-02:
good day terjeio, friend I have an emco compact 5 CNC lathe and a stm32f411 with grblHAL but I do not know how to configure the encoder I can not find where it is configured and the input pins as I do I remain attentive thanks

terjeio, 2022-07-02:
@Juank23 See the [st_morpho_map.h](https://github.com/grblHAL/STM32F4xx/blob/master/Inc/st_morpho_map.h) file, search for the SPINDLE_SYNC_ENABLE symbol. You will have to add those definitions to the map file you are using. Note that currenly only D2 is supported as input for the spindle encoder pulse input. Info in [this comment](https://github.com/grblHAL/STM32F4xx/issues/71#issuecomment-1121359326) above.

Juank23, 2022-07-02:
Good morning thanks for your prompt response, friend I added those lines to the code but there is a problem, you have to make any other modification to the code ?? in iosender when it is in lathe mode gives a problem when trying to make the threads I remain attentive thank you very much.

terjeio, 2022-07-03:
> ... but there is a problem ...

And the problem is...?

First you have to ensure that the spindle encoder input is handled correctly. When enabled the RPM display in ioSender goes "live" and displays the real RPM as read from the encoder. Does it? and does the "live" value match reality?

For spindle synced motion the index pulse is needed, you can use the `$SD` command to check. This outputs a list of comma separated values: current index pulse count, pulse count, error count and angular position. Error count is a count of misread pulses for each revolution (> 0 if pulses per revolution does not match the PPR setting - $38). The `$SR` command resets the counters and angular position.

When the encoder pulses are read correctly then tuning of the PID loop has to be done with settings $80 - $82. To aid that it is best to enable PID logging. This is a compile time option found in [config.h](https://github.com/grblHAL/core/blob/32f905abb3f95af10771398697a2b349e57e28d5/config.h#L503-L504). When enabled a new tab is presented in ioSender where the data from the log is presented graphically. To tune issue a G33 command and fetch the data afterwards. Then adjust the PID parameters and repeat until you get a good response. [Here is one how-to description of PID tuning](https://realpars.com/pid-tuning/), many more can be found by searching for [pid tuning](https://duckduckgo.com/?t=ffsb&q=pid+tuning&ia=web).

Juank23, 2022-07-04:
buen dia terjeio, amigo descargue grblhall pero no encuentro la menara de conifurar, modifique las lineas de codigo que me digiste pero no funciona no aparece ninguna pestaña de pidtuning en iosender como hago para configurarlo estoy compilando en stm32cubeIDE a continuacion te envio unas capturas espero me puedas ayudar muchas gracias 
[image]
[image]
[image]
[image]
[image]

terjeio, 2022-07-04:
> good day terjeio, friend download grblhall but I can not find the menara of conifurar, modify the lines of code that you told me but it does not work does not appear any tab of pidtuning in iosender as I do to configure it I am compiling in stm32cubeIDE then I send you some captures I hope you can help me thank you very much

If I read this translation correctly you did not manage to compile with PID logging. This is a compile time option found in [config.h](https://github.com/grblHAL/core/blob/32f905abb3f95af10771398697a2b349e57e28d5/config.h#L503-L504) - uncomment the PID_LOG definition, recompile and reflash and the PID tab should show up.

From the images above it also looks like you have not added/uncommented [this line](https://github.com/grblHAL/STM32F4xx/blob/be9965052f8ec7d0e56ccf56ca7c3c06123981cd/Inc/st_morpho_map.h#L44) in your map file?

Juank23, 2022-07-04:
Thanks for your prompt reply, I am using the blackpill_map.h map, I added the lines you told me before but I notice there are errors
here I send you a copy of the map that I am using thanks
[image]
[image]
[image]
[image]
[image]
[image]

terjeio, 2022-07-08:
The 411 Blackpill does not have the PD2 pin brought out so more code changes are needed to use [one of the alternatives](https://github.com/grblHAL/STM32F4xx/issues/71#issuecomment-1123354284). PA15 is currently not in use so that seems to be the best candidate.

Juank23, 2022-07-08:
good day terjeio, I tell you that already appears the pid tuning tab, I have changed the turret motor and I have adapted a stepper motor so that the tool change is automatic, there are 6 tools in the turret, I also added some limit switches, but I do not know how to create the macro to make the tool change from the gcode, I have done some tests with some small macros and I have added them to the gcode to see how it works and it works fine but it should be automatic, from the gcode I see a t1, t2, iosender pauses to make the tool change, but as I have not made the appropriate macro does not work then a video to see the project thanks for your help greetings
https://www.youtube.com/watch?v=5Jw0SQXnH1w

terjeio, 2022-07-09:
I have just uploaded code changes for swapping the timers, and thus using some port A pins, for spindle pulse input. Only tested with a simulator and there is a smallish risk for the interrupt driven extension of the RPM timer to 32 bits to produce invalid values. Testing in a real machine is required to validate this change. The blackpill map has been updated to use this option.

@Juank23 There is not much I can do to help without detailed information of how the turret is controlled. And please open another issue as it is not related to this.

Juank23, 2022-07-12:
good morning terjeio, I did some small tests 
and with a bad connection I burned the card, I was sad but I have already ordered 2 more cards, I am waiting for them to arrive to continue with the project, thank you very much for your help, best regards

closed 2023-08-26 by terjeio