grblHAL/STM32F4xx #74: stm32f411 with SPI and I2C

muhammadelmogy, 2022-05-30:
I am designing my own controller around stm32f411(the same one as in the blackpill) and I want to use I2C FRAM [MB85RC16PN](https://eu.mouser.com/datasheet/2/1113/MB85RC16_DS501_00001_11v0_E-2329168.pdf) or [MB85RC64VPNF](https://eu.mouser.com/datasheet/2/1113/MB85RC64V_DS501_00013_7v0_E-2329204.pdf) and also going to use Trinamic TMC5160 with SPI.
I want to assign the proper ports and pins for both I2C & SPI communication and make sure those assignments would not conflict with any other assignments in the code.
On the first sight I have chosen I2C1 (PB6/PB7) and SPI1 (PA4/PA5/PA6/PA7)
Should I change this?
How should I assign the mentioned ports and pins in the code, I know it should be in XXXX_map.h but not sure about the used definitions.(Is there a place for all default definitions used in the project?)
Also Is there any issues with using the mentioned FRAM(First one does not have hardware address pins)?

terjeio, 2022-05-31:
> On the first sight I have chosen I2C1 (PB6/PB7) and SPI1 (PA4/PA5/PA6/PA7)
Should I change this?

That depends on the other pin assignments - STM32 MCUs are a bit odd in that for a given pin number only one port can have interrupt enabled. This dictates which pins can be used for limit and system control inputs.

> Is there a place for all default definitions used in the project?

Not for pin/port assignments - the closest to a default is perhaps [generic_map.h](https://github.com/grblHAL/STM32F4xx/blob/master/Inc/generic_map.h).
The core defaults can be overidden in [grbl/config.h](https://github.com/grblHAL/core/blob/6d00f743f09db2a07adadf4dc996e3d55de37cdb/config.h) - many of these can be set at run time with setting commands.

> Also Is there any issues with using the mentioned FRAM(First one does not have hardware address pins)?

I guess these will be ok with the standard [EEPROM plugins](https://github.com/grblHAL/Plugin_EEPROM). The address is in a #define statement so can be easily changed.

terjeio, 2022-05-31:
You can have (GPIO) inputs on whatever port you like, but two inputs should not share the same pin number. E.g. having inputs on A.7 and B.7 will only allow one of them to have an interrupt assigned. Input pins routed to other peripherals like I2C or SPI uses a different routing for interrupts so does not block the pin number (for GPIO interrupts).

closed 2023-08-26 by terjeio