grblHAL/STM32F4xx #75: Stm32 f411(blackbill) for XYYZA config

ibrahimballoul, 2022-06-01:
Hi am planning to build a XYYZ with a rotary axis with the stm32 f411 (blackbill), can i make this config with the blackbill? Plus can i use auto squaring for the Y axis with this config?....

terjeio, 2022-06-01:
[BOARD_CNC3040](https://github.com/grblHAL/STM32F4xx/blob/master/Inc/cnc3040_map.h) supports up to 5 motors but does not have the required input for the second Y limit. If there are no free pins you can likely use one of the other assigned input or output pins that you do not use for the intended purpose (e.g the SAFETY_DOOR_PIN) as the limit input - see t[his driver](https://github.com/grblHAL/STM32F4xx/blob/master/Inc/st_morpho_map.h) for how it is done.

ibrahimballoul, 2022-06-01:
Thank you for the fast replay....okay so i will assign the second Y limit to the Safety door pin i will not be using it..i ordered a couple of blackpills as soon as they arrive  i will test the build and close the issue.

closed 2023-04-10 by ibrahimballoul