grblHAL/STM32F4xx #76: PC interface

osmargarnica, 2022-06-06:
Hello Guys.
Working on a STM32f411CEU6 (blackpill) board with grbl hal. Does this board has ethernet interface capabilities? Is possible to do so? Saw some GRBl firmwares that connect ehternet via SPI.
Will an UART to ethernet converter work for this purpose? 
Thanks.

hankecnc, 2022-06-07:
F4 does not seem to have its own Ethernet, it needs W5500 module,
This version of grbl has implementation https://github.com/Schildkroet/GRBL-Advanced

But why don't you use ESP32 grbl? You can use esp32 wifi to connect your router, you can use ioSender written by terjeio for ethernet control, I tried it and it worked

terjeio, 2022-06-08:
The simplest road to network connectivity (wifi) is to use a [Wemos D1](https://github.com/terjeio/grblHAL/discussions/341#discussioncomment-1877055) (or similar module). I do not know if there are UART based modules available providing a wired connection.
A W5500 module will require driver code for the SPI interface and the higher level networking protocols as the current [networking plugin](https://github.com/grblHAL/Plugin_networking) protocols are layered on top of the raw lwIP interface.

> F4 does not seem to have its own Ethernet, it needs W5500 module,

AFAIK the F411 does not have an on-chip ethernet peripheral, however the F446 (and some other variants?) has.

osmargarnica, 2022-06-08:
I was thinking about this module:
https://www.wiznet.io/product-item/wiz750sr/
It can be easily configured with their tool.

terjeio, 2022-06-09:
Ok, the WIZ750SR-TTL module is what you want.

osmargarnica, 2022-06-09:
Ok. Thank you very much, I`ll give it a try.

terjeio, 2023-05-21:
The STM32F4xx driver has gotten support for W5100S (untested) and W5500 breakouts. SPI pins has to be defined in the map file for it to work - like in [this map](https://github.com/grblHAL/STM32F4xx/blob/c06947d825e3cbd96a4c0b4f204ad30a7e01a2a6/Inc/st_morpho_map.h#L225-L234).

closed 2023-05-21 by terjeio