grblHAL/STM32F4xx #77: Wrong baud generation for Serial on STM32F407VGT6

ho-ho-ho, 2022-06-13:
I'm using the STM32F4 Discovery which has the STM32F407VGT6 controller onboard and when I tried to connect a device on the USART3 (PD8/9, selected by SERIAL_MOD = 3) I only received corrupted data. So after digging a little more I noticed that in https://github.com/grblHAL/STM32F4xx/blob/c5b8e6a0dd225c07200681ab71bc03c10000e16d/Src/serial.c#L216-L229 the wrong clock domain is used. After changing to the right one I got the correct data :)

From the RM for STM32F405/415, STM32F407/417, STM32F427/437 and STM32F429/439:
> Only USART1 and USART6 are clocked with PCLK2. Other USARTs are clocked with PCLK1

Now my solution so far is a big #if mess, but I'm not sure if that's the best solution right now:
```c
static bool serialSetBaudRate (uint32_t baud_rate)
{
#if defined(STM32F405xx) || defined(STM32F415xx) || defined(STM32F407xx) || defined(STM32F417xx) || defined(STM32F427xx) || defined(STM32F429xx) || defined(STM32F437xx) || defined(STM32F439xx)
#if SERIAL_MOD == 3
    USART->CR1 = USART_CR1_RE|USART_CR1_TE;
    USART->BRR = UART_BRR_SAMPLING16(HAL_RCC_GetPCLK1Freq(), baud_rate);
    USART->CR1 |= (USART_CR1_UE|USART_CR1_RXNEIE);
#else
    USART->CR1 = USART_CR1_RE|USART_CR1_TE;
    USART->BRR = UART_BRR_SAMPLING16(HAL_RCC_GetPCLK1Freq(), baud_rate);
    USART->CR1 |= (USART_CR1_UE|USART_CR1_RXNEIE);
#endif
#else
#if SERIAL_MOD == 2
    USART->CR1 = USART_CR1_RE|USART_CR1_TE;
    USART->BRR = UART_BRR_SAMPLING16(HAL_RCC_GetPCLK1Freq(), baud_rate);
    USART->CR1 |= (USART_CR1_UE|USART_CR1_RXNEIE);
#else
    USART->CR1 = USART_CR1_RE|USART_CR1_TE;
    USART->BRR = UART_BRR_SAMPLING16(HAL_RCC_GetPCLK1Freq(), baud_rate);
    USART->CR1 |= (USART_CR1_UE|USART_CR1_RXNEIE);
#endif
#endif

    return true;
}
```

ho-ho-ho, 2022-08-23:
Works like a charm :)

closed 2022-08-23 by ho-ho-ho

datdadev, 2024-07-18:
Have the devs removed SERIAL_MOD?

terjeio, 2024-07-18:
> Have the devs removed SERIAL_MOD?

Replaced by [SERIALn_PORT](https://github.com/grblHAL/STM32F4xx/blob/17440571911851c55f9aac2de8ff239200df3e04/Src/serial.c#L32-L57).

datdadev, 2024-07-19:
Hi @terjeio, can you guide me how to change the USART default serial port to different USART pins?
I am using STM32F407VET6, in GRBLHAL, there is STM32F407VGT6 using BTT SKR 2.0 board with default USART1, I have changed to USART2 in the code, but it did not work when I connect that 2 pins between the laptop and the MCU
I use STM32CubeIDE, to build the selected project which issues STM32F4 8MHz

This is what I have modified in the code:
```c
#define SERIAL_PORT     2   // GPIOA: TX = 2, RX = 3
```