grblHAL/STM32F4xx #78: Oops, on STM32F401RCT6, some parameters cannot be saved, some parameters can be saved, and some parameters are not saved after power off

hankecnccc, 2022-06-30:
Oops, on STM32F401RCT6, some parameters cannot be saved, some parameters can be saved, and some parameters are not saved after power off, such as the maximum spindle speed, I use the F401blackpill compile option,

hankecnccc, 2022-06-30:
Later I will test to see if the F401CCU6 also has such a problem

hankecnccc, 2022-06-30:
amazing! Some parameters cannot be saved on F401CCU6, such as the maximum spindle speed

hankecnccc, 2022-07-01:
While waiting for your reply, I tested it again. After he needs to set the $31 parameter value, he also needs to set the $36 parameter value before the value can be saved, which is a strange and incomprehensible behavior.

terjeio, 2022-07-10:
I do not have a F401 to test with, however testing with a F411 Blackpill (NVS storage in flash) does not show this issue.
Can you compare the first 512 bytes starting from 0x8004000 before and after saving a setting (but before restart) that does not survive a restart? No changes?

>  After he needs to set the $31 parameter value, he also needs to set the $36 parameter value before the value can be saved, which is a strange and incomprehensible behavior.

The spindle parameters $30 - $36 are [written directly to the settings memory area](https://github.com/grblHAL/core/blob/e8530a45ab5b386aa05956eb368e856ab5c520d9/settings.c#L446-L452) so indeed incomprehensible unless writing the settings memory area to NVS storage (flash or I2C EEPROM) somehow randomly fails.