grblHAL/STM32F4xx #83: error: expected identifier

duinzicht8, 2022-07-24:


duinzicht8, 2022-07-25:
All downloaded today

jan@jan-LubuntuESP32:~$ git clone https://github.com/grblHAL/STM32F4xx.git

Cloning into 'STM32F4xx'...
remote: Enumerating objects: 1156, done.
remote: Counting objects: 100% (570/570), done.
remote: Compressing objects: 100% (98/98), done.
remote: Total 1156 (delta 503), reused 493 (delta 472), pack-reused 586
Receiving objects: 100% (1156/1156), 1.96 MiB | 6.48 MiB/s, done.
Resolving deltas: 100% (723/723), done.

jan@jan-LubuntuESP32:~$ cd STM32F4xx

jan@jan-LubuntuESP32:~/STM32F4xx$ cp -R ~/STM32F4xx/core/* ~/STM32F4xx/grbl

jan@jan-LubuntuESP32:~/STM32F4xx$ /home/jan/.platformio/penv/bin/platformio run --list-targets

Environment               Group     Name    Title    Description
------------------------  --------  ------  -------  -------------
nucleo_f446re_protoneer   Platform  upload  Upload

nucleo_f446re_morpho_cnc  Platform  upload  Upload

btt_skr_2_tmc5160         Platform  upload  Upload

btt_skr_pro_1_1           Platform  upload  Upload

fysetc_s6_tmc2209         Platform  upload  Upload

fysetc_s6                 Platform  upload  Upload

btt_skr_2_tmc2130         Platform  upload  Upload

nucleo_f411re_protoneer   Platform  upload  Upload

blackpill_f411ce          Platform  upload  Upload

nucleo_f411re_morpho_cnc  Platform  upload  Upload

jan@jan-LubuntuESP32:~/STM32F4xx$ /home/jan/.platformio/penv/bin/platformio run --environment nucleo_f446re_protoneer

Processing nucleo_f446re_protoneer (board: nucleo_f446re; platform: ststm32; framework: stm32cube)
-----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f446re.html
PLATFORM: ST STM32 (15.4.1) > ST Nucleo F446RE
HARDWARE: STM32F446RET6 180MHz, 128KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-stm32cubef4 @ 1.26.2 
 - tool-ldscripts-ststm32 @ 0.2.0 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 80 compatible libraries
Scanning dependencies...
Dependency Graph
|-- bluetooth
|-- grbl
|-- keypad
|-- laser
|-- motors
|-- odometer
|-- spindle
|-- Core
|-- Class
|-- App
|-- Target
|-- eeprom
[... 39 lines cut ...]
Compiling .pio/build/nucleo_f446re_protoneer/src/driver.o
Compiling .pio/build/nucleo_f446re_protoneer/src/flash.o
Compiling .pio/build/nucleo_f446re_protoneer/src/i2c.o
Src/driver.c:1235:28: error: expected ')' before '.' token
     spindle_update_caps(hal.spindle.cap.variable);                                                                          
                            ^                                                                                                
Src/driver.c:1237:5: error: expected identifier or '(' before 'return'                                                       
     return true;                                                                                                            
     ^~~~~~                                                                                                                  
Src/driver.c:1238:1: error: expected identifier or '(' before '}' token
 }                                                                                                                           
 ^                                                                                                                           
Src/driver.c: In function 'spindleConfig':                                                                                   
Src/driver.c:1229:5: warning: control reaches end of non-void function [-Wreturn-type]                                       
     }                                                                                                                       
     ^                                                                                                                       
Compiling .pio/build/nucleo_f446re_protoneer/src/ioports.o                                                                   
*** [.pio/build/nucleo_f446re_protoneer/src/driver.o] Error 1
================================================ [FAILED] Took 2.65 seconds ================================================ 

Environment              Status    Duration
-----------------------  --------  ------------
nucleo_f446re_protoneer  FAILED    00:00:02.655
=========================================== 1 failed, 0 succeeded in 00:00:02.655 ===========================================
jan@jan-LubuntuESP32:~/STM32F4xx$

terjeio, 2022-07-25:
Ok, there is a bug lurking after all - it shows up when there is no spindle PWM output defined in the map file. Sorry for not catching that earlier.

Change these lines:

https://github.com/grblHAL/STM32F4xx/blob/d4b3e9ce126e198a8a444957fb2a247ee664cdbe/Src/driver.c#L1224-L1229

to:
```
    } else {
        if(pwmEnabled)
            hal.spindle.set_state((spindle_state_t){0}, 0.0f);
#endif // SPINDLE_PWM_TIMER_N
        hal.spindle.set_state = spindleSetState;
#ifdef SPINDLE_PWM_TIMER_N
    }
#endif
```

duinzicht8, 2022-07-25:
I think it works now!

jan@jan-LubuntuESP32:~/STM32F4xx$ /home/jan/.platformio/penv/bin/platformio run --environment nucleo_f446re_protoneer

Processing nucleo_f446re_protoneer (board: nucleo_f446re; platform: ststm32; framework: stm32cube)
-----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f446re.html
PLATFORM: ST STM32 (15.4.1) > ST Nucleo F446RE
HARDWARE: STM32F446RET6 180MHz, 128KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-stm32cubef4 @ 1.26.2 
 - tool-ldscripts-ststm32 @ 0.2.0 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 80 compatible libraries
Scanning dependencies...
Dependency Graph
|-- bluetooth
|-- grbl
|-- keypad
|-- laser
|-- motors
|-- odometer
|-- spindle
|-- Core
|-- Class
|-- App
|-- Target
|-- eeprom
Building in release mode
Compiling .pio/build/nucleo_f446re_protoneer/src/driver.o
Compiling .pio/build/nucleo_f446re_protoneer/src/main.o
Compiling .pio/build/nucleo_f446re_protoneer/src/serial.o
Compiling .pio/build/nucleo_f446re_protoneer/src/spi.o
Compiling .pio/build/nucleo_f446re_protoneer/src/st_morpho.o
Compiling .pio/build/nucleo_f446re_protoneer/src/st_morpo_dac.o
Compiling .pio/build/nucleo_f446re_protoneer/src/stm32f4xx_hal_msp.o
Compiling .pio/build/nucleo_f446re_protoneer/src/stm32f4xx_it.o
Compiling .pio/build/nucleo_f446re_protoneer/src/syscalls.o
Compiling .pio/build/nucleo_f446re_protoneer/src/sysmem.o
Compiling .pio/build/nucleo_f446re_protoneer/src/system_stm32f4xx.o
Compiling .pio/build/nucleo_f446re_protoneer/src/tmc_uart.o
Compiling .pio/build/nucleo_f446re_protoneer/src/usb_serial.o
Compiling .pio/build/nucleo_f446re_protoneer/FrameworkCMSISDevice/gcc/startup_stm32f446xx.o
Compiling .pio/build/nucleo_f446re_protoneer/FrameworkCMSISDevice/system_stm32f4xx.o
Archiving .pio/build/nucleo_f446re_protoneer/libFrameworkCMSISDevice.a
Indexing .pio/build/nucleo_f446re_protoneer/libFrameworkCMSISDevice.a
Linking .pio/build/nucleo_f446re_protoneer/firmware.elf
Checking size .pio/build/nucleo_f446re_protoneer/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   8.6% (used 11240 bytes from 131072 bytes)
Flash: [==        ]  22.1% (used 115892 bytes from 524288 bytes)
Building .pio/build/nucleo_f446re_protoneer/firmware.bin
================================================ [SUCCESS] Took 2.60 seconds ================================================

Environment              Status    Duration
-----------------------  --------  ------------
nucleo_f446re_protoneer  SUCCESS   00:00:02.602
================================================ 1 succeeded in 00:00:02.602 ================================================

jan@jan-LubuntuESP32:~/STM32F4xx$ /home/jan/.platformio/penv/bin/platformio run -v -e nucleo_f446re_protoneer --target upload

Processing nucleo_f446re_protoneer (board: nucleo_f446re; board_build.ldscript: STM32F411CEUX_FLASH.ld; build_flags: -I ., -I FatFS, -I Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc, -I Middlewares/ST/STM32_USB_Device_Library/Core/Inc, -I USB_DEVICE/Target, -D OVERRIDE_MY_MACHINE, -D BOARD_PROTONEER_3XX=, -D NUCLEO_F446=; lib_deps: bluetooth, grbl, keypad, laser, motors, odometer, spindle, Core, Class, App, Target, eeprom; lib_extra_dirs: ., FatFS, Middlewares/ST/STM32_USB_Device_Library, USB_DEVICE; upload_protocol: stlink; platform: ststm32; platform_packages: framework-stm32cubef4 @ ~1.26.2; framework: stm32cube; lib_archive: False; lib_ldf_mode: off)
-----------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f446re.html
PLATFORM: ST STM32 (15.4.1) > ST Nucleo F446RE
HARDWARE: STM32F446RET6 180MHz, 128KB RAM, 512KB Flash
[... 75 lines cut ...]
Info : flash size = 512 kbytes                                                                                               
Info : Flash write discontinued at 0x080001c4, next section at 0x08008000                                                    
** Programming Finished **                                                                                                   
** Verify Started **                                                                                                         
** Verified OK **                                                                                                            
** Resetting Target **                                                                                                       
Info : Unable to match requested speed 2000 kHz, using 1800 kHz                                                              
Info : Unable to match requested speed 2000 kHz, using 1800 kHz                                                              
shutdown command invoked                                                                                                     
=============================================== [SUCCESS] Took 12.39 seconds =============================================== 

Environment               Status    Duration
[... cut ...]
btt_skr_pro_1_1           IGNORED
btt_skr_2_tmc2130         IGNORED
btt_skr_2_tmc5160         IGNORED
fysetc_s6                 IGNORED
fysetc_s6_tmc2209         IGNORED
nucleo_f411re_morpho_cnc  IGNORED
nucleo_f411re_protoneer   IGNORED
nucleo_f446re_morpho_cnc  IGNORED
nucleo_f446re_protoneer   SUCCESS   00:00:12.387
blackpill_f411ce          IGNORED
================================================ 1 succeeded in 00:00:12.387 ================================================
jan@jan-LubuntuESP32:~/STM32F4xx$

closed 2022-07-26 by terjeio