grblHAL/STM32F4xx #85: After processing nucleo_f446re_protoneer, the stm32nucleo don't startup

duinzicht8, 2022-07-29:
After processing nucleo_f446re_protoneer, the stm32nucleo don't startup.
LD2 don't flash.
The processing was oke, look at ; error: expected identifier #83, "https://github.com/grblHAL/STM32F4xx/issues/83".

terjeio, 2022-07-30:
What do you mean by not starting up?
When I compile and flash with the Protoneer board map it starts up ok.

Is the Protoneer board plugget on? If so can you try with it removed?

> LD2 don't flash.

Sorry, I do not understand this statement.

duinzicht8, 2022-07-30:
>What do you mean by not starting up?<
If I connect the nucleo_f446re connect with the computer, There is no light flashing.
If I startup bCNC there is a  connection via serialport with the ST-LINK, but there is no moving with the x and y axis on the screen.

>Is the Protoneer board plugget on? If so can you try with it removed?<
If the Protoneer board is plugget on or removed it don't matter, there is no difference.

>LD2 don't flash.<
Led1 > LD1 is red or green by conection
Led3 > LD3 is always red light.
Led2 > LD2 is flashing when the software is working.

It seems the software is not working oke, or Did I forget something .... a jumper or a button?

terjeio, 2022-07-30:
If you connect with a terminal program such as PuTTY and try these commands:
```
$I
?
```
Do you get any responses back? If so, what?

> Led1 > LD1 is red or green by conection

This is used by the onboard ST-Link and is blinking when there is a debug session active. If not it is a steady red on my board.

> Led3 > LD3 is always red light.

This is the power on LED so ok.

> Led2 > LD2 is flashing when the software is working.

This is connected to one of the GPIO pins, cannot  remember which. If flashing then this pin is used by one of the grblHAL outputs and and the flashing is purely accidental - e.g. the pin is used as a stepper direction or enable signal..

duinzicht8, 2022-07-31:
This is without Protoneer board.

minicom -D /dev/ttyACM0
$$

$0=10.0
$1=25
$2=0
$3=0
$4=7
$5=0
$9=1
$10=511
$11=0.010
$12=0.002
$13=0
$14=0
$15=0
$16=0
$17=0
$18=0
$20=0
$21=0
$22=0
$23=0
$24=25.0
$25=500.0
$26=250
$27=1.000
$28=0.100
$29=0.0
$32=0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$62=0
$63=2
$64=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=100.0
$384=0
ok

$I

[VER:1.1f.20220722:]
[OPT:NSL,35,1024,3,0]
[NEWOPT:ENUMS,RT+,NOPROBE,ES,TC,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32F446]
[... 113 lines cut ...]
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Controller state changed to: Idle (Running: False)
Job done. Purging the controller. (Running: False)
Stream requested by CYCLE START machine button
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/jan/.local/lib/python3.8/site-packages/bCNC/Sender.py", line 786, in serialIO
    elif self.mcontrol.parseLine(line, cline, sline):
  File "/home/jan/.local/lib/python3.8/site-packages/bCNC/controllers/_GenericController.py", line 217, in parseLine
    self.parseBracketAngle(line, cline)
  File "/home/jan/.local/lib/python3.8/site-packages/bCNC/controllers/GRBL1.py", line 188, in parseBracketAngle
    self.master.event_generate("<<Run>>", when = 'tail')
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1858, in event_generate
    self.tk.call(args)
RuntimeError: main thread is not in main loop

terjeio, 2022-07-31:
OK, so it is bCNC that is crashing.
If you connect to the controller with minicom after bCNC crashes is it still responding? If it is you may try to compile with a different [compatibility level](https://github.com/grblHAL/core/wiki/Compatibility-level).

duinzicht8, 2022-08-07:
It is working now, UGS and bCNC work well, UGS must have an extra configuration; 
1/ "export MESA_LOADER_DRIVER_OVERRIDE=i965".
2/ NucleoF446RE ;  "$10=1".

closed 2023-08-26 by terjeio