grblHAL/STM32F4xx #86: hi, is terjeio there? grblhal if only one system parameter is set at a time, then this system parameter will not be saved after power failure

hankecnccc, 2022-07-30:


hankecnccc, 2022-07-30:
> > I manually sent the command "$x=x", this bug can be reproduced accurately
> 
> I do not have a F401 to check with - the code difference between F446 and F401 is only the clock setup and the loader script since the flash memory layout is different.
> 
> > can you move this problem to STM32F4xx? ? Maybe someone will fix it
> 
> Yes, that someone has to be someone with access to a F401 and preferably a debugger. The initial place to [set a breakpoint is here](https://github.com/grblHAL/core/blob/7dac639da81f4518bb2890d17b84a06867b96b8a/nvs_buffer.c#L310). This is where most of the settings are written on a change.
> 
> Nice board BTW and I guess you are storing the settings in flash since I cannot see any EEPROM/FRAM chip.

hi, I'm very happy, I found the bool protocol_execute_realtime (void) according to the save function you pointed out, I called nvs_buffer_sync_physical(); twice in this function,
problem has been resolved

bool protocol_execute_realtime (void)
{
     if(protocol_exec_rt_system()) {

         if (sys.suspend)
             protocol_exec_rt_suspend();

       #ifdef BUFFER_NVSDATA
        if((state_get() == STATE_IDLE || (state_get() & (STATE_ALARM|STATE_ESTOP))) && settings_dirty.is_dirty && !gc_state.file_run)
        {
     nvs_buffer_sync_physical();
     nvs_buffer_sync_physical();
        }

       #endif
     }

     return !ABORTED;
}

terjeio, 2022-07-31:
I have just committed a change that [outputs a message](https://github.com/grblHAL/core/blob/4b36ffbfbf62fbc0484e907b11cd432182f21cb7/nvs_buffer.c#L363-L364) if writing to flash fails.
And I do not understand how your fix above can work since the dirty flag is reset on the first call...

hanke-cnc, 2022-07-31:
> nvs_buffer_sync_physical();

> I have just committed a change that [outputs a message](https://github.com/grblHAL/core/blob/4b36ffbfbf62fbc0484e907b11cd432182f21cb7/nvs_buffer.c#L363-L364) if writing to flash fails. And I do not understand how your fix above can work since the dirty flag is reset on the first call...

> I have just committed a change that [outputs a message](https://github.com/grblHAL/core/blob/4b36ffbfbf62fbc0484e907b11cd432182f21cb7/nvs_buffer.c#L363-L364) if writing to flash fails. And I do not understand how your fix above can work since the dirty flag is reset on the first call...

I don't know, who cares, anyway, calling nvs_buffer_sync_physical() twice can solve my problem

hanke-cnc, 2022-07-31:
> I have just committed a change that [outputs a message](https://github.com/grblHAL/core/blob/4b36ffbfbf62fbc0484e907b11cd432182f21cb7/nvs_buffer.c#L363-L364) if writing to flash fails. And I do not understand how your fix above can work since the dirty flag is reset on the first call...

In addition, for the details of this bug, if only one parameter is set, and then "$$" is sent, the message obtained is that the parameter is its set value, but when the power of the control board is unplugged and powered on again, the parameter value is not saved.

closed 2022-07-31 by hankecnccc

terjeio, 2022-07-31:
> In addition, for the details of this bug, if only one parameter is set, and then "$$" is sent, the message obtained is that the parameter is its set value

That is normal, there are three copies of the settings when buffering is enabled: the copy used by the core code, the buffered copy where changes are stored before written (together with the dirty flags) and the copy in flash (or in EEPROM). The first two are restored from flash/EEPROM at startup.