grblHAL/STM32F4xx #87: 4-axis with protoneer V3.5 and NucleoF446

duinzicht8, 2022-08-07:
terje io,

Is it possible to make 4-axis run on the NucleoF446RE with protoneer V3.5 board?

terjeio, 2022-08-11:
Yes, change [N_AXIS to 4](https://github.com/grblHAL/core/blob/53855a4bf8a2c31fae417c692d512cd10af00b80/config.h#L35-L41) and rebuild.

jielemon, 2022-09-15:
I changed N_AXIS to 4 and rebuilt. But looks like the ioSender (2.0.40) installed on my Windows10 does not support the A axis. When I send "G0 A5", the console returns error 20. How can I change the grbl settings to support A, B, C axes? or do I need to define ST_ROTATIONAL_MASK (A_AXIS_BIT|B_AXIS_BIT|C_AXIS_BIT) and rebuild?

terjeio, 2022-09-16:
Changing `N_AXIS` should be enough.
Is the A axis showing up in the ioSender DRO or not? If not then the rebuild/reflash did not work out as it should?
What is the $I response (or what is showing in the ioSender _Help > About dialog_)?

jielemon, 2022-09-16:
> Changing `N_AXIS` should be enough. Is the A axis showing up in the ioSender DRO or not? If not then the rebuild/reflash did not work out as it should? What is the $I response (or what is showing in the ioSender _Help > About dialog_)?

Thank you for your prompt reply.

Rebuilt with "N_AXIS = 4". This time the A axis shows up in the ioSender DRO, but the red square above "A" in Signals section is not there. And both the lower and upper limits of A axis is 0 (I disabled soft and hard limits). BTW, how to program limits for X, Y, Z, A axes?
[image]

Here is the response of $I:
[VER:1.1f.20220906:]
[OPT:NSL,35,1024,4,0]
[NEWOPT:ENUMS,RT+,HOME,NOPROBE,TC,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32F446]
[DRIVER VERSION:220907]
[BOARD:Protoneer v3]
[VER:1.1f.20220906:]
[OPT:NSL,35,1024,4,0]
[NEWOPT:ENUMS,RT+,HOME,NOPROBE,TC,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32F446]
[DRIVER VERSION:220907]
[BOARD:Protoneer v3]

terjeio, 2022-09-16:
> the red square above "A" in Signals section is not there

Check the signal level on the A limit pin, use $pins to find it if you do not know. The input might also be inverted with the $5 setting. I normal operation (no limit or control switches asserted) all red lights should be off. Oh wait, the Protoneer board does not have an input for A-limit so it will always be off.

> BTW, how to program limits for X, Y, Z, A axes?

Use $130-$133 - axis maximum travel. Set $133 to 0 if no limit (A is a rotary) and $376 to 1 if you ever use inches mode (this disables inches to mm conversion for A).

terjeio, 2022-09-16:
> BTW, how to program limits for X, Y, Z, A axes?

Perhaps you meant the _Program limits_ box in ioSender - the data shown there is calculated from movements in the loaded gcode program.

jielemon, 2022-09-17:
Finally got the A axis moved. Thank you so much for your help, Terje Io.

Does PROTONEER_3XX support 5- or 6-axis if changing N_AXIS to 5 or 6? If so, what other edits should be made?

terjeio, 2022-09-18:
> Does PROTONEER_3XX support 5- or 6-axis if changing N_AXIS to 5 or 6? If so, what other edits should be made?

The protoneer shield has, AFAIK, only four motor driver sockets so no. If there are versions with more than four sockets a new map file should be created with definitions for M4 and M5 motor pins (if there are any free pins available for that).

closed 2023-08-26 by terjeio