grblHAL/STM32F4xx #91: STM32F446RE (BOARD_PROTONEER_3XX) VCP

jielemon, 2022-09-06:
Virtual COM Port can't be detected.

jielemon, 2022-09-06:
Windows can only find the STLink Virtual COM Port.

terjeio, 2022-09-06:
This is with a Nucleo-64 board? If so then there is no USB port available with direct connection to the MCU. You have to use the STLink virtual com port that connects to the MCU via UART.

jielemon, 2022-09-06:
Yes, with a Nucleo-64 board. I tried to use the STLink virtual com port detected by Windows to communicate with the target MCU (via Candle), but can't get connected.

terjeio, 2022-09-06:
Have you selected a Nucleo-64 build from the Build drop down (hammer icon) or Debug drop down (bug icon)?
Many of the available build targets uses native USB for comms and won't work.

jielemon, 2022-09-06:
I selected "Debug F446 Nucleo64" from the Build drop down.  Does that one work? If not, which one should I select?

jielemon, 2022-09-07:
Got responses if connected with PuTTY, but can't execute GCODE correctly. BTW, what is your PuTTY configurations for the serial port? 
1. If the reset button is pressed, the response is as below:
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:'$H'|'$X' to unlock]
2. If I type "$$", the response is as below:
GrblHAL 1.1f ['$' or '$HELP' for help]
[MSG:'$H'|'$X' to unlock]
$0=10.0
$1=25
$2=0
$3=0
$4=7
$5=0
$10=511
$11=0.010
$12=0.002
$13=0
$14=0
$15=0
$16=0
$17=0
$18=0
$20=0
$21=0
$22=0
$23=0
$24=25.0
$25=500.0
$26=250
$27=1.000
$28=0.100
$29=0.0
$32=0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$62=0
$63=2
$64=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=100.0
$384=0
ok
3. If I type "G0 X5", it returns "error: 9".
4. If I type "$H", it returns "error: 5".
5. If I type "$X", it returns "error: 18".

jielemon, 2022-09-07:
I also installed the TM Terminal plug-in in Eclipse and got the same responses as in PuTTY.

dresco, 2022-09-07:
Is probably in an alarm state, and the error responses are because it's locked.. What does typing in `?` respond with?

jielemon, 2022-09-07:
> Is probably in an alarm state, and the error responses are because it's locked.. What does typing in `?` respond with?
Exactly as what you said. It's in an alarm state. It responded with the following when typing in '?'.
<Alarm|MPos:0.000,0.000,0.000|Bf:35,1023|F:0|Pn:XYZRHS|Ov:100,100,100>
And it stays in the alarm state after the "reset" button is pressed.
How to go into a normal state? 
Thank you for your help!

jielemon, 2022-09-07:
> > Is probably in an alarm state, and the error responses are because it's locked..
> 
> It is...
> 
> $H returns error 5 because homing is not enabled - `$22=0`, set to at least 1 to enable. Send `$$=22` to list options. $X returns error 18 because the [reset/e-stop input is asserted](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings), connect it to ground (or invert it by $14) and try again.
> 
> Since you are using Windows [ioSender](https://github.com/terjeio/ioSender) can be used for easy access to setting options.

Thank you for the reply.   Homing is enabled by sending "$22=1".  What should I type in to invert the reset/e-stop input?