grblHAL/STM32F4xx #93: No source available for "Reset_Handler() at 0x800ad40" with BOARD_PROTONEER_3XX and Nucleo F446RE

jielemon, 2022-09-12:
The active build configuration is "Debug F446 Nucleo64". The console shows that "Download verified successfully" and the target MCU can execute the gcode correctly. 
But why this happened?

terjeio, 2022-09-14:
I have no idea, when I search for _Reset_Handler_ in the source code nothing is found in .c or .h files, however there is a short assembly function in the startup code with that name.  Which version of the IDE are you using?

jielemon, 2022-09-14:
STM32CubeIDE
Version: 1.9.0
Build: 12015_20220302_0855 (UTC)

dresco, 2022-09-14:
Can confirm I see the same now on a Nucleo F446RE build. I did update the STM32F4 MCU package a while ago (to v1.27.1) - it may have started then?

jielemon, 2022-09-15:
I downloaded the latest STM32F4xx-master.zip and core-master.zip. And the gcodes I sent were as following:
G0 X5 Y5 Z5
G0 X-2 Y-2 Z-2
G0 X1 Y1 Z1
The MCU can start the cycle and jog normally on all the three axes though with the Reset_Handler issue.

dresco, 2022-09-15:
Yeah, I was just referring to the version of the F4 MCU support package within STMCubeIDE. 

I **was** seeing the same issue as you. After an STLink upload (not debug), the editor window would change to display the `No source available for Reset_Handler()` message, as if it were a debug session with missing debug information. 

But I just tried downgrading the F4 support package, and it went away. Have again reinstalled the latest, and now I can't recreate it at all. Pretty confident is just a glitch in the IDE, rather than any problem with the generated code..

jielemon, 2022-09-15:
> STM32F4 MCU package

When you reinstalled the latest, did you install the latest patch?

dresco, 2022-09-16:
Hi @jielemon

I just used the Help -> Manage Embedded Software Packages menu. 

I added 1.26.2 & removed 1.27.1, and the warning went away. But it didn't reappear when I re-added 1.27.1 & removed 1.26.2

I did also clean and close the project, and exit and re-open the IDE in between each test - is possible that cleared something up..?

jielemon, 2023-03-19:
Hi Terjeio,

I can't find the "STM32F4xx" directory on https://github.com/grblHAL. Have you moved it? Will you update STM32F4xx  in the future?

Thanks.

terjeio, 2023-03-19:
> I can't find the "STM32F4xx" directory on https://github.com/grblHAL. Have you moved it?

It is still [there](https://github.com/grblHAL/STM32F4xx) - FYI github moves the repositories in the list, those who has the most recent commits get promoted to the top for some reason.

closed 2023-08-26 by terjeio