grblHAL/STM32F4xx #94: I don't know how to get the 6 axis compilation to pass, got an error

hanke-cnc, 2022-09-22:
I have noticed that some users who purchased the grblhal control board are not used for cnc engraving machine, but pure motor control, so I tried to compile 6 axes, but encountered an error,
I defined N_AXIS 6 in grbl-config.h,
cnc3040_map.h is enabled, and M5 is defined
, but encountered a compilation error (the five-axis can be compiled and passed)
[image]

terjeio, 2022-09-22:
> cnc3040_map.h is enabled, and M5 is defined

You have modified the map file?

hanke-cnc, 2022-09-22:
> > cnc3040_map.h is enabled, and M5 is defined
> 
> You have modified the map file?

no, i just defined m5
[image]

hanke-cnc, 2022-09-22:
Five axes can be perfectly compiled through

terjeio, 2022-09-22:
>>     You have modified the map file?

> no, i just defined m5

Well, that is a modification...

There are two missing commas, [here](https://github.com/grblHAL/core/blob/c7c64366e39f8bcab612d24f9196ab9d6d7e58ad/stepdir_map.h#L84) and [here](https://github.com/grblHAL/core/blob/c7c64366e39f8bcab612d24f9196ab9d6d7e58ad/stepdir_map.h#L163). I you beat me to it because I am about to update this file to support 8 axes by generating the file programmatically.

terjeio, 2022-09-22:
Use this file, there is a couple of further typos/bugs in the original.

[stepdir_map.zip](https://github.com/grblHAL/STM32F4xx/files/9623291/stepdir_map.zip)

closed 2022-09-22 by hanke-cnc