grblHAL/STM32F4xx #101: Compilation error with Nucleo-F446RE (BOARD_PROTONEER_3XX)

jielemon, 2022-10-31:
Hi,

I just downloaded the newest versions of grblHAL for STM32F4xx and core, but the build failed.
https://github.com/grblHAL/STM32F4xx
https://github.com/grblHAL/core

Could you please look into the problem? Thank you!

Best regards,
Judy

PS: Console messages
12:18:11 **** Build of configuration Debug F446 Nucleo64 for project GRBL Driver STM32F401 ****
make -j4 all 
arm-none-eabi-gcc "../grbl/alarms.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DSTM32F446xx -DNUCLEO_F446 -DDEBUG -c -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401" -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"grbl/alarms.d" -MT"grbl/alarms.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "grbl/alarms.o"
arm-none-eabi-gcc "../grbl/coolant_control.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DSTM32F446xx -DNUCLEO_F446 -DDEBUG -c -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401" -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"grbl/coolant_control.d" -MT"grbl/coolant_control.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "grbl/coolant_control.o"
arm-none-eabi-gcc "../grbl/corexy.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DSTM32F446xx -DNUCLEO_F446 -DDEBUG -c -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401" -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"grbl/corexy.d" -MT"grbl/corexy.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "grbl/corexy.o"
arm-none-eabi-gcc "../grbl/errors.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DSTM32F446xx -DNUCLEO_F446 -DDEBUG -c -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401" -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"grbl/errors.d" -MT"grbl/errors.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "grbl/errors.o"
arm-none-eabi-gcc "../grbl/gcode.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DSTM32F446xx -DNUCLEO_F446 -DDEBUG -c -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401" -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"grbl/gcode.d" -MT"grbl/gcode.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "grbl/gcode.o"
arm-none-eabi-gcc "../grbl/grbllib.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DSTM32F446xx -DNUCLEO_F446 -DDEBUG -c -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401" -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"grbl/grbllib.d" -MT"grbl/grbllib.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "grbl/grbllib.o"
arm-none-eabi-gcc "../grbl/ioports.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DSTM32F446xx -DNUCLEO_F446 -DDEBUG -c -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401" -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"grbl/ioports.d" -MT"grbl/ioports.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "grbl/ioports.o"
arm-none-eabi-gcc "../grbl/machine_limits.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DSTM32F446xx -DNUCLEO_F446 -DDEBUG -c -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401" -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"grbl/machine_limits.d" -MT"grbl/machine_limits.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "grbl/machine_limits.o"
arm-none-eabi-gcc "../grbl/maslow.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DSTM32F446xx -DNUCLEO_F446 -DDEBUG -c -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401" -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"grbl/maslow.d" -MT"grbl/maslow.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "grbl/maslow.o"
arm-none-eabi-gcc "../grbl/motion_control.c" -mcpu=cortex-m4 -std=gnu11 -g -DUSE_HAL_DRIVER -DSTM32F446xx -DNUCLEO_F446 -DDEBUG -c -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401" -I"C:/LJ/WFH/STM32F/STM32CubeIDE/workspace_1.9.0a/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"grbl/motion_control.d" -MT"grbl/motion_control.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "grbl/motion_control.o"
[... cut ...]
  371 |   #include "motors/trinamic.h"
      |            ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [USB_DEVICE/App/subdir.mk:25: USB_DEVICE/App/usbd_cdc_if.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [Src/subdir.mk:73: Src/btt_skr_1.1.o] Error 1
make: *** [Src/subdir.mk:73: Src/btt_skr_2.0.o] Error 1
make: *** [Src/subdir.mk:73: Src/diskio.o] Error 1
"make -j4 all" terminated with exit code 2. Build might be incomplete.

12:18:25 Build Failed. 13 errors, 0 warnings. (took 13s.178ms)

terjeio, 2022-10-31:
`../Inc/protoneer_3.xx_map.h:27:2: error: #error "Trinamic plugin not supported!"`

IIRC it is not possible to use Trinamic drivers (in SPI or UART mode that is) with the Protoneer board without wiring the signals with "flying leads" to the relevant driver pins (which has to be cut off or soldered so they point upwards instead of into the socket). And for this to be feasible the signals required has to be available from unused pins on the ST Morpho headers.
Perhaps it is possible - but then what kind of Trinamic drivers in which configuration?

jielemon, 2022-10-31:
> `../Inc/protoneer_3.xx_map.h:27:2: error: #error "Trinamic plugin not supported!"`
>
> IIRC it is not possible to use Trinamic drivers (in SPI or UART mode that is) with the Protoneer board without [...]

Thank you for your prompt reply.
If I do not need the Trinamic plugin right now, where can I uncomment the Trinamic  macros?

jielemon, 2022-10-31:
> Thank you for your prompt reply. If I do not need the Trinamic plugin right now, where can I uncomment the Trinamic macros?

Sorry, should be comment TRINAMIC.

terjeio, 2022-10-31:
My bad - I have left a line uncommented. Comment out this line too fix:

https://github.com/grblHAL/STM32F4xx/blob/f94850d64cac7787bea9645daa4402308ecc6508/Inc/my_machine.h#L71

closed 2023-08-26 by terjeio