grblHAL/STM32F4xx #102: hi, axis pulse and direction are defined in other ports, what should I do if there is no pulse output?

hanke-cnc, 2022-10-31:
hi, since terjeio has updated 8 axes for grblhal, I am modifying the original circuit board, I use the cnc3040_map.h file, but since the GPIOA port is not enough, I define the C, U, V axes in GPIOB, but there is no pulse Output, if there is output defined on port A, it means that it can work normally,

hanke-cnc, 2022-10-31:
Where will the final pulse output of grbl be called? Can I directly modify it to HAL_GPIO_TogglePin? This way, I don’t have to trouble terjeio

terjeio, 2022-11-01:
Max number of axes supported by drivers varies, the core and currently only the [STM32F7xx driver](https://github.com/grblHAL/STM32F7xx/blob/main/Src/driver.c) supports up to 8. Other drivers/board maps has to be modified and tested when max number of axes is increased.

> Where will the final pulse output of grbl be called?

Pulse output for the STM32F4xx driver is in _driver.c_ [here](https://github.com/grblHAL/STM32F4xx/blob/9f2a250e1e35545627bd1bd06fdfdaeb4708e611/Src/driver.c#L509-L563) and [here](https://github.com/grblHAL/STM32F4xx/blob/9f2a250e1e35545627bd1bd06fdfdaeb4708e611/Src/driver.c#L574-L627). Handling of direction output, motor enable and possibly limit inputs has to be added as well - and definitions for the pins added at the [start of driver.c](https://github.com/grblHAL/STM32F4xx/blob/9f2a250e1e35545627bd1bd06fdfdaeb4708e611/Src/driver.c#L218-L336). Trinamic driver support requires even more pins/code (and testing)... 

> Can I directly modify it to HAL_GPIO_TogglePin?

No, you should use the same methods for output as used in the driver. There are usually several options, with different overhead and flexibility for port use. Fastest is when all pins in a group (such as step output) can be set in a single read/modify/write cycle, this obviously requires all the pins to be assigned to the same port. Slowest is when each pin is set separately, allowing them to be spread over available ports.

closed 2022-11-01 by hanke-cnc