grblHAL/STM32F4xx #103: Bugs - to be fixed in the next commit. Z-limit has to be polled since it has the same pin number as X preventing enabling interrupt handling for it.

jielemon, 2022-11-01:
Bugs - to be fixed in the next commit. Z-limit has to be polled since it has the same pin number as X preventing enabling interrupt handling for it.

_Originally posted by @terjeio in https://github.com/grblHAL/STM32F4xx/issues/99#issuecomment-1277043161_

jielemon, 2022-11-01:
Now Homing in XYZ axes still works when "$H" is sent. But all the three hard limit switches in the XYZ axes can't be triggered.
Additionally, if a command of single axis homing is sent, like "$HX", "$HY", or "$HZ", it stays in "homing" status even if it finishes homing. I tried to reset the machine, then alarm 6 was triggered. And I had to unlock to proceed.

terjeio, 2022-11-01:
>  But all the three hard limit switches in the XYZ axes can't be triggered.

All three works for me for triggering hard limit alarm. Did you forget to enable hard limits with $21=1?

> Additionally, if a command of single axis homing is sent, like "$HX", "$HY", or "$HZ", it stays in "homing" status even if it finishes homing.

Oops, a regression. This line

https://github.com/grblHAL/core/blob/bf6823abf91986bc840e5e9497ed9b861000b5d8/motion_control.c#L891

should be

` homed_status = limits_go_home(cycle);`

I'll commit a fix later (will be a few days).

jielemon, 2022-11-02:
>But all the three hard limit switches in the XYZ axes can't be triggered.

>All three works for me for triggering hard limit alarm. Did you forget to enable hard limits with $21=1?

I did enable hard limits with Settings: Grbl, otherwise the homing would not work.

Just sent "$21=0" and "$21=1" with MDI input, weird that there was no response (the Settings: Grbl were not changed).

BTW, is it possible to send the grbl commands in a CNC file?

Thank you for your help.

jielemon, 2022-11-02:
| But all the three hard limit switches in the XYZ axes can't be triggered.

| All three works for me for triggering hard limit alarm. Did you forget to enable hard limits with $21=1?

| I did enable hard limits with Settings: Grbl, otherwise the homing would not work.

|Just sent "$21=0" and "$21=1" with MDI input, weird that there was no response (the Settings: Grbl were not changed).

Sorry, my bad. I mixed homing enable up with hard limit enable. All three work now for triggering hard limit alarm. And $21=1 works as well.

jielemon, 2022-11-02:
| https://github.com/grblHAL/core/blob/bf6823abf91986bc840e5e9497ed9b861000b5d8/motion_control.c#L891

|should be

|homed_status = limits_go_home(cycle);

Changed this line as suggested and rebuilt. The homing in Z axis failed with alarm 8 even if I increased the pull-off setting to 10mm. The XY axes completed homing with 10mm pull-off.